lvgl/lv_drivers #5: mouse_hid driver shoud rename to evdev and support EV_ABS

lbzhung, 2018-03-19:
Hi, 

I use littlevgl in embed linux, and i have a touch screen use input evdev interface. I notice that mouse_hid use evdev interface, and only support EV_REL event (my touch screen need support EV_ABS and BTN_TOUCH event). So maybe mouse_hid shoud rename to evdev more precisely and add EV_ABS and BTN_TOUCH support.

Below is the patch to support EV_ABS and BTN_TOUCH:

diff --git a/indev/mouse_hid.c b/indev/mouse_hid.c
index b2e91de..6907c32 100644
--- a/indev/mouse_hid.c
+++ b/indev/mouse_hid.c
@@ -76,15 +76,22 @@ bool mouse_hid_read(lv_indev_data_t * data)
             if (in.code == 1)
                 mouse_hid_root_y += in.value;
         } else if (in.type == 1) {
-            if (in.code == 272) {
+            if (in.code == 272 || in.code == BTN_TOUCH) {
                 if (in.value == 0)
                     mouse_hid_button = LV_INDEV_STATE_REL;
                 if (in.value == 1)
                     mouse_hid_button = LV_INDEV_STATE_PR;
             }
+        } else if (in.type == EV_ABS)
+        {
+            if (in.code == 0)
+                mouse_hid_root_x = in.value;
+            if (in.code == 1)
+                mouse_hid_root_y = in.value;
         }
     }
 
+
     if (mouse_hid_root_x < 0)
         mouse_hid_root_x = 0;
     if (mouse_hid_root_y < 0)

kisvegabor, 2018-03-19:
Hi,

The improvement and the renaming is also a good idea! Thank you!

Can you create pull request to better see the changes?

lbzhung, 2018-03-20:
OK, i will.

kisvegabor, 2018-03-20:
I merged the pull request!
Thank you very much!

closed 2018-03-20 by kisvegabor