lvgl/lv_drivers #10: IRQ and input

Zaltora, 2018-04-10:
I think i found a way to use IRQ to read input device without need change lvgl.  This way to work seems good to you ?  I use 5-2 dev because input_dev got user_data.
in the input library i will get input function and irq function like this:

Library (WIP):
```c
static bool ar10xx_input_get(lv_indev_data_t * data)
{
    ar10xx_t* dev = data->user_data;  //get input device
    if (!dev->flag) //no irq, return false
        return false;
   //
    int16_t xpos;
    int16_t ypos;
    bool pen = AR10XX_GetPos(&xpos, &ypos);
    /*Store the collected data*/
    data->point.x = xpos;
    data->point.y = ypos;
    data->state = pen ? LV_INDEV_STATE_PR : LV_INDEV_STATE_REL;
    if(dev->flag) dev->flag-- ;
    return (dev->flag) ? true : false;
}

inline void ar10xx_irq(const* ar10xx_t dev)
{
        dev->flag++ ; //event on irq
}
```
Main:
```c
void indev_init(void)
{
    lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);          /*Basic initialization*/
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read = ar10xx_input_get;         /*This function will be called periodically*/
    indev_drv.user_data = &ar10xx_dev,
    lv_indev_drv_register(&indev_drv);
}

inline void interupt_fn(void)
{
    ar10xx_irq(&dev);
}
```
lvgl functions are called constantly. I wonder if we can activate a task only when an event incoming.
A bit like using "Queue" in FreeRTOS. The task is waiting for a new information (here the flag) to activate the task.
The advantage of the interupt system above is do not do transmission when it is not needed.

kisvegabor, 2018-04-10:
Hi,

It looks good to me as well! As I see the `user_data` increases only the flexibility because you use directly the `ar10xx_t dev` if you are if you have only one touch controller from this type (no multiple display). Is it correct?

I didn't plan to use a more complex OS in LittlevGL hence now you have the opportunity to immediately return from the task if the condition is not met. I has only a very small overhead. (NO much more then checking the condition before calling the task)

Zaltora, 2018-04-10:
You re right,
I will use a static variable to store indev. lvgl API will be upgraded later for multi-display usage.
Okay, no more complexity :)
If you want overhead the return system, do it with "int", it will prepare error management. ( same for flush/map/fill). What did you think ?

kisvegabor, 2018-04-10:
Okay :)

For error handling: Good idea! In v6.0 this part of the API also should be updated as well!

closed 2018-04-10 by Zaltora