lvgl/lv_drivers #14: SSD1306 (probably SH1106)

jano2358, 2018-06-24:
Hi!

I'm trying the driver (new_api) for the screen I have (it was sold as a SSD1306, but works better when I select SH1106 in the typedef). This one have the pins for I2C.
I'm using this with an STM32F91RC nucleo board, to make the code I like IDEs and as much debugging possibilities as I can have, then I'm using Eclipse with the GNU MCU Eclipse plugins + OCD, and CubeMX to generate the base.
I don't know if this is the 'correct' way, but the only way I was able to make this work was adding a call for the function ssd1306_load_frame_buffer(&dev) in the main loop of my main, don't know if this is documented somewhere, I didn't find it, just figured out.

here is the relevant part of the main:

  lv_init();
  ssd1306_t scr;
scr.height = 64;
scr.width =128;
scr.protocol = SSD1306_PROTO_I2C;
scr.screen = SH1106_SCREEN; //SSD1306_SCREEN ; //
scr.i2c_dev = &hi2c1;
scr.rst_pin = 0;
  ssd1306_init(&scr);

  lv_disp_drv_t disp_drv;                         /*Descriptor of a display driver*/
  lv_disp_drv_init(&disp_drv);

 disp_drv.disp_flush = ssd1306_flush;            /*Used in buffered mode (LV_VDB_SIZE != 0  in lv_conf.h)*/
//  disp_drv.disp_fill = ssd1306_fill;              /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
//  disp_drv.disp_map = ssd1306_map;                /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/

  lv_disp_drv_register(&disp_drv);

///////////////////////////*******\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\

  lv_obj_t * label1 =  lv_label_create(lv_scr_act(), NULL);

  /*Modify the Label's text*/
  lv_label_set_text(label1, "Hello world!");

  /* Align the Label to the center
   * NULL means align on parent (which is the screen now)
   * 0, 0 at the end means an x, y offset after alignment*/
  lv_obj_align(label1, NULL, LV_ALIGN_CENTER, 0, 0);

  /* USER CODE END 2 */

  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
	  HAL_Delay(5);
	  lv_tick_inc(5);
	  lv_task_handler();
	  if (ssd1306_need_redraw()) ssd1306_load_frame_buffer(&scr);

  /* USER CODE END WHILE */

  }

the next problem I had was with the function to talk with i2c in lv_drv_conf.h, here is the code I wrote:
static inline int lv_i2c_write(lv_i2c_handle_t i2c_dev, uint8_t reg, void* data_out, uint8_t datalen)
{
    //Do the dependant port here
	uint8_t buffer[datalen +1];
	uint8_t *ptr = data_out;
	buffer[0] = reg;

	for (int i = 1; i <= datalen +1; i++) buffer[i] = *(ptr++);
    HAL_I2C_Master_Transmit(i2c_dev, 0x78, buffer ,datalen + 1, HAL_MAX_DELAY);
}

this function is called from lv_drv_common.h with 
static inline int i2c_send(const lv_i2c_handle_t i2c_dev, uint8_t reg, uint8_t* data, uint8_t len)
{
    return lv_i2c_write(i2c_dev, &reg, data, len);
}
which is called form SSD1306.c in the command and data sections, the command is more easy to read:
err_control(i2c_send(dev->i2c_dev, 0, &cmd, 1));

when the function was called all the time with the third argument as '0' I never get a 0 at the interface (was reading that with the scope all the time), I tried several ways to write 'reg' but I always get 63 or 60 or 6B when should be 0 and 47 when should be 40 (both hex)
I ended writing the function inside ssd1306_command and _load_frame_buffer, like: 
    {
#if (SSD1306_I2C_SUPPORT)
    case SSD1306_PROTO_I2C:
//        err_control(i2c_send(dev->i2c_dev, 0, &cmd, 1));
        ;
        uint8_t buffer[2];
        buffer[0] = 0x00;
       	buffer[1] = cmd;

        HAL_I2C_Master_Transmit(&hi2c1, 0x78, buffer, 2, HAL_MAX_DELAY);
        break;
the other is a bit more long but same idea.

Any idea about what I'm doing wrong here?

and lastly, in the init procedure, this option should be false:
            && !ssd1306_set_whole_display_lighting(dev, false)	// was true!!
if is set to true the whole screen remains white.

Zaltora, 2018-06-25:
that the way i use my ssd1306 (work too with the sd1106) in spi3 protocol.
The next big PR will be documentation and fixes. (like for ssd1306 init protocol).
In task:
```c
static void _vTaskScreen(void *pvParameters)
{
    while(1)
    {
            lv_task_handler();
            if(ssd1306_need_redraw())
            {
                ssd1306_load_frame_buffer(&xScreenDev);
            }
    }
}
```
in a timer:
```c
static void _vTimerTickCounting(TimerHandle_t xTimerHandle)
{
    lv_tick_inc(coreDISPLAY_TICK);
}
```
at start:
```c
    ssd1306_t* _pxdev = &xScreenDev;

    _pxdev->spi_dev = (lv_spi_handle_t)&_xSpiDev;
    _pxdev->protocol = SSD1306_PROTO_SPI3;
    _pxdev->screen = SH1106_SCREEN; //SSD1306_SCREEN,  SH1106_SCREEN
    _pxdev->width    = DISPLAY_WIDTH;
    _pxdev->height   = DISPLAY_HEIGHT;
    _pxdev->rst_pin  = DISPLAY_RST;

    while (ssd1306_init(_pxdev) != 0)
    {
        debug("%s: failed to init SSD1306 lcd\n", __func__);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    //ssd1306_set_whole_display_lighting(&xScreenDev, true);
    ssd1306_set_whole_display_lighting(_pxdev, false);

    //inverse screen (180°)
    ssd1306_set_scan_direction_fwd(_pxdev,true);
    ssd1306_set_segment_remapping_enabled(_pxdev, false);

    //lvgl register
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    /*Set up the functions to access to your display*/
    disp_drv.disp_flush = ssd1306_flush; /*Used in buffered mode (LV_VDB_SIZE != 0  in lv_conf.h)*/
    disp_drv.disp_fill = ssd1306_fill;//ex_disp_fill;   /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
    disp_drv.disp_map = ssd1306_map;//ex_disp_map;    /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
    lv_disp_drv_register(&disp_drv);

```
Second Problem:
----------------------------------------------------

i found a bug in your i2C. you write  into "datalen+1",  you erase your "data_out pointer".
The code above will fix it (i guess)

```c
static inline int lv_i2c_write(lv_i2c_handle_t i2c_dev, uint8_t reg, void* data_out, uint8_t datalen)
{
    //Do the dependant port here
    uint8_t buffer[datalen +1];
     uint8_t *ptr = data_out;
    buffer[0] = reg;

    for (int i = 1; i < (datalen+1) ; i++)
    {
        buffer[i] = *(ptr++);
    }
    HAL_I2C_Master_Transmit(i2c_dev, SSD1306_I2C_ADDR_0 << 1, buffer ,datalen + 1, HAL_MAX_DELAY);
}
```

jano2358, 2018-06-25:
Hi!

Adding a timer with lv_tick_inc, was already in my todo list, first I wanted to have it working :)
the form of the function you recommend me didn't work either, I tried it for commands and I get 7B when the command is single and 77 when the command is double, but both numbers are followed by a non operative code (something my scope can't recognize as a valid code, I also tried several speeds from 1MHz to 100k, it's always the same)
In the other hand, I use (almost) the same code in the _load_frame_buffer function:

`else
        {
            for (j = y1; j <= y2; j++)
            {
                err_control(sh1106_go_coordinate(dev, x1, j));
                for (i = x1; i <= x2 ; ) //column
                {
                     if((x2-i) >= 15) //(x2-i+1) >= 16
                    {
//                        err_control(i2c_send(dev->i2c_dev, 0x40, &_buffer[j*dev->width+i], 16));

                        uint8_t buffer[17];
			uint8_t *ptr = &_buffer[j*dev->width+i];
			buffer[0] = 0x40;

			for (int i = 1; i <= 16; i++) buffer[i] = *(ptr++);
			HAL_I2C_Master_Transmit(&hi2c1, 0x78, buffer, 17, HAL_MAX_DELAY);

                        i+=16;
                    }
                    else
                    {
//                        err_control(i2c_send(dev->i2c_dev, 0x40, &_buffer[j*dev->width+i], x2-i+1));

                        uint8_t buffer[x2-i+2];
			uint8_t *ptr = &_buffer[j*dev->width+i];
			buffer[0] = 0x40;

			for (int i = 1; i <= (x2-i+1); i++) buffer[i] = *(ptr++);
			HAL_I2C_Master_Transmit(&hi2c1, 0x78, buffer, x2-i+2, HAL_MAX_DELAY);

                        break; //Last data send for this page
                    }
                }
            }`

you can see it's in the original form I copied it before in lv_drv_conf.h, all the codes have the 0x40 head and they arrive to the screen without problems
&xScreenDev is something you defined in that particular code, I don't have anything named like that (just wondering, do we have the same version?, how to know it?)

Zaltora, 2018-06-27:
Sry for `&xScreenDev `, it is specific to my app code. The real basic delaration for declare a driver screen is:
```c
static ssd1306_t xScreenDev; 
```
your i2c_dev need be static too (or global) and you can name like you want :)

---
I see a strange bug. the prototype in the lib is :
```c
static inline int lv_i2c_write(lv_i2c_handle_t i2c_dev, const uint8_t* reg, const void* data_out, uint16_t datalen);
```
your prototype from "comment 1" is :
```c
static inline int lv_i2c_write(lv_i2c_handle_t i2c_dev, uint8_t reg, void* data_out, uint8_t datalen);
```
i didn't notice that when i send you the new function. You are up to date ?

---
After do test with SH1106 and I2C setting, i see normal data go into parameters. try to see the output `printf` of this function (updated with the correct api) if it is possible and compare to your scope value:

```c
static inline int lv_i2c_write(lv_i2c_handle_t i2c_dev, const uint8_t* reg, const void* data_out, uint16_t datalen)
{
    //
    uint8_t buffer[datalen +1];
     uint8_t *ptr = data_out;
    buffer[0] = *reg;

    printf("i2c Transaction: %02X\n", *reg);
    for (uint16_t i = 0 ; i < datalen ; i++)
    {
      printf("%02X ",ptr[i]); 
    }
    printf("\n");

    for (uint16_t i = 1; i < (datalen+1) ; i++)
    {
        buffer[i] = *ptr;
        ptr++;
    }
    HAL_I2C_Master_Transmit(i2c_dev, SSD1306_I2C_ADDR_0 << 1, buffer ,datalen + 1, HAL_MAX_DELAY);
}

```

jano2358, 2018-07-01:
hi!

In the first comment I made, I have changed the pointers in the function declaration to test what was not working, but seems like I changed something else (who knows what or where) and the real problem was as you said before in the erase your "data_out pointer" (I don't know yet what that means, there is some obscure parts of the pointers I don't know yet), then when I paste the first code you send me the damage was done and it didn't work. 
Today I made a pull from git, overwrite the folder, pasted the function and it works! Hello World is there! (I didn't go until the printf, to make this to work I need to figure how).
Thanks for your support, we can close this one now.

Zaltora, 2018-07-01:
Great then :)

More details: It is about the memory declaration order.
```c
    uint8_t buffer[datalen +1];
    uint8_t *ptr = data_out;
```
If you go outside of your buffer with this assignation: 
` buffer[datalen +1] = 0`
It is like  write:
` ptr = 0 `
Because the declaration of `ptr` is below  your array.
This is a forbidden zone but with the declaration order, we can deduce which variable will be overwritten.

jano2358, 2018-07-01:
I see that, but it's seems a bit unrelated.
Just now I was testing it a bit, and comparing with the older code. The initial error seems to be over here:
In my first function I wrote: 
```
uint8_t buffer[datalen +1];
uint8_t *ptr = data_out;
buffer[0] = reg;

```
and it should be (the one you send me):
```
uint8_t buffer[datalen +1];
uint8_t *ptr = data_out;
buffer[0] = *reg;    //<if i delete the * here it not longer works
```
in that declaration, the content of the first position of the array is being equal to the address of the function parameter, right?
because in the for loop:
```

    for (uint16_t i = 1; i < (datalen+1) ; i++)
    {
        buffer[i] = *ptr;
        ptr++;
    }
```

behaves the same as (I just tested that):

`for (int i = 1; i <= datalen +1; i++) buffer[i] = *(ptr++);`

Zaltora, 2018-07-01:
Ok then change `buffer[0] = reg;` to  `buffer[0] = *reg; ` fix  one problem.
The `for` loop dont get the same effect in both case, it is the second problem:
First `for` loop go from `1` to `datalen` : `for (uint16_t i = 1; i < (datalen+1) ; i++)`
Second `for` loop go from `1` to `datalen+1` : `for (uint16_t i = 1; i <= (datalen+1) ; i++)`
The second `for` loop write outside of the buffer and erase `uint8_t *ptr`

Zaltora, 2018-07-01:
yeah, you a maybe right but the compiler do optimisation too. So we don't know really what happens in what order.

jano2358, 2018-07-01:
yes, that belongs to the compiler mystery :)
I'm closing this now

closed 2018-07-01 by jano2358