lvgl/lv_drivers #26: lvgl with bit sized vdb

Zaltora, 2018-10-07:
With the new lvgl 5.2, i see some special function/configuration  for monochrome display. with discussion here, i see more function : littlevgl/lvgl#300
```c
#define LV_VDB_PX_BPP
...
 void (*vdb_wr)(uint8_t * buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color, lv_opa_t opa);
```
The `vdb_rd` function doesn't exist yet ?
What is the  `rounder` function described in the commit ?
If i set 1bpp color for vdb, the buffer size will be lower than use 1 byte by pixel ?

It is a WIP feature ?

kisvegabor, 2018-10-08:
1. There is no `vdb_rd` because the background pixel will be in the buffer. So you can easily read it from the buffer to handle opacity. Look the `lv_vdb_wr_gray` function in the mentioned comment

2. The rounder function is used to extend the invalidated areas according to your display's requirements. For example, if you can write to your display 8 bits at once (1 byte for a monochrome display) and the bytes are mapped vertically then you should be sure that refreshed areas are `N*8` px tall. This way you can simply flush the VDB to your display. And a `8*LV_HOR_RES` VDB would be enough without any other frame buffers in the driver.  

3. If you set `LV_VDB_PX_BPP` to 1 and have a `LV_VDB_SIZE (128*64)` then the VDB will be `128*64 bit` -> `1 kbyte`

4. It's not WIP. it should already work properly.

Zaltora, 2018-10-08:
Here my new functions:
```c
void ssd1306_rounder(lv_area_t *a)
{
    /*Make the area bigger in y*/
    a->y1 = a->y1 & ~(0x7);
    a->y2 = a->y2 |  (0x7);
}

void ssd1306_vdb_wr(uint8_t * buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color, lv_opa_t opa)
{
    buf += buf_w * (y >> 3) + x;
    if(color.full)
    {
        (*buf) &= ~(1 << (y % 8));
    }
    else
    {
        (*buf) |= (1 << (y % 8));
    }
}

void ssd1306_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
    /*Return if the area is out the screen*/
    if (x2 < 0 || y2 < 0 || x1 > LV_HOR_RES - 1 || y1 > LV_VER_RES - 1)
    {
        lv_flush_ready();
        return;
    }

    /*Truncate the area to the screen*/
    int32_t act_x1 = x1 < 0 ? 0 : x1;
    int32_t act_y1 = y1 < 0 ? 0 : y1;
    int32_t act_x2 = x2 > LV_HOR_RES - 1 ? LV_HOR_RES - 1 : x2;
    int32_t act_y2 = y2 > LV_VER_RES - 1 ? LV_VER_RES - 1 : y2;

    _load_frame_buffer(_dev, (uint8_t*)color_p, x1, y1, x2, y2);
    lv_flush_ready();
}
```
my settings:
```c
#define LV_HOR_RES          (128)
#define LV_VER_RES          (64)
#define LV_VDB_SIZE         (LV_VER_RES * LV_HOR_RES / 2) 
#define LV_VDB_PX_BPP       1  
#define LV_COLOR_DEPTH     1  
```

Everything work fine !!  I can use VDB as main buffer now ! ( work with different VDB size too like 1/10 screen). Smaller RAM usage possible :).

- I will remove map and fill function, they are totally unnecessary with this driver now.
- I will remove my little trick function to avoid the use of VDB (ssd1306_update, ...).

Remarks:
- we need a special include to use `lv_refr_set_round_cb`, it is `#include "lvgl/lv_core/lv_refr.h"`. It not intuitive.
- I was thinking it is  WIP because in the comment of `lv_conf.h `we got this : (Set `disp_drv->vdb_wr` and `disp_drv->vdb_rd` too)

Question:
- The truncate operation in flush is necessary ? It is prevent a bad order setting  by user when he use the API ?

5.2 is a really nice update !!  good job.

kisvegabor, 2018-10-09:
> Everything work fine !! I can use VDB as main buffer now ! ( work with different VDB size too like 1/10 screen). Smaller RAM usage possible :).

So glad to hear! :) 
The minimum LV_VDB_SIZE is LV_HOR_RES * 8 to get at least one full row every time. 

> we need a special include to use lv_refr_set_round_cb, it is #include "lvgl/lv_core/lv_refr.h". It not intuitive.

I added `lv_core/lv_refr.h` to `lvgl.h` to make it more intuitive.

> I was thinking it is WIP because in the comment of lv_conf.hwe got this : (Set disp_drv->vdb_wr and disp_drv->vdb_rd too)

I fixed the comment, thanks!

> The truncate operation in flush is necessary ? It is prevent a bad order setting by user when he use the API ?

It is not necessary. It is used if you set larger LV_HOR_RES/LV_VER_RES then your display's resolution. In this case, you will get larger coordinates in the `disp_flush` function.

Zaltora, 2018-10-09:
So if i control at start of programm that LV_HOR_RES and LV_VER_RES are not greater that 128 and 64 respectively, i can remove the control code from flush function ?

kisvegabor, 2018-10-09:
Exactly!

closed 2018-10-09 by Zaltora