lvgl/lv_drivers #29: VDB update

Zaltora, 2018-10-17:
Since i update the library SSD1306, i got a lot of visual problem in my final application. the "hello world" example work but my app got more objects.

I need to clean some thing.  
- I need a custom (perfect) monochrome style setting
- I didn't found the way to set  a custom style by default like "font"

For my current Test, i use VDB screen sized (128*64). My first thought about the problem is the wrong style setting.
I got a second thought about the rounder function that will erase object already updated (not confirmed).
I will test more.

kisvegabor, 2018-10-17:
> I didn't found the way to set a custom style by default like "font"

You can modify the built-in styles (`lv_style_...`) before creating objects. 

I'm curiously waiting for the news. :)

Zaltora, 2018-10-17:
My style settings:
```c
    lv_style_copy(&style, &lv_style_transp);
    style.text.font = &lv_font_dejavu_10;   /*Unicode and symbol fonts already assigned by the library*/
    style.text.color.full = 1;
    style.text.opa = 255;

    style.body.main_color.full = 0;
    style.body.grad_color.full = 0;
    style.body.shadow.color.full = 0;
    style.body.border.color.full = 0;
    style.body.empty = 0;

    style.image.color.full = 1;
    style.image.intense = 255;
    style.image.opa = 255;

    style.line.color.full = 1;
    style.line.opa = 255;
    style.line.width = 1;
    style.line.rounded = false;
```
Well, when i set the rounder to all the screen i fix the problem:
```c
static void inline _rounder(lv_area_t *a)
{
    a->y1 = 0;
    a->y2 = LV_VER_RES-1;
    a->x1 = 0;
    a->x2 = LV_HOR_RES-1;
}
```
The problem is the following on a screen with 5 objects (all labels):
The first object at the top left is correctly drawn: "Selection"
All other objects are in their position except that part of the object has a piece of text "selection" at their bottom. The top of these objects are correctly traced.

Is not a style problem. I will check the function that send the buffer but I do not think the problem comes from there

kisvegabor, 2018-10-17:
Can you send a code which I can test in the PC simulator?

Zaltora, 2018-10-17:
I create something similar to my application. Very quickly made. the visual bug is different but same behavior.
[GEF_test.zip](https://github.com/littlevgl/lv_drivers/files/2488243/GEF_test.zip)

Zaltora, 2018-10-17:
I updated with pc_simulator but i got segmentation fault when i execute the program with my lv_conf.h setting.
The segmentation fault come too without my code.
[example_simulator.zip](https://github.com/littlevgl/lv_drivers/files/2488437/example_simulator.zip)

kisvegabor, 2018-10-18:
I have no segfault with the latest `master` branch of lvgl. There were some updates to solve segfaults in some cases. Hopefully, it will solve it for you too.

And `custom_init();` should be after `lv_init()`. Does it solve something?

This way I got this:
[image]

Zaltora, 2018-10-18:
I think  my SDL lib got the problem. Valgrind tell me that but i will test more.

Yes it is the resulte intended. What happen if you add this:

Zaltora, 2018-10-18:
I tested and it iw work. I will take a look to the send algorithm now.

Zaltora, 2018-10-18:
Found a way to generate the same problem.
New source :

Zaltora, 2018-10-18:
Here the final example : [exemple_simulator_3.zip](https://github.com/littlevgl/lv_drivers/files/2491206/exemple_simulator_3.zip)
I zoom screen. I remove mouse.
You got the same problem as i got with my screen. It is exactly the same screen.

kisvegabor, 2018-10-19:
Hi,

I've checked your code and found the mistake in `monitor flush`:
```c
void monitor_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
    /*Return if the area is out the screen*/
    if(x2 < 0 || y2 < 0 || x1 > MONITOR_HOR_RES - 1 || y1 > MONITOR_VER_RES - 1) {
        lv_flush_ready();
        return;
    }
    uint8_t i;
    uint16_t j;
    y1 = y1 >> 3;
    y2 = y2 >> 3;
    uint8_t xlen = x2-x1;
    uint8_t plen = y2-y1;
    lv_coord_t area_w = x2 - x1 + 1;     /*!!! color_p is not MONITOR_HOR_RES wide but (x2*x1+1)*/
    for (j = 0; j <= plen; j++)
    {
        for (i = 0; i <= xlen ; i++) //column
        {
            _send(i+x1, j+y1, (uint8_t*)&color_p[j* area_w +i]);
        }
    }

    sdl_refr_qry = true;

    /*IMPORTANT! It must be called to tell the system the flush is ready*/
    lv_flush_ready();
}
```

Zaltora, 2018-10-20:
Very nice ! how i missed that !! Of course, the size of VDB depend of the current size to refresh.
Thank for your help. I will fix that!

kisvegabor, 2018-10-20:
Did it completely solve the issue?

Zaltora, 2018-10-20:
Yes :)

closed 2018-10-20 by Zaltora

kisvegabor, 2018-10-20:
Cool :)