lvgl/lv_drivers #33: [new_api] vdb SSD1306 visual artifacts

BrianPugh, 2018-10-29:
I just updated to the latest commit of the `new_api` branch and am having visual artifacts. I looked through the new SSD1306 code, but I don't seem anything obviously wrong with it; however, I am not super familiar with the VDB system. Previously, everything worked fine with the old fill/flush method. The odd thing is it only happens sometimes; full screen refreshes seem fine, as well as *some* scrolling text.

Microcontroller: ESP32
Display: I2C SSD1306

Changing the VDB size seems to have no impact.

Attached gif demonstrating the issue:
https://gfycat.com/AgileDemandingHound

@Zaltora , can you offer any insight on this?

Zaltora, 2018-10-29:
I will take a look on it. Can you verify your setting ?
Here its mine :  [lv_conf.h.txt](https://github.com/littlevgl/lv_drivers/files/2526546/lv_conf.h.txt)
I will take a look on I2C transmission buffer.

Zaltora, 2018-10-29:
I found maybe the problem. I have done some change on I2C algorithm. Can you test it ?

BrianPugh, 2018-10-29:
Ya I'll test it in about 3 hours; thanks!

BrianPugh, 2018-10-30:
your PR #34 fixed it! Thank you so much!

closed 2018-10-30 by BrianPugh