lvgl/lv_drivers #43: Suggestions for setting input devices.

chaos4284, 2018-12-03:
In raspberry pie, the device name of the touch screen and the mouse changes each time depending on whether the mouse is used or not. When the mouse is connected / dev / input / event0 (mouse) and / dev / input / event1 (touch screen), If the mouse is not connected / dev / input / event1 (touch screen).
Therefore,  I want to put the function of setting the touch screen device in the way that we defined in the existing lv_drv_conf.h by passing the driver parameter to the function.

**~before
evdev_init()

~after**
evdev_init(char* device_name)

kisvegabor, 2018-12-03:
Hi,

It sounds like a reasonable improvement. However, I would rather prefer a new function because now all the `init` and `read` function have the same prototype. 

For example: `evdev_set_file(dev_name)`

kisvegabor, 2018-12-05:
See #44

closed 2018-12-05 by kisvegabor