lvgl/lv_drivers #49: Driver PAR, init_9341 and options

autoprog, 2019-01-10:
Hi, my test wiht ILI9341 parallel port.
Lvgl library compiling ok (Eclipse NXP).

how to init driver sequence
ex 1: 
       static ili9341_t  _xParDev;
       ili9341_t* _pxdev = &_xParDev;
       _pxdev->par_dev = (lv_par_handle_t)&_xParDev;
       _pxdev->protocol = ILI9341_PROTO_8080_8BIT;
....
...
 lv_label_set_text(label1, "Hello world!");

ex 2:
how to modifier bus ILI9341_PROTO_8080_8BIT to ILI9341_PROTO_8080_16BIT 
in driver : lv_par_write();

thanks
Carlos.

kisvegabor, 2019-01-11:
I suppose you are using the `new_api` branch. @Zaltora can you have a look at this question?

autoprog, 2019-01-11:
Hi, yes, using new_api.
questions 1: example init driver.
quaestion 2: driver_out ( modifier par 8080_8bits to 8080_16bits) in driver( lv_par_write ).

seee pseude code:
for (uint32_t i = 0; i < (len+1) ; i++)
	    {
	       GPIOB->PDOR = (uint8_t)  *ptr;   8080_8bits ?
               GPIOB->PDOR = (uint16_t)  *ptr; 8080_16bits ?
	        ptr++;
	        PAR_WR_LO
	        	//delay();
	     	PAR_WR_HI
	    }

thanks,
Carlos.

kisvegabor, 2019-01-11:
> questions 1: example init driver.

What is the question here?

autoprog, 2019-01-11:
hi,
example, how to init driver, steps by step.

ex 1:
static ili9341_t _xParDev;
ili9341_t* _pxdev = &_xParDev;
_pxdev->par_dev = (lv_par_handle_t)&_xParDev;
_pxdev->protocol = ILI9341_PROTO_8080_8BIT;
....
...
lv_label_set_text(label1, "Hello world!");
whie(1);

thanks,
Carlos

kisvegabor, 2019-01-11:
Hopefully @Zaltora could help in it.

Zaltora, 2019-01-12:
I do not program a lot right now but I can help.
I never test the parallel port, you can still found bug.
I will try answer the both question.
You need create your own parallel object option

First you need change and create your structure to handle the parallel port  with your custom option in `lv_drv_conf.h`:
```c
typedef const void* lv_par_handle_t;  //<== line to remove

typedef struct {
    uint8_t bus; // custom option to know bus option that can be used in  lv_par_write()
   ....
} lv_par_dev_t, *lv_par_handle_t;
```
Second, declare your port and screen object and init them:
```c
static ili9341_t _xScreenDev;
static lv_par_dev_t _xParDev;
ili9341_t* _pxdev = &_xScreenDev;

//parallel port initialization
_xParDev.bus = 1;
_xParDev.....
//fonction to init parallel port
....

//screen initialization
_pxdev->par_dev = (lv_par_handle_t)&_xParDev;
_pxdev->protocol = ILI9341_PROTO_8080_8BIT;
_pxdev->width = 320;
_pxdev->height = 240;
_pxdev->r = LV_ROT_DEGREE_90;
_pxdev->rst_pin = LV_DRIVER_NOPIN //If not used
_pxdev->bl_pin = LV_DRIVER_NOPIN //If not used

if (ili9341_init(&_xScreenDev, LV_ROT_DEGREE_90))
    return -1;
```
Then do lvgl init:
```c
    /*Gui initialization*/
    lv_init();

    //register driver into lvgl
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    /*Set up the functions to access to your display*/
    disp_drv.disp_flush = ili9341_flush; /*Used in buffered mode (LV_VDB_SIZE != 0  in lv_conf.h)*/
    disp_drv.disp_fill = NULL;//ex_disp_fill;   /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
    disp_drv.disp_map = NULL;//ex_disp_map;    /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
    lv_disp_drv_register(&disp_drv);
```

It is WIP but i already test this kind of init with SPI port with this screen and make them working fine :)
Hope it is help.

autoprog, 2019-01-17:
hello Zaltora,
I'm testing but still unsuccessful,
initializes driver pair but display lcd off. 
You do not have SPI code tested to help me, I may not be setting up properly.
see code:

 LV_MEM_CUSTOM   =   1  

typedef struct {
       uint8_t bus; // custom option to know bus option that can be used in  lv_par_write()
       lv_gpio_handle_t rst_pin;      //!< Reset pin
       lv_gpio_handle_t bl_pin;       //!< Backlight pin
       uint16_t width;                //!< Screen width, 320 or 240
       uint16_t height;               //!< Screen height, 240 or 320
} lv_par_dev_t, *lv_par_handle_t;

  // err_control(par_send(dev->spi_dev, 0, &cmd, 1, 1));
to
       err_control(par_send(dev->par_dev, 0, &cmd, 1, 1));

//------------------------------------------------------------------------------
int init_tft_9341(void){

	static ili9341_t _xScreenDev;
	static lv_par_dev_t _xParDev;

	ili9341_t* _pxdev = &_xScreenDev;

	
	//screen initialization
	_pxdev->par_dev = (lv_par_handle_t)&_xParDev;
	_pxdev->protocol = ILI9341_PROTO_8080_8BIT;
	_pxdev->width = 320;
	_pxdev->height = 240;
	_pxdev->r = LV_ROT_DEGREE_90;
	_pxdev->rst_pin = 1; //If not used
	_pxdev->bl_pin =  1; //If not used

	count_b=0;  count_buf =0;

	if (ili9341_init(&_xScreenDev, LV_ROT_DEGREE_90)){
	    return 255;
	}
//--------------------------------------------

	           //Gui initialization
	              lv_init();
	              //register driver into lvgl
	              lv_disp_drv_t disp_drv;
	              lv_disp_drv_init(&disp_drv);
	              //Set up the functions to access to your display
	              disp_drv.disp_flush = ili9341_flush; //Used in buffered mode (LV_VDB_SIZE != 0  in lv_conf.h)
	              disp_drv.disp_fill = NULL;//ex_disp_fill;   //Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)
	              disp_drv.disp_map = NULL;//ex_disp_map;    //Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)
	              lv_disp_drv_register(&disp_drv);

	return count_b;
//-------------main
init label +  hello_world("test");

//----------------------
thanks,
Carlos

autoprog, 2019-01-17:
hi Zaltora,
now my test with code Arduino on my board.

simple code:
//------------------------
 LCD_command_write(0xF7); // Pump ratio control
	  delayus_par( 10);
	  LCD_data_write(0x20); //
	  delayus_par( 20);
	  LCD_command_write(0x3A); // COLMOD: Pixel Format Set
	  delayus_par( 10);
	  LCD_data_write(0x55);
	  delayus_par( 20);
	  LCD_command_write(0x36); // Memory Access Control
	  delayus_par( 10);
	  LCD_data_write(0x08); 
	  delayus_par( 20);
	  LCD_command_write(0x11); // Sleep OUT
	  delayus_par( 10);
	  LCD_command_write(0x29); // Display ON

          LCD_rect(10,10,50,50, BLACK);
	  LCD_rect(70,10,150,50, RED); 
	  LCD_rect(180,80,50,50, BLUE);
	  LCD_rect(270,190,50,50, WHITE);
//-------------------------------------------------
results:

[image]

thanks,
Carlos.

Zaltora, 2019-01-20:
I didn't show you more details with SPI command because this part is hardware dependent.
Here an example with my SSD1306 and SPI configuration with esp-open-rtos

```c
//Objects declaration
static lv_spi_dev_t _xSpiDev =
{
    .bus = SPI_BUS,
    .cs = DISPLAY_CS,
    .dc = 0,
};
ssd1306_t __xScreenDev__;

//init peripherals system
    spi_init(SPI_BUS, SPI_MODE0, SPI_FREQ_DIV_8M, true, SPI_LITTLE_ENDIAN, false);
    portable_gpio_enable(DISPLAY_RST, PORT_GPIO_MODE_OUTPUT);

//screen configuration
    ssd1306_t* _pxdev = &__xScreenDev__;
    _pxdev->spi_dev = (lv_spi_handle_t)&_xSpiDev;
    _pxdev->protocol = SSD1306_PROTO_SPI3 ; //SSD1306_PROTO_SPI3;
    _pxdev->screen = SH1106_SCREEN; //SSD1306_SCREEN,  SH1106_SCREEN
    _pxdev->width    = DISPLAY_WIDTH;
    _pxdev->height   = DISPLAY_HEIGHT;
    _pxdev->rst_pin  = LV_DRIVER_NOPIN;//DISPLAY_RST;

    while (ssd1306_init(_pxdev) != 0)
    {
        vDebugLevel1("%s: failed to init SSD1306 lcd\n", __func__);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }

//Gui initialization
    lv_init();

    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    /*Set up the functions to access to your display*/
    disp_drv.disp_flush = ssd1306_flush; /*Used in buffered mode (LV_VDB_SIZE != 0  in lv_conf.h)*/
    disp_drv.disp_fill = NULL;//ex_disp_fill;   /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
    disp_drv.disp_map = NULL;//ex_disp_map;    /*Used in unbuffered mode (LV_VDB_SIZE == 0  in lv_conf.h)*/
    disp_drv.vdb_wr = ssd1306_vdb_wr;
    lv_disp_drv_register(&disp_drv);
```
lv_conf_drv configuration File:
```c
typedef struct {
    uint8_t bus;
    uint8_t cs;
    uint8_t dc;
} lv_spi_dev_t, *lv_spi_handle_t;

static inline void lv_spi_wr_cs(lv_spi_handle_t spi_dev, uint8_t lvl)
{
    gpio_write(spi_dev->cs, lvl);
}

static inline void lv_spi_wr_dc(lv_spi_handle_t spi_dev, uint8_t lvl)
{
    gpio_write(spi_dev->dc, lvl);
}

static inline int lv_spi_transaction(lv_spi_handle_t spi_dev, void* data_in, const void* data_out, uint16_t len, uint8_t word_size)
{
    spi_transfer(spi_dev->bus, data_out, data_in, len, word_size);
	return 0;
}
```

You make an error with your structure for parallel bus configuration, you don't need to add screen parameters, you need add parameters from your parallel library protocol. e.g:
```c
typedef struct {
uint8_t bus; // If your library support multiple BUS, otherwise, useless.
uint8_t cs;
uint8_t dc;
} lv_par_dev_t, *lv_par_handle_t;

static inline void lv_par_wr_cs(lv_par_handle_t par_dev, uint8_t lvl)
{
    digitalWrite(par_dev->cs, lvl); //Arduino API
    return;
}

static inline void lv_par_wr_dc(lv_par_handle_t par_dev, uint8_t lvl)
{
    digitalWrite(par_dev->dc, lvl); //Arduino API
    return;
}

static inline int lv_par_write(lv_par_handle_t par_dev, const void* data_out, uint16_t len, uint8_t word_size)
{
    //???????????
    return 0;
}
```

For the error you catch, I had review the code to do some corrections. (update soon)

autoprog, 2019-02-14:
Hi,
thanks,
Carlos.

closed 2019-02-14 by autoprog