lvgl/lv_drivers #50: ST7565 driver results in black background

jdinh1, 2019-01-20:
Hi, I'm currently porting lvgl to work with mbed-os using platformio. The screen I'm using is an ST7565 LCD 128x64 monochrome [(specs).](http://www.newhavendisplay.com/specs/NHD-C12864A1Z-FSRGB-FBW-HT1.pdf) I was able to get "Hello World!" to draw on the screen, however the entire screen background is solid black (the text can be seen if the screen is tilted, see attached images). I'm struggling to figure out what is wrong. I do have a feeling that my `lv_conf.h` is mis-configured, so I will keep digging on my end. If anyone have any ideas or suggestions, please do let me know.

[main.cpp](https://github.com/jdinh1/lvgl-mbed-test/blob/master/src/main.cpp)
[lv_conf.h](https://github.com/jdinh1/lvgl-mbed-test/blob/master/src/lv_conf.h)

[image]
[image]

embeddedt, 2019-01-20:
Try using the mono theme. You'll need to enable it in lv_conf.h and call
lv_theme_set_current.

jdinh1, 2019-01-21:
Thanks for the suggestion! Unfortunately, it did not help with the black bg. I even tried playing around with the hue value but no result. 

I did eventually got it to work properly. Turns out the command to set the lcd operational voltage in the `st7565_init()` function was incorrect. It should have been 

`st7565_command(CMD_SET_RESISTOR_RATIO | 0x4);` 
isntead of 
`st7565_command(CMD_SET_RESISTOR_RATIO | 0x6);`

I also noticed various other things in the st7565 driver that needs to be cleaned up and fixed as well. I will submit a PR for it soon.

[image]

kisvegabor, 2019-01-21:
@jdinh1 The setting of `CMD_SET_RESISTOR_RATIO` is specific to the display, right?

jdinh1, 2019-01-22:
@kisvegabor Yes, it is. The specsheet said to use `0x26`, which is what the `ST7565` driver in the repo is currently defaulted to. But it was strange that it did not work correctly. I sampled a few other implementations of the same driver, and most seems to use `0x24` or `0x27` so I ended up using `0x24` that and it seems to worked.

kisvegabor, 2019-01-22:
@jdinh1 I see, thank you! Can you send a PR with a comment for this line in the driver?

jdinh1, 2019-01-22:
@kisvegabor Done. Thank you for the awesome library!

closed 2019-01-22 by jdinh1