lvgl/lv_drivers #61: Incorrect path in libinput.h define

danjulio, 2019-07-18:
https://github.com/littlevgl/lv_drivers/blob/2423e655b0b6b7fc82d40889ca7c6ada1ec5528f/indev/libinput.h#L26

Should be #include "lvgl/src/lv_hal/lv_hal_indev.h"

embeddedt, 2019-07-18:
@danjulio You can send a PR with the change (that way you get credited for it).

danjulio, 2019-07-19:
@embeddedt sorry I'm not sure how to do this. I tried to create and publish a branch to do a pull request from but was not allowed to publish the branch.  My knowledge of github is pretty limited.

kisvegabor, 2019-07-19:
@danjulio 

Have look at this guide: https://blog.littlevgl.com/2018-12-06/contributing#how-can-i-send-fixes-and-improvements

Or as it's a very simple fix you can even edit the file on GitHub. However, I suggest learning Pull requests because it's a useful knowledge anyway. :)

embeddedt, 2019-08-06:
Closed via #62.

closed 2019-08-06 by embeddedt