lvgl/lv_drivers #66: Drivers questions

puzrin, 2019-07-29:
When working with PIO drivers, i could not understand concept of this repo. It looks like "assorted collection of different useful things", and may be tries to emulate features of missed package manager.

Of cause, i understand, we need to do something to make lvgl work, but there are no clean limits "where to stop" and everything is tight-coupled via template header.

I had to copy-paste files and edit hardcoded paths. That's strange and not good for maintenance.

## Goals

I'd like to understand concept and make some (or all parts) useable from PIO with ease.

## Problems

1. It's not clear, which drivers are about SDL.
2. Grouping by `display` / `indev` looks strange. And it's not clean why `win_drv` placed separate, in root.
3. Global header required, which defines everything, even if i need only SDL drivers with couple of defines.

## Suggestions

1. Global header should NOT be mandatory.

It worth to cleanup such patterns:

```
#ifndef USE_SSD1963
#  define USE_SSD1963         0
#endif
```

User should be able to pass #define-s only for used "drivers"

2. Improve isolation

https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.c#L20-L22 - it's very suspicious, when one driver goes for another one, out of it's scope (folder)

Probably, may worth to more flat layout with prefixes, OR may be move SDL drivers to separate repo.

If drivers are independent, probably those could be in separate folder each (something like mono repository, but not sure).

embeddedt, 2019-07-29:
> I had to copy-paste files and edit hardcoded paths. That's strange and not good for maintenance.

Why? You can just compile all of the files (only the enabled ones in `lv_drv_conf.h` are actually used).

> And it's not clear why `win_drv` placed separate, in root.

Because the nature of Windows required me to put the display and input routines together.

> https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.c#L20-L22 - it's very suspicious, when one driver goes for another one, out of it's scope (folder)

That's because the SDL core is in the `monitor` driver, but parts of it are also used by the SDL mouse, keyboard, and mousewheel drivers.

puzrin, 2019-07-29:
> Why? You can just compile all of the files (only the enabled ones in lv_drv_conf.h are actually used).

That's a bad practice to force user use huge header, when only couple of drivers required. Ideally, each should be in separate package. I understand reasons to put all in single repo. But that should not break isolation (make all dependent on single file with mandatory lines)

> Because the nature of Windows required me to put the display and input routines together.

> That's because the SDL core is in the `monitor` driver, but parts of it are also used by the SDL mouse, keyboard, and mousewheel drivers.

That means a simple thing - current folder structure does not satisfies technical requirements and should be improved somehow.

Usually folder means isolation level, not just "good intents".

embeddedt, 2019-07-29:
Our goal is not to have a perfect folder structure. Our goal is to make this as simple and easy to use as possible, especially for new users.

The current system, while probably not perfect, is also not difficult to add to a project. Copy the folder into your tree, and make an `lv_drv_conf.h` enabling the drivers you need.

kisvegabor, 2019-07-29:
> Our goal is not to have a perfect folder structure. Our goal is to make this as simple and easy to use as possible, especially for new users.

Exactly! Maybe its directory structure can be improved somehow but the basic idea of using `lv_drv_conf.h` should remain becaue its ease of use in most of the cases.

puzrin, 2019-07-29:
It may sound strange, but if you see first post, my goal is the same - make that easy to use :).

I see no problem to use big header if someone likes to copy-paste files. But that should not be mandatory. It looks like reinventing of primitive package system, and force everyone use it even when better alternatives available.

embeddedt, 2019-07-29:
> better alternatives available.

Better alternatives may be available, but are they simpler for the majority of users?

kisvegabor, 2019-07-29:
What we can do is to add `lv_drv_conf_checker.h` similarly to `lv_conf_checker.h` to allow using `_D....` compiler flags to set configs externally.
It this case an empty `lv_drv_conf.h` is still required.

puzrin, 2019-07-29:
> Better alternatives may be available, but are they simpler for the majority of users?

This question can not have constructive answer :). Anyone can continue use existing approach, if he wish. I suggest to not destroy it, but to remove limitations.

puzrin, 2019-07-29:
> What we can do is to add `lv_drv_conf_checker.h` similarly to `lv_conf_checker.h` to allow using `_D....` compiler flags to set configs externally.
> It this case an empty `lv_drv_conf.h` is still required.

Why not add something like "LV_DRV_NO_CONFIG" or "LV_DRV_USE_GLOBAL_CONF" (depends on prefered defaults) and include file only if condition matched?

kisvegabor, 2019-07-29:
> Why not add something like "LV_DRV_NO_CONFIG" or "LV_DRV_USE_GLOBAL_CONF" (depends on prefered defaults) and include file only if condition matched?

Sounds good! @embeddedt ?

embeddedt, 2019-07-29:
Fine with me.

puzrin, 2019-07-29:
https://github.com/littlevgl/lv_drivers/blob/master/indev/mouse.c#L10 but conditions like this should be updated. Driver should not be enabled on not-existing vars

kisvegabor, 2019-07-29:
> but conditions like this should be updated. Driver should not be enabled on not-existing vars.

It's not enabled. Undefined defines are evaluated as 0.

> May be names should be prefixed for more safety

They can get `lv_` prefix.

puzrin, 2019-07-29:
How about flatten folders structure and use prefixes instead? Example (very preliminary):

- `disp_r61581`
- `in_xpt2046`
- `win`
- `sdl2`

Reason - win & sdl drivers do not match existing structure.

kisvegabor, 2019-07-29:
I think the current folder structure is ok, only win and sdl are the exceptions.

> Also, probably, join SDL parts into single file - extensions (kbd/mouse) are small enougth.

I've also considered it but didn't implement in lack of time.

So I could imagine:
- `indev`
- `disp`
- `sdl`
- `win`

Or `sdl` and `win` in a folder called `both` (bad name just an example)

puzrin, 2019-07-29:
IMO that's artificial division, based on personal preferences/feelings instead of technical needs. I can suggest you very simple checkpoints:

1. Imagine every driver is in separate repo - should be possible to use (verify isolation and self-sufficience).
2. There should be simple (unified) way to joins multiple repos into one and split back. If that's possible (verify isolation is not broken on merge).

From this point of view, 2 possibilities available:

1. Keep all in `./src`, 2 files per driver (`.c` + `.h`).
2. Keep each in `./src/driver_name` subfolders.

IMO (1) should be enough. May be, worth add `lv_` prefix to avoid file names conflicts (not sure)

embeddedt, 2019-07-29:
> IMO that's artificial division, based on personal preferences/feelings instead of technical needs. 

@puzrin You continually refer to "technical needs" (even in other issues), however, we still haven't seen much objective evidence for changing the folder structure of this repository.

kisvegabor, 2019-07-29:
I also don't see why would it be better to use 10-20 repos with 2 files in each.

puzrin, 2019-07-29:
> @puzrin You continually refer to "technical needs" (even in other issues), however, we still haven't seen much objective evidence for changing the folder structure of this repository.

Currently there is no clean logic where to place each driver. Only "we would like displays & inputs, but if we fails - let's do special case". From my point of view - that's broken architecture.

> I also don't see why would it be better to use 10-20 repos with 2 files in each.

I did not suggested this. Because that can work only when package manager avilable. I suggested different thing - checkpoint, how to verify your files in single repo are ok. All you need - just imagine described actions and check if collisions exist or not.

kisvegabor, 2019-07-30:
> I did not suggested this. Because that can work only when package manager avilable. I suggested different thing - checkpoint, how to verify your files in single repo are ok. All you need - just imagine described actions and check if collisions exist or not.

In light of that, it would solve the issue if we handled SDL similarly to Win, and applied the same philosophy for new common drivers (both display and indev).

puzrin, 2019-07-30:
> In light of that, it would solve the issue if we handled SDL similarly to Win, and applied the same philosophy for new common drivers (both display and indev).

I think, driver type (display/indev/common) is a bit artificial thing, which does not add value. Each driver is independent unit. You may use or not use it :), and having category will not change anything. The only required thing is good name. For example, it's not trivial to guess "monitor" / "keyboard" / "mouse" are about SDL.

Well, since we need move forward, critical things are:

- Add guards to disable config #include
- Improve files location/names in any way you prefer
- May be (not sure), put all under `./src`

As soon as this done, i can start experimenting with PIO support & demo update, to use this repo directly.

embeddedt, 2019-07-30:
> For example, it's not trivial to guess "monitor" / "keyboard" / "mouse" are about SDL.

Then they can be renamed to `sdl_monitor`, `sdl_keyboard`, and `sdl_mouse`. That's trivial to do and shouldn't affect existing functionality.

puzrin, 2019-07-30:
I will accept any decision. All preferences are listed, but only `Add guards to disable config #include` is really blocking me. Everything else can be used.

kisvegabor, 2019-07-30:
I've added `LV_DRV_NO_CONF`. See 5c6535d

The directory structure should be modified when v6.1 is released to save avoid updating the projects twice.

puzrin, 2019-07-30:
There may be problem with tagging. It looks convenient to have the same naming with lvgl, but that makes difficult to release patches separate. That's a common problem, not specific for this project.

Do you really need so much to hard sync version numbers or semantic versioning could be enougth? It's still not late to change, while only one tag exists and not used wide.

One more note - tags could be also good without letter `v` at start :)

puzrin, 2019-07-30:
One more build problem:

https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.h#L26

This should be `#include "lvgl.h"` for PIO. In other words, foler name should not be hardcoded, or there should be possibility to change defaults. Need fix with alternatives.

kisvegabor, 2019-07-31:
> There may be problem with tagging. It looks convenient to have the same naming with lvgl, but that makes difficult to release patches separate. That's a common problem, not specific for this project.

Here and other repos the tags refer to the lvgl release which they were released. So these tags are used to get a compatible combination of repositories. 

> Do you really need so much to hard sync version numbers or semantic versioning could be enougth? It's still not late to change, while only one tag exists and not used wide.

Isn't vX.Y.Z the semantic versioning? Our modification is we us only vX.Y and, instead of releasing Z patches, we push bugfixes directly into master to make the fixes delivered faster.

> One more note - tags could be also good without letter v at start :)

I add `v` because theoretically there can be other tags than versions.

> This should be #include "lvgl.h" for PIO. In other words, folder name should not be hardcoded, or there should be possibility to change defaults. Need fix with alternatives.

There could be an other define like `LV_LVGL_H_INCLUDE_DIRECT`. It can be used in other repos too.

puzrin, 2019-07-31:
> Isn't vX.Y.Z the semantic versioning?

1. I know many projects, using X.X.X or Y.Y. Both are ok.
2. I know some projects, changed `vX.X.X` => `X.X.X` after some time, for simplicity.
3. I don't know any project changed `X.X.X` => `vX.X.X` :)

> Our modification is we us only vX.Y and, instead of releasing Z patches, we push bugfixes directly into master to make the fixes delivered faster.

Master updates are not compatible with semversioning rules. Those rules uses tags (exact commits), and definition how to match best preferable tag name AND keep situation reproduceable. So you can describe exact combination of packages when install from scratch.

But master is "floating" and if you pin dependency to master, you will get different result after each update. And, for example, project can be broken afte couple of years, when you release 7.+.

> There could be an other define like `LV_LVGL_H_INCLUDE_DIRECT`. It can be used in other repos too.

Could you add anything you prefer? Seems this is last change needed to build sources. Sorry missed that in previous request.

kisvegabor, 2019-07-31:
> But master is "floating" and if you pin dependency to master (or any other live branch), you will get different result after each update. And, for example, project can be broken afte couple of years, when you release 7.+.

As usually lvgl is added as a submodule which points to specific commit it should be fine. In case of PIO it really could cause issues. I'll think about it.

> Could you add anything you prefer? Seems this is last change needed to build sources. Sorry missed that in previous request.

I don't understand :smile:  How do you mean "Could you add anything you prefer?"

puzrin, 2019-07-31:
> As usually lvgl is added as a submodule which points to specific commit it should be fine. In case of PIO it really could cause issues. I'll think about it.

Note, principal difference is not PIO but having package manager. It has 2 goals:

1. Simplify installs
2. Simplify updates (maintenance)

You still can hard-pin to commits in PIO, but that will restrict useful features.

> I don't understand smile How do you mean "Could you add anything you prefer?"

I mean, i'd like to get update asap to move forward, and don't care about details if it works :). If you think `LV_LVGL_H_INCLUDE_DIRECT` is good - i'm ok with it.

embeddedt, 2019-07-31:
> You still can hard-pin to commits in PIO, but that will restrict useful features.

The PC simulator gets pinned to a commit and no one has had an issue thus far (we don't add new features to `master`, only bugfixes). So I think we can safely do it in PIO as well.

puzrin, 2019-07-31:
@embeddedt 

1. big changes appears in master after each major release.
2. if you pin to direct commit, you will loose simplicity of minor updates.

embeddedt, 2019-07-31:
> big changes appears in master after each major release.

This is irrelevant if the commit is pinned.

> if you pin to direct commit, you will lose simplicity of minor updates.

As far as I know that hasn't been an issue thus far.

puzrin, 2019-07-31:
@embeddedt 

Your statements are absolutely correct for your situation - you don't use package managers (just because those almost not exists for C, that's not your fault).

But IMHO it really worth improve compatibility with "modern" approaches. That does not require to reinvent something new, only copy-paste well-tested things.

kisvegabor, 2019-07-31:
@embeddedt the problem is that PIO (and other package managers) will get the latest master if no tag or commit specified. So they can get even a new major version automatically which will cause issues.

A simple solution could be to still send bugfixes into master but make a Patch release every month.

embeddedt, 2019-07-31:
> patch release every month.

I suppose that could work. It wouldn't affect the people who pull from `master` anyways but PIO would still get updates at a reasonable rate.

puzrin, 2019-07-31:
> A simple solution could be to still send bugfixes into master but make a Patch release every month.

There are several possibilities:

1. Increase releases frequencies
2. Use 3 digits in version, and unsync last one
   Examples: `6.0.1`, `6.0.2`, ...
3. Unsync versions at all and leave notes in readme/changelog
   Examples: `from 1.x for lvgl 6.x, from 3.0 for lvgl 7.0`

From my point of view, less dependencies for separate repos are usually more simple, or it worth consider mono repo. That's not specific for lvgl. And of cause frequent releases are always great.

Since i'm not C dev and not lvgl dev, i can not say what will be better for you.

kisvegabor, 2019-08-01:
I've added `LV_LVGL_H_INCLUDE_SIMPLE` in  a124106

> Use 3 digits in version, and unsync last one (and make independent releases)
Examples: 6.0.1, 6.0.2, ...

I like this option the most

lvgl master receives updates anyway so there the periodic releases will work. In other repost we can make patch releases on demand.

@embeddedt 
It's 1st of Agust so we can release lvgl v6.0.1, do you agree?

embeddedt, 2019-08-01:
Sounds good.

puzrin, 2019-08-02:
Since topic may be "endless" and all fatal blocking issues were solved, i tend to close it. If there are other important questions, those can be moved to separate, more concrete issues.

kisvegabor, 2019-08-02:
@puzrin 
Ok, let's close it. I'll apply the same changes in the other example/app repos too later.

closed 2019-08-02 by kisvegabor

puzrin, 2019-08-02:
A first glance, examples and apps are not self-sufficient (no main function in each).

Prior to add configs, need decide how do you expect those to be used. Personally, when you design each lib/app, i strongly recommend check it first for compatibility with PIO (read as "any package manager capable system"). It has more high requirements for architecture. If your repo works with PIO, it will be very easy to make it work with other approaches. The opposite is not true.

embeddedt, 2019-08-02:
@puzrin They're not expected to be self-sufficient, but rather to be called from your existing `main` function.

puzrin, 2019-08-02:
I know. That means they conflict with concept of "1-click install and use". Solution could be to have guarded `main`, hidden by default. OR may be some are considered as library units, then worth move those to separate repo, provide demo sample etc.

Problem is to understand full list of requirements for final "product", prior to try add pio configs. Current PIO demo is almost polished, anyone can clone it's `platform.ini` and modify as he wish.

kisvegabor, 2019-08-02:
To use for example a settings example, you need to get the [lv_apps](https://github.com/littlevgl/lv_apps) repo and call `lv_ex_settings_2()`;

It's not one click but they are examples, so IMO it's usable enough. 
It's quite easy to try a lot of examples this way. E.g. just call `lv_ex_slider_1()` or `lv_es_slider_2()` from the [lv_examples](https://github.com/littlevgl/lv_apps/tree/rework) and you can quickly try al lteh examples by installing only one repo and just calling different functions.

embeddedt, 2019-08-02:
I think we need to separate the PlatformIO architecture and the average project that LittlevGL uses. I don't have any statistics, but based on the questions we get, very few people use PlatformIO at present. So I think that, as @kisvegabor said, the current method is fine.

puzrin, 2019-08-02:
1. I don't like idea to use extra files, out of app folder. If you don't target app as library, there are no reasons for foreign examples, because app is example itself. Mono-repo approach means you keep multiple things in single repo, but each one stays isolated and self-sufficient (keep all in it's folder).
2. IMO easy use could be "user clones repo, then open each folder as standalone project and it works".

embeddedt, 2019-08-02:
> IMO easy use could be "user clones repo, then open each folder as standalone project and it works".

That's too many individual IDE projects to maintain and update when something changes. And like @kisvegabor said, it makes it easy to try examples.

puzrin, 2019-08-02:
> I think we need to separate the PlatformIO architecture and the average project that LittlevGL uses. I don't have any statistics, but based on the questions we get, very few people use PlatformIO at present.

There is nothing "special" with PIO. IMO current layout just breaks design approaches (or mix completely different things), but this was not critical before. Now such mixture is visible. You can fix it or not use PIO.

Also note, statement "very few people use PlatformIO at present" is not precise. More correct is "very few people use PlatformIO *with LlittlevGL* at present". And reasons are completely different - they don't use because library was not useable with PIO :). And not because code is bad, but because it was organized such way.

embeddedt, 2019-08-02:
> And reasons are completely different - they don't use because library was not useable with PIO :). And not because code is bad, but because it was organized such way.

What is the evidence for that?

> IMO current layout just breaks design approaches (or mix completely different things), but this was not critical before.

LittlevGL is not supposed to have a perfect design. It's supposed to be easy to use for the majority of people.

I see no reason why someone can't add a simple function call into `main` in PlatformIO.

puzrin, 2019-08-02:
> > And reasons are completely different - they don't use because library was not useable with PIO :).
> 
> What is the evidence for that?
>

Hm... do you need evidence that pio is widely used? See their forum actibity, registry size and so on. If your question is about lvgl only - IMO my fixes across lvgl/lv_drivers/demo show the difference.

> LittlevGL is not supposed to have a perfect design. It's supposed to be easy to use for the majority of people.
> 
> I see no reason why someone can't add a simple function call into `main` in PlatformIO.

I respect your opinion. But IMO you speak as qualified developper, who know very well all details. Other people may be not as you. "Trivial thing" for you may mean "useless waste of time" for someone else.

embeddedt, 2019-08-02:
> If your question is about lvgl only

My question is about whether people are using PlatformIO+LittlevGL.

> IMO my fixes across lvgl/lv_drivers/demo show the difference.

No, I would say your fixes make it easier to use, but they don't automatically increase usage. It would be better if you had some objective proof that people are using PlatformIO+LittlevGL and having issues due to bad organization, instead of just stating that.

puzrin, 2019-08-02:
> My question is about whether people are using PlatformIO+LittlevGL.

I don't know such people. But i don't know lot of people you use lvgl anyhow too :).

> No, I would say your fixes make it easier to use, but they don't automatically increase usage. It would be better if you had some objective proof that people are using PlatformIO+LittlevGL and having issues due to bad organization, instead of just stating that.

I can repeat again - it's not correct to demand evidences of `PlatformIO+LittlevGL` use. From my point of view, that was unuseable at all in the past. Your statement about "your fixes make it easier to use, but they don't automatically increase usage" also have no proofs :).

Also note, all of us should have motivation to do something. When i did PIO-related things, lv_font_conv, lv_i18n - that's not just to improve world, but because i need that for my projects. But i have no motivation to improve lv_apps and lv_samples. So, i can advice improvments, once, but will not push those.

kisvegabor, 2019-08-04:
@puzrin 
As @embeddedt described, PlatformIO is only one way of using LittlevGL. We can't improve one way while making less usable the rest.

IMO the current way (call the create function of examples/apps) is a good compromise in the use cases we've seen so far. Not one-click but two-clicks in PIO. :slightly_smiling_face:

Let's leave this part as it is. `lv_examples` and `lv_apps` should be updated in the same way as we improved `lv_drivers`, however, they are being reworked anyway.

puzrin, 2019-08-04:
> We can't improve one way while making less usable the rest.

I did not suggested to make the rest less useful. At least, not intended to to so. And i don't know how making each app more isolated/independent can decrease usability.

embeddedt, 2019-08-04:
> And i don't know how making each app more isolated/independent can decrease usability.

It's because it makes it harder to try many of them. The current system allows you to try them by just changing a function call in `main`.

puzrin, 2019-08-04:
IMO you mix completely different things. For apps isolation and consistency do not mean inability to run from outer code. Those mean no hard dependencies on outer code, and ability to run standalone.

For example, each app can have guarded `main()`, toggleable by build options (depending on usage type). I can understand any use-cases, but don't understand ignoring all other approaches and design principles.

kisvegabor, 2019-08-05:
> Those mean no hard dependencies on outer code, and ability to run standalone.

Would it mean to include `lvgl` to the example repos too?

---

I can't see `lv_drivers` in PIO registry:
https://platformio.org/lib/search?query=littlevgl

Do you knoe something about it?

puzrin, 2019-08-05:
> Would it mean to include lvgl to the example repos too?

Personally, i don't need those. No ideas about other people. IMO it could be useful add readme into each folder with very short description. If app can run, it's an example itself.

> I can't see `lv_drivers` in PIO registry:

I'm still waiting too. Probably they have moderation pause on weekend. See https://platformio.org/lib "Resently Added". If those refresh today and drivers not appears, i will ask on forum.

kisvegabor, 2019-08-06:
> That does not require lvgl sources, only pio config in app folder (if main() exists in some form).

Adding pio config to the app repo is not a problem once we've established this practice. 

How would you handle the driver and system initialization if `main` was in the app repo? I.e. how could you call the driver inits without modifying the app repo's main?

[17 more replies]