lvgl/lv_drivers #75: Build failure due to incorrect header path

thelunatic, 2019-08-29:
Hi,
I'm currently working on a project where I'm using lvgl and lv_drivers as git submodules.
The issue is that the following include:
https://github.com/littlevgl/lv_drivers/blob/117812eec9a5b82fe879f53ee0d1bf39cb9027c8/display/fbdev.h#L29

Breaks the build as the header lvgl.h is in "../../lvgl/lvgl.h" and not "lvgl/lvgl.h" as lv_drivers and lvgl are two separate directories.

Is this intentional or can this be updated to match  "../../lv_drv_conf.h" include above it?

Best regards,
Vijay

embeddedt, 2019-08-29:
You should add whatever directory has `lvgl` in it to your include path.

```
- root
| - lvgl
| - lv_drivers
```

`root` needs to be in your include path.

thelunatic, 2019-08-29:
OK. Thank you for the quick reply!
I'm closing the issue since it's solved now.

closed 2019-08-29 by thelunatic