lvgl/lv_drivers #76: FBDEV driver always tries to use the FBSD code

ArmyOfMonkies, 2019-09-04:
HI!

If i try to use the linux framebuffer code it alwas ends with a compile fail. The failure occurs inside codeblocks that belongs to fbsd. This occurs irrespecitively of how i configre littlvgl in lv_drv_conf.h

in lv_drv_conf.h the use of the freebsd framebuffer code is defined like this

```
#ifndev USE_BSD_FBDEV
#   deine USE_BSD_FBDEV   0
#endif
```

Its the same pattern used in all config files, and that means that USE_BSD_FBDEV is always defined and is either 0 och 1.

But in fbdev.c the code checks if USE_BSD_FBDEV is defined or not, and it will always be defined, so the bsd code will always be used.

```
#ifdef USE_BSD_FBDEV
#include <sys/fcntl.h>
#include <sys/time.h>
#include <sys/consio.h>
#include <sys/fbio.h>
#else  /* USE_BSD_FBDEV */
#include <linux/fb.h>
#endif /* USE_BSD_FBDEV */
```
The code works fine if its altered to check if USE_BSD_FBDEV is 0 or 1 instead of if it is defined att all. Eg just replace #ifdef with #if in the few locations where the code checks for USE_BSD_FBDEV

embeddedt, 2019-09-04:
Please send a pull request.

ArmyOfMonkies, 2019-09-04:
I think i have managed to fix a pull request, but this is the first time i use git so i might have done something wrong.

embeddedt, 2019-09-04:
Merged. https://github.com/littlevgl/lv_drivers/commit/82f48926dac9333b2b2fa2c5a301a43a619b6f8c

closed 2019-09-04 by embeddedt