lvgl/lv_drivers #86: X axis not inverting

criscola, 2019-11-21:
Hi, I'm trying to use this library to interface with an Adafruit TFT display hooked to a raspberry pi 3. 
I enabled the marker to work with the calibration and I swapped the X and Y axis through ``EVDEV_SWAP_AXES`` in ``lv_drv_conf.h``. I enabled ``EVDEV_SCALE`` and ``EVDEV_CALIBRATE`` as well. By reading the comment just to the right of ``EVDEV_HOR_MIN`` I swapped the former with its counterpart ``EVDEV_HOR_MAX`` but the axis is not inverting. Could this be a bug?

criscola, 2019-11-21:
Yep... I changed the code in ```evdev.c``` from:
```
#if EVDEV_SCALE
    data->point.x = map(evdev_root_x, 0, EVDEV_SCALE_HOR_RES, 0, LV_HOR_RES);
    data->point.y = map(evdev_root_y, 0, EVDEV_SCALE_VER_RES, 0, LV_VER_RES);
#else
#if EVDEV_CALIBRATE
	data->point.x = map(evdev_root_x, EVDEV_HOR_MIN, EVDEV_HOR_MAX, 0, LV_HOR_RES);
	data->point.y = map(evdev_root_y, EVDEV_VER_MIN, EVDEV_VER_MAX, 0, LV_VER_RES);
#else
    data->point.x = evdev_root_x;
    data->point.y = evdev_root_y;
#endif
#endif
```
to: 
```
#if EVDEV_SCALE
    data->point.x = map(evdev_root_x, 0, EVDEV_SCALE_HOR_RES, 0, LV_HOR_RES);
    data->point.y = map(evdev_root_y, 0, EVDEV_SCALE_VER_RES, 0, LV_VER_RES);
#endif
#if EVDEV_CALIBRATE
	data->point.x = map(evdev_root_x, EVDEV_HOR_MIN, EVDEV_HOR_MAX, 0, LV_HOR_RES);
	data->point.y = map(evdev_root_y, EVDEV_VER_MIN, EVDEV_VER_MAX, 0, LV_VER_RES);
#else
    data->point.x = evdev_root_x;
    data->point.y = evdev_root_y;
#endif
```
Now it inverts the direction correctly. Can I submit a pull request?

embeddedt, 2019-11-21:
> Can I submit a pull request?

Please do!

embeddedt, 2019-11-21:
Let's continue the discussion in #87.

closed 2019-11-21 by embeddedt