lvgl/lv_drivers #88: fbdev.c: Re-use of previously confined area can result in out-of-range memcpy

M-Bab, 2019-12-11:
The function fbdev_flush nicely confines the set area if it exceeds the screen size.

But in the line https://github.com/littlevgl/lv_drivers/blob/master/display/fbdev.c#L175 the original unconfined area value is used for width calculation which can result in memory corruption.

I don't have a monochrome display but line https://github.com/littlevgl/lv_drivers/blob/master/display/fbdev.c#L225 may also be affected by this problem.

kisvegabor, 2019-12-11:
Hi,

So do you suggest using area_x2 - area_x1 + 1` in the mentioned cases?

M-Bab, 2019-12-18:
No rather the truncated values act_x/y1/2. Certainly not the values that where previously found to be potentially out-of-range.

kisvegabor, 2019-12-19:
Could you send a Pull request with the proposed changes?

PR #89 (2020-01-12): Use truncated width in fbdev.c
May fix #88.
closed 2020-01-12 by kisvegabor