lvgl/lv_drivers #90: libinput driver not linking

j-clay, 2020-01-15:
Hello!

I'm setting up my project to use libinput for the touchscreen. When I enable this through the config file and make, I get several linking errors when calling libinput functions. I haven't modified the files within lv_drivers folder.

It seems the libinput library isn't being compiled into the lv_indev.o file correctly. Any ideas?

These are the errors below:

arm-poky-linux-gnueabi-gcc  -march=armv7-a -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 --sysroot=/opt/fsl-imx-x11/4.14-sumo/sysroots/cortexa9hf-neon-poky-linux-gnueabi -o demo ./main.o  lv_group.o lv_indev.o lv_disp.o lv_obj.o lv_refr.o lv_style.o lv_debug.o lv_hal_disp.o lv_hal_indev.o lv_hal_tick.o lv_arc.o lv_bar.o lv_cb.o lv_cpicker.o lv_ddlist.o lv_kb.o lv_line.o lv_mbox.o lv_preload.o lv_roller.o lv_table.o lv_tabview.o lv_tileview.o lv_btn.o lv_calendar.o lv_chart.o lv_canvas.o lv_gauge.o lv_label.o lv_list.o lv_slider.o lv_ta.o lv_spinbox.o lv_btnm.o lv_cont.o lv_img.o lv_imgbtn.o lv_led.o lv_lmeter.o lv_page.o lv_sw.o lv_win.o lv_font.o lv_font_fmt_txt.o lv_font_roboto_12.o lv_font_roboto_16.o lv_font_roboto_22.o lv_font_roboto_28.o lv_font_unscii_8.o lv_circ.o lv_area.o lv_task.o lv_fs.o lv_anim.o lv_mem.o lv_ll.o lv_color.o lv_txt.o lv_math.o lv_log.o lv_gc.o lv_utils.o lv_async.o lv_printf.o lv_bidi.o lv_theme_alien.o lv_theme.o lv_theme_default.o lv_theme_night.o lv_theme_templ.o lv_theme_zen.o lv_theme_material.o lv_theme_nemo.o lv_theme_mono.o lv_draw_basic.o lv_draw.o lv_draw_rect.o lv_draw_label.o lv_draw_line.o lv_draw_img.o lv_draw_arc.o lv_draw_triangle.o lv_img_decoder.o lv_img_cache.o fbdev.o monitor.o R61581.o SSD1963.o ST7565.o UC1610.o SHARP_MIP.o FT5406EE8.o keyboard.o mouse.o mousewheel.o evdev.o libinput.o XPT2046.o win_drv.o -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed
libinput.o: In function `libinput_set_file':
lv_drivers/indev/libinput.c:72: undefined reference to `libinput_device_unref'
lv_drivers/indev/libinput.c:73: undefined reference to `libinput_path_remove_device'
lv_drivers/indev/libinput.c:76: undefined reference to `libinput_path_add_device'
lv_drivers/indev/libinput.c:81: undefined reference to `libinput_device_ref'
libinput.o: In function `libinput_init':
lv_drivers/indev/libinput.c:98: undefined reference to `libinput_path_create_context'
lv_drivers/indev/libinput.c:103: undefined reference to `libinput_get_fd'
libinput.o: In function `libinput_read':
lv_drivers/indev/libinput.c:132: undefined reference to `libinput_dispatch'
lv_drivers/indev/libinput.c:134: undefined reference to `libinput_event_get_type'
lv_drivers/indev/libinput.c:149: undefined reference to `libinput_event_destroy'
lv_drivers/indev/libinput.c:133: undefined reference to `libinput_get_event'
lv_drivers/indev/libinput.c:138: undefined reference to `libinput_event_get_touch_event'
lv_drivers/indev/libinput.c:139: undefined reference to `libinput_event_touch_get_x_transformed'
lv_drivers/indev/libinput.c:140: undefined reference to `libinput_event_touch_get_y_transformed'
collect2: error: ld returned 1 exit status
Makefile:37: recipe for target 'default' failed
make: *** [default] Error 1

embeddedt, 2020-01-15:
I don't see you linking `libinput` anywhere within that command. Maybe you need `-linput`, or something similar?

j-clay, 2020-01-15:
This is my makefile. I am using a toolchain for cross compiling so ${CC} is  arm-poky-linux-gnueabi-gcc -march=armv7-a -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 --sysroot=/opt/fsl-imx-x11/4.14-sumo/sysroots/cortexa9hf-neon-poky-linux-gnueab

However with this makefile has the linker flags and still not showing on the output. Maybe the toolchain isn't including it for some reason? 

#CC ?= gcc
LVGL_DIR ?= ${shell pwd}
CFLAGS ?= -O3 -g0 -I$(LVGL_DIR)/ -Wall -Wshadow -Wundef -Wmaybe-uninitialized -Wmissing-prototypes -Wno-discarded-qualifiers -Wall -Wextra -Wno-unused-function -Wundef -Wno-error=strict-prototypes -Wpointer-arith -fno-strict-aliasing -Wno-error=cpp -Wuninitialized -Wmaybe-uninitialized -Wno-unused-parameter -Wno-missing-field-initializers -Wtype-limits -Wsizeof-pointer-memaccess -Wno-format-nonliteral -Wno-cast-qual -Wunreachable-code -Wno-switch-default -Wno-switch-enum -Wreturn-type -Wmultichar -Wformat-security -Wno-ignored-qualifiers -Wno-error=pedantic -Wno-sign-compare -Wno-error=missing-prototypes -Wdouble-promotion -Wclobbered -Wdeprecated -Wempty-body -Wtype-limits -Wshift-negative-value -Wstack-usage=1024 -Wno-unused-value -Wno-unused-parameter -Wno-missing-field-initializers -Wuninitialized -Wmaybe-uninitialized -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wtype-limits -Wsizeof-pointer-memaccess -Wno-format-nonliteral -Wpointer-arith -Wno-cast-qual -Wmissing-prototypes -Wunreachable-code -Wno-switch-default -Wswitch-enum -Wreturn-type -Wmultichar -Wno-discarded-qualifiers -Wformat-security -Wno-ignored-qualifiers -Wno-sign-compare
LDFLAGS ?= -lSDL2 -lm -linput
BIN = demo

#Collect the files to compile
MAINSRC = ./main.c

include $(LVGL_DIR)/lvgl/lvgl.mk
include $(LVGL_DIR)/lv_drivers/lv_drivers.mk
#include $(LVGL_DIR)/lv_examples/lv_examples.mk

OBJEXT ?= .o

AOBJS = $(ASRCS:.S=$(OBJEXT))
COBJS = $(CSRCS:.c=$(OBJEXT))

MAINOBJ = $(MAINSRC:.c=$(OBJEXT))

SRCS = $(ASRCS) $(MAINSRC)
OBJS = $(AOBJS) $(COBJS)

## MAINOBJ -> OBJFILES

all: default

%.o: %.c
        @$(CC)  $(CFLAGS) $(LDFLAGS) -c $< -o $@
        @echo "CC $<"

default:$(COBJS) $(MAINOBJ)
        $(CC) $(LDFLAGS) -o $(BIN) $(MAINOBJ) $(COBJS)

clean: 
        rm -f $(BIN) $(AOBJS) $(COBJS) $(MAINOBJ)

The output is the same as original comment.

embeddedt, 2020-01-15:
Try moving `LDFLAGS` after everything else. Usually the objects have to specified first and then the libraries.

j-clay, 2020-01-15:
Same issue :(

j-clay, 2020-01-15:
I was able to compile it without it sourced to the cross compiler. So something about the cross compiler isn't including the flags.

j-clay, 2020-01-15:
I was able to force the flags in this way. This is by no means the proper way, so if you have any ideas on how to fix it the correct way that would be great!

Since I'm sourcing the cross compiler, it is already using the env flag ${CC}, so it was being overidden within the makefile. I tried removing the -libinput from the CPC variable since it is in the LDFLAGS, however, it still doesn't seem the LDFLAGS are being added since the problem still persisted. 

```
CPC ?= ${CC} -linput
LVGL_DIR ?= ${shell pwd}
CFLAGS ?= -O3 -g0 -I$(LVGL_DIR)/ -Wall# -Wshadow -Wundef -Wmaybe-uninitialized -Wmissing-prototypes -Wno-discarded-qualifiers -Wall -Wextra -Wno-unused-function -Wundef -Wno-error=strict-prototypes -Wpointer-arith -fno-strict-aliasing -Wno-error=cpp -Wuninitialized -Wmaybe-uninitialized -Wno-unused-parameter -Wno-missing-field-initializers -Wtype-limits -Wsizeof-pointer-memaccess -Wno-format-nonliteral -Wno-cast-qual -Wunreachable-code -Wno-switch-default -Wno-switch-enum -Wreturn-type -Wmultichar -Wformat-security -Wno-ignored-qualifiers -Wno-error=pedantic -Wno-sign-compare -Wno-error=missing-prototypes -Wdouble-promotion -Wclobbered -Wdeprecated -Wempty-body -Wtype-limits -Wshift-negative-value -Wstack-usage=1024 -Wno-unused-value -Wno-unused-parameter -Wno-missing-field-initializers -Wuninitialized -Wmaybe-uninitialized -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wtype-limits -Wsizeof-pointer-memaccess -Wno-format-nonliteral -Wpointer-arith -Wno-cast-qual -Wmissing-prototypes -Wunreachable-code -Wno-switch-default -Wswitch-enum -Wreturn-type -Wmultichar -Wno-discarded-qualifiers -Wformat-security -Wno-ignored-qualifiers -Wno-sign-compare 
LDFLAGS ?= -lSDL2 -lm -linput
BIN = demo

#Collect the files to compile
MAINSRC = ./main.c

include $(LVGL_DIR)/lvgl/lvgl.mk
include $(LVGL_DIR)/lv_drivers/lv_drivers.mk
#include $(LVGL_DIR)/lv_examples/lv_examples.mk

OBJEXT ?= .o

AOBJS = $(ASRCS:.S=$(OBJEXT))
COBJS = $(CSRCS:.c=$(OBJEXT))

MAINOBJ = $(MAINSRC:.c=$(OBJEXT))

SRCS = $(ASRCS) $(MAINSRC)
OBJS = $(AOBJS) $(COBJS)

## MAINOBJ -> OBJFILES

all: default

%.o: %.c
        @$(CPC)  $(CFLAGS) -c $< -o $@ $(LDFLAGS) 
        @echo "CC $<"

default: $(COBJS) $(MAINOBJ)
        $(CPC) -o $(BIN) $(MAINOBJ) $(COBJS) $(LDFLAGS) 

clean: 
        rm -f $(BIN) $(AOBJS) $(COBJS) $(MAINOBJ)
```

embeddedt, 2020-01-15:
Try changing `LDFLAGS ?=` to `LDFLAGS :=`, and the same for CFLAGS. Maybe the Makefile is bringing its own built-in LDFLAGS which `?=` avoids overwriting.

j-clay, 2020-01-15:
This worked!

Thank you for your help @embeddedt

closed 2020-01-15 by j-clay