lvgl/lv_drivers #91: Need to improve sdl2 drivers support for multiple OS-es.

puzrin, 2020-02-29:
Refs:

- https://github.com/littlevgl/lv_platformio/pull/14
- https://forum.littlevgl.com/t/lv-drivers-sd2-builds-use-unificaion-for-different-os-es/1780

Problem: user still needs to apply additional OS-dependent actions to build emulator (sdl2-based PC drivers) for different OS. That's not convenient, and worth to improve.

In ideal world, user should only install sdl2 bin+headers and add those into PATH. Everything else should work without changes.

kisvegabor, 2020-02-29:
Thanks for opening this issue.
My proposal is: in case of OSX create an `lv_task` in `monitor_init`  where the `*_handler`s are called. `lv_task`s are called from `lv_task_handler` (so from `main` as before).

What do you think?

@husky-koglhof would you be interested in adding this?

puzrin, 2020-02-29:
Why should we do it for OSX only? IMO this should work in linux/webassembly/win too, and using unified method will reduce number of guards.

kisvegabor, 2020-02-29:
Great idea. I've replaced the thread with `lv_task` and it's working on Linux. See here: https://github.com/littlevgl/lv_drivers/commit/4f17cce49dbbf6f19ec3aaefa26928e5d22cf891

Can you confirm if it's working on Windows and OSX as well?

@embeddedt Does it affect Emscripten port?

puzrin, 2020-02-29:
@kisvegabor 

Quick review:

- https://github.com/littlevgl/lv_drivers/blob/4f17cce49dbbf6f19ec3aaefa26928e5d22cf891/display/monitor.c#L43 `MONITOR_APPLE` is not used anymore.
- https://github.com/littlevgl/lv_drivers/blob/4f17cce49dbbf6f19ec3aaefa26928e5d22cf891/display/monitor.c#L358 - can we always use software render and reduce OS divergence even more?

kisvegabor, 2020-02-29:
I agree with both. Fixed here: c6175e4

puzrin, 2020-02-29:
LGTM. The last kludge in this file is `MONITOR_EMSCRIPTEN` guard. I guess, this may be related to thread emulation warkaround (not needed with new `lv_task` approach).

@embeddedt could you check if this guard can be removed (or replaced with runtime check)?

@kisvegabor next point of refactoring is related to https://github.com/littlevgl/lv_drivers/blob/master/win_drv.c. Can anyone explain, why windows implemetation is different and located in separate file? Ideally, this should be in the same place with linux/osx/web display driver (and without guards).

embeddedt, 2020-02-29:
The Windows driver uses direct Win32 API calls for rendering. It doesn't make use of SDL.

kisvegabor, 2020-02-29:
@embeddedt Could you try if it still works with Emscripten?

puzrin, 2020-02-29:
> The Windows driver uses direct Win32 API calls for rendering. It doesn't make use of SDL.

Ah, got it. This is like using "alternate framework". Does sdl2 driver still work for windows? I need to understand possibilities, how to implement PIO demo better.

**Problem**

We still need platform-independent "ticker" to call `lv_tick_inc()` and "usleep" for main loop.

**Possible solutions:**

**A.** Use sdl2 everywhere, and care about automatic win install.

**B.** Make those methods a part of winapi & sdl2 drivers.

embeddedt, 2020-02-29:
> Does sdl2 driver still work for windows?

SDL works on Windows (the Visual Studio project uses it). I chose to also add support for direct Win32 calls because people often have trouble getting SDL to work on Windows.

puzrin, 2020-02-29:
> because people often have trouble getting SDL to work on Windows.

Can you guide, where i can read about problems? Is this installation difficulties or something else?

---

After inspecting `win_drv.c`, seems it's completely different from sdl2:

- lv_driver is created with hardcoded presets
- no keyboard support
- different API at all

Can this be unified to simplify switch between sdl2/winapi?

embeddedt, 2020-03-01:
> Can this be unified to simplify switch between sdl2/winapi?

I think it would be simpler to just keep the Win32 driver the way it is for CodeBlocks, and use the SDL driver for PlatformIO.

> Can you guide, where i can read about problems? Is this installation difficulties or something else?

They had trouble getting the library installed and configuring their IDE to link with it.

embeddedt, 2020-03-01:
Emscripten works; I pushed [a commit](https://github.com/littlevgl/lv_drivers/commit/7d71907c1d6b02797d066f50984b866e080ebeed) that gets rid of the `MONITOR_EMSCRIPTEN` logic.

kisvegabor, 2020-03-01:
> Emscripten works; I pushed a commit that gets rid of the MONITOR_EMSCRIPTEN logic.

Great, thank you!

puzrin, 2020-03-01:
Let's close as solved. All critical issues gone.

I would suggest factor out sdl init & event loop from monitor.c to separate shared module, but that's not critical. Also, renaming files to sdl_* could be nice (not critical too).

kisvegabor, 2020-03-01:
It still needs testing on OSX and Windows.

puzrin, 2020-03-29:
Ups, seems branch still not merged to master. Please merge or reopen.

kisvegabor, 2020-03-30:
I haven't merged it for 2 reasons:
- Needs to be tested at least on Windows (+ Mac would be even better)
- We can't merge it to master because it wouldn't be compatible with the current projects where `main` still has the `#if`s.

I suggest releasing it with `v7.0` when the simulator projects are also updated.

puzrin, 2020-03-30:
I'd prefer to have this released in 6.x, because `lv_platformio` stays in flying state. But if that's absolutely impossible - i'm ok with 7.0.

embeddedt, 2020-03-30:
> Needs to be tested at least on Windows (+ Mac would be even better)

I can try it on Windows if I get time, but I don't have a Mac to test it on.

kisvegabor, 2020-03-31:
> I'd prefer to have this released in 6.x, because lv_platformio stays in flying state. But if that's absolutely impossible - i'm ok with 7.0.

There are no new features to release now in v6 so I let's wait for v7. Sorry for the inconvenience. 

> I can try it on Windows if I get time, but I don't have a Mac to test it on.

Great!

kisvegabor, 2020-03-31:
Works on Mac.

embeddedt, 2020-05-01:
I'll have the Emscripten port's `dev-7.0` branch use `monitor_no_thread` until it gets merged.

kisvegabor, 2020-05-01:
Perfect!

puzrin, 2020-08-01:
Is this solved?

PS. Seems a lot of branches can be removed.

kisvegabor, 2020-08-03:
Yes, it was already merged.

I've cleaned up the old and unused branches.

closed 2020-08-03 by kisvegabor