lvgl/lv_drivers #96: SDL mouse driver ignores touchscreen events

stawnPTL, 2020-04-22:
We're using the SDL display and mouse drivers.  We had naively assumed that if the UI worked with a mouse then it'd work with a touchscreen, however touchscreen events are completely ignored.

In our use case I fixed this by adding the following to `mouse_handler()`, but not sure if this is the correct general solution.  Maybe you'd prefer it if this went into a separate touchscreen driver.

```
        case SDL_FINGERUP:
            left_button_down = false;
            last_x = LV_HOR_RES_MAX * event->tfinger.x / MONITOR_ZOOM;
            last_y = LV_VER_RES_MAX * event->tfinger.y / MONITOR_ZOOM;
            break;
        case SDL_FINGERDOWN:
            left_button_down = true;
            last_x = LV_HOR_RES_MAX * event->tfinger.x / MONITOR_ZOOM;
            last_y = LV_VER_RES_MAX * event->tfinger.y / MONITOR_ZOOM;
            break;
        case SDL_FINGERMOTION:
            last_x = LV_HOR_RES_MAX * event->tfinger.x / MONITOR_ZOOM;
            last_y = LV_VER_RES_MAX * event->tfinger.y / MONITOR_ZOOM;
            break;
```

embeddedt, 2020-04-22:
Hmm, that's interesting. I thought that mouse events would still be emulated for compatibility purposes. 

I think your solution is reasonable and shouldn't break anything on non-touchscreen devices.

kisvegabor, 2020-04-23:
I think it's fine to add it to the mouse driver for simplicity.

Can you send a Pull request?

closed 2020-04-23 by embeddedt