lvgl/lv_drivers #98: master branch does not compile with lvgl master branch

mibcat, 2020-05-21:
Hi !

I'm running into trouble by using both repositories with the current **master** branch.
- lvgl: 2a78353ca485e9d07e2aa5f5543db13e6370580a
- lv_driver: bc24e9fd80db4746b3595245ba4a881265104f0d

Maybe the **lvgl** update to V7 is not considered fully in **lv_drivers** - I found the following issues:

## wrong header file location
The **lv_drv_conf_templ.h** has 3 devices **enabled** per default

- `#define USE_SSD1306        1`
- `#define USE_SSD1306        1`
- `#define USE_AR10XX (1)`

Additionally for these devices the normally used code template is missing:

```c
#ifndef USE_<device>
#  define USE_<device>         0
#endif
```

This leads for all 3 devices to a header include error
```bash
fatal error: lvgl/lv_misc/lv_color.h: No such file or directory 
#include "lvgl/lv_misc/lv_color.h"
```
The lv_color.h file is found at lvgl/**src**/lv_misc/lv_color.h
(quickly fixed by disabling the drivers which are not used in my project anyway)

## the cpp keyword "template" is used as function argument
I'm using the lvgl framework in a c/cpp project and this error pops up after updating to master branch

```bash
In file included from /home/pi/work/lora/Rfm96Raspberry/Application/lv_drivers/display/fbdev.h:20,                                                
                 from /home/pi/work/lora/Rfm96Raspberry/Application/gui.h:12,                                                                     
                 from /home/pi/work/lora/Rfm96Raspberry/Application/main.cpp:9:                                                                   
/home/pi/work/lora/Rfm96Raspberry/Application/lv_drivers/display/../../lv_drv_conf.h: At global scope:                                            
/home/pi/work/lora/Rfm96Raspberry/Application/lv_drivers/display/../../lv_drv_conf.h:235:70: error: expected ‘,’ or ‘...’ before ‘template’       
 static inline int lv_spi_repeat(lv_spi_handle_t spi_dev, const void* template, uint32_t repeats, uint8_t template_size)
```
Renaming the function argument **template** in [lv_drv_conf_templ.h](https://github.com/lvgl/lv_drivers/blob/master/lv_drv_conf_templ.h#L235) fixes this issue.

## warnings
Compiling the master branch shows a lot of new warnings, mainly 
- `warning: unused parameter`
- `warning: no return statement in function`

Currently I'm ignoring this warning and after fixing the other issues temporarily my little demo application is running nearly the same as for V6 (except of the look and feel due to the changed theme handling ;-) )

embeddedt, 2020-05-21:
> Compiling the master branch shows a lot of new warnings, mainly

I compile LittlevGL with many warnings enabled and I only get warnings from one file. What warnings are you using?

> the cpp keyword "template" is used as function argument

Fixed here: https://github.com/lvgl/lv_drivers/commit/627b3d190ea4e1cf3473dd77ccebfcb7d383db69

> The lv_drv_conf_templ.h has 3 devices enabled per default

I'm not entirely sure why these are enabled out-of-the-box. The folder structure they're assuming appears to predate 6.0 so I'm guessing these haven't been updated in a while.

mibcat, 2020-05-22:
Thanks for your feedback.

regarding the warnings:
I'm using this setting (and doesn't change it for v6 or v7)
```cmake
# set compiler settings
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -Wextra -O3")
```
seems to be a cpp related issue (`extern "C"`) in the **lv_drv_conf.h** header:
e.g. each function definition with arguments leads to an unused parameter warning.
[image]

regarding the default enabled devices:
This is strange indeed - I had no compile issues with the **release/v6** branch. 
Should be examined more closely ...

kisvegabor, 2020-05-22:
Sorry, it's my fault. During merging branches I messed up something. 

I reverted it, and now should work.

mibcat, 2020-05-22:
Ok - jumping back will eliminate those issues ;-)

but I still get one warning type (`warning: missing initializer for member`) from **lv_color.h** like

```bash
/home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/../lv_misc/lv_color.h: In function ‘void lv_color_mix_with_alpha(lv_color_t, lv_opa_
t, lv_color_t, lv_opa_t, lv_color_t*, lv_opa_t*)’:
/home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/../lv_misc/lv_color.h:536:47: warning: missing initializer for member ‘lv_color16_t:
:<unnamed struct>::green’ [-Wmissing-field-initializers]
         static lv_color_t fg_color_save = {{0}};
                                               ^
```

I'm not sure but I think those warnings were emitted also in the past ...

kisvegabor, 2020-05-22:
> Ok - jumping back will eliminate those issues ;-)

Great!

> but I still get one warning type (warning: missing initializer for member) from lv_color.h like

That's strange why to complains only about `green` as `red` and `blue` are just next to it. With GCC I don't get a warning for this.

mibcat, 2020-05-23:
sorry my last comment may be misleading:
it's only one **type** of warning - of course there also warnings for the other colors red and blue

I'm using gcc with the additional compiler settings `-Wall -Wextra`
```bash
pi@raspi3b:~ $
> gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

mibcat, 2020-05-23:
yep: `-Wextra` contains `-Wmissing-field-initializers`

kisvegabor, 2020-05-24:
Test run with [these settings](https://github.com/lvgl/lvgl/blob/master/tests/Makefile#L8) and it doesn't show warnings.

Could you send the whole build log and your lv_conf.h?

mibcat, 2020-05-24:
You are right !
The warnings only occurs due my attempt to use lvgl in my class `gui.cpp` and are thrown if this module is including lvgl headers, e.g.
```bash
[ 89%] Building CXX object CMakeFiles/rfm96_test.dir/gui.cpp.o                                                                                    
In file included from /home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/lv_hal_disp.h:21,                                             
                 from /home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/lv_hal.h:16,
                 from /home/pi/work/lora/Rfm96Raspberry/Application/lvgl/lvgl.h:22,
                 from /home/pi/work/lora/Rfm96Raspberry/Application/lv_drivers/display/fbdev.h:29,
                 from /home/pi/work/lora/Rfm96Raspberry/Application/gui.h:12,
                 from /home/pi/work/lora/Rfm96Raspberry/Application/gui.cpp:8:
/home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/../lv_misc/lv_color.h: In function ‘void lv_color_mix_with_alpha(lv_color_t, lv_opa_
t, lv_color_t, lv_opa_t, lv_color_t*, lv_opa_t*)’:
/home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/../lv_misc/lv_color.h:536:47: warning: missing initializer for member ‘lv_color16_t:
:<unnamed struct>::green’ [-Wmissing-field-initializers]
         static lv_color_t fg_color_save = {{0}};
                                               ^
/home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/../lv_misc/lv_color.h:536:47: warning: missing initializer for member ‘lv_color16_t:
:<unnamed struct>::red’ [-Wmissing-field-initializers]
/home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/../lv_misc/lv_color.h:537:47: warning: missing initializer for member ‘lv_color16_t:
:<unnamed struct>::green’ [-Wmissing-field-initializers]
         static lv_color_t bg_color_save = {{0}};
                                               ^
/home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/../lv_misc/lv_color.h:537:47: warning: missing initializer for member ‘lv_color16_t:
:<unnamed struct>::red’ [-Wmissing-field-initializers]
/home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/../lv_misc/lv_color.h:538:49: warning: missing initializer for member ‘lv_color16_t:
:<unnamed struct>::green’ [-Wmissing-field-initializers]
         static lv_color_t res_color_saved = {{0}};
                                                 ^
/home/pi/work/lora/Rfm96Raspberry/Application/lvgl/src/lv_hal/../lv_misc/lv_color.h:538:49: warning: missing initializer for member ‘lv_color16_t:
:<unnamed struct>::red’ [-Wmissing-field-initializers]
```
But in the end, due to the static c implementation of lvgl, I had to make so many "exceptions" in the class that it really doesn't make sense to force a `cpp` class for this.

Reverting back the code to `c` removes all warnings - so we can close this issue.

Thanks for your support !!

embeddedt, 2020-05-24:
We should still fix it, as C++ code should be able to include public headers without warnings.

mibcat, 2020-05-24:
OK, fine !

Adding the missing parts (in my case green and blue because of  `LV_COLOR_DEPTH` = 16) will eliminate the warning

e.g. replacing line 536:
`static lv_color_t fg_color_save = {{0}};`
by this one:
`static lv_color_t fg_color_save = { { 0, 0, 0 } };`

But due to the different implementation possibilities (dependency on `LV_COLOR_DEPTH` and others) a new **init macro** should be created ...

kisvegabor, 2020-05-25:
I fixed it like this: 966b4b11

Do you still see the warning?

mibcat, 2020-05-25:
No - no warnings anymore (for both c and cpp) :+1:

embeddedt, 2020-05-25:
Great! I guess we can close this issue then.

closed 2020-05-25 by embeddedt

kisvegabor, 2020-05-25:
Great! :)