lvgl/lv_drivers #111: VGA and PS/2 driver

bil-ash, 2020-10-20:
Someone please implement a VGA display and PS/2(keyboard and mouse) LVGL driver for embedded devices which is implemented on software side and depends only on GPIO access. I came across two such similar libraries- [FabGL](https://github.com/fdivitto/FabGL) and [vgax](https://github.com/smaffer/vgax) but both of them are hardware (ESP32/ARDUINO) specific, would be happy if there is no hardware-specific code in the new driver.

kisvegabor, 2020-10-21:
I thought these interfaces are quite outdated now. 
Could you describe your use case to better understand the need for these drivers?

bil-ash, 2020-10-26:
I would like to output graphics from a [linkit smart](https://labs.mediatek.com/en/platform/linkit-smart-7688) to a computer monitor over VGA port. This is because the other port over which this is possible(at least,theoretically) is HDMI but that protocol is more complex. At the same time I would like to have provision for the library to be usable on [Arduino Portenta H7](https://store.arduino.cc/usa/portenta-h7) or any other board having similar processing power and RAM. This is only possible if no-hardware specific code is used and the only one board-specific (linux,arduino,etc) function is used-DIGITAL WRITE.

kisvegabor, 2020-10-27:
I see, thank you.

Regarding Portenta, I had a collaboration with Ardunio to port LVGL to Portenta. So I can confirm that it supports LVGL.

bil-ash, 2020-10-27:
> I would like to output LVGL graphics from a [linkit smart](https://labs.mediatek.com/en/platform/linkit-smart-7688) to a computer monitor over VGA port

My main concern is the ability to show output on desktop monitors using VGA port(via board's GPIO pins), I am not much concerned about hardware on which it runs(linkit smart/portenta).
What I am able to see from this repo is that no such drivers exist,so I have opened this issue **to request such a driver for lvgl** . It is not an immediate need,I would be fine if the drivers are ready by next 2 months.

bil-ash, 2020-11-27:
@kisvegabor Are you interested in building the vga driver?There is a nice demo at https://github.com/abelykh0/VGA-demo-on-bluepill . It would be nice if you could make something similar but generic in nature,i.e., make timer,pwm and GPIO non-specific to any board so that it can be used on ESP32, Arduino due,linkit smart or any other board.

kisvegabor, 2020-12-01:
@bil-ash 
Personally, I'm not interested in it. :slightly_frowning_face: My main focus is on the development of LVGL and unfortunately have no time to develop special drivers.

I suggest opening a topic on the Forum. Maybe someone will get interested!

bil-ash, 2020-12-01:
Ok,so closing the issue for now

closed 2020-12-01 by bil-ash