lvgl/lv_drivers #113: How to rotate LVGL output to linux framebuffer driver?

roddehugo, 2020-11-13:
Hi,

In the current release of LVGL drivers, the linux framebuffer driver is present in `display/fbdev.{h,c}`. But does it use the `fbdev(4)` as described here https://manpages.debian.org/stretch/xserver-xorg-video-fbdev/fbdev.4.en.html?

We are developing an GUI using LVGL and a custom linux distribution built with yocto on targeting an i.MX6ULL platform. The display attached is by default in landscape mode, but we want the LVGL GUI to be in portrait mode. We currently set `USE_FBDEV=1` and `USE_EVDEV=1` in the config file. 

Hence, with such a setup, would it be possible to rotate the screen CCW by 90° to achieve expected portrait mode?

In the yocto recipe, we are using the framebuffer distribution, no X11, no Wayland. I'd like to keep it like that because our need is simple: a fullscreen GUI without any desktop environment. I could eventually add X11 to the image and use `Xorg.conf` file as stipulated int the link above, but wouldn't LVGL bypass X11 since the `display/fbdev.c` directly use mapped memory to flush graphics?

Thanks for the help,
Hugo

kisvegabor, 2020-11-13:
Hi,

Mainly for performance reasons, LVGL doesn't support screen rotation by default. However, if you have some extra memory and computational power (which assume you have on Linux based system) you can try this conversation in the `disp_flush` function:
https://forum.lvgl.io/t/setting-rotated-option-in-lv-disp-drv-t-causes-hard-fault/440/10

roddehugo, 2020-11-13:
Thanks for the snippet. I'll try it, indeed we have spare computational power. 

Would it be appreciated to submit a PR with the feature conditionally available using preprocessor, alike in `evdev`?

kisvegabor, 2020-11-16:
> Would it be appreciated to submit a PR with the feature conditionally available using preprocessor, alike in evdev?

For `fbdev` it'd be great, thanks!
In the long term, we should integrate it into LVGL in some form. But it's a more complicated story :slightly_smiling_face:   

For `evdev` already have the `EVDEV_SWAP_AXES` config so the user can configure it accordingly.

roddehugo, 2020-11-17:
Alright, it is not that obvious! I'll need a playground and time to get this right for various bpp formats.

> In the long term, we should integrate it into LVGL in some form. But it's a more complicated story

What do you have in mind?

kisvegabor, 2020-11-18:
> What do you have in mind?

LVGL could automatically do the rotation before sending the data to the `flush` function. It's no ideal because this way 
1. only 1 line (the rotated) is sent to the `flush` function at a time
2. the whole rotated image is rotated but it needs a lot of extra memory

closed 2020-12-29 by stale[bot]