lvgl/lv_drivers #115: [fbdev] Shouldn't the framebuffer driver use color channels fb_bitfield from fb_var_screeninfo

roddehugo, 2020-11-17:
Hi,

I have the following issue with my setup. The data lines for red and blue channels of the LCD are inverted. This leads to having red and blue colors swapped between the UI on desktop (dev env) and UI on target (production env).

I noticed the returned value for `red`, `green` and `blue` channels are not really as expected and as assumed by LVGL (and plenty other tools like https://github.com/phoenix826/fb-test-app). Each pixel is not RGBA (32bits) but rather BGRA (32bits).

One can read this info from `struct fb_var_screeninfo`. I have not tested, but it looks like this guy is leveraging that data to properly fill pixels into the memory mapped framebuffer https://github.com/sigma-embedded/fbtest/blob/master/fbtest.c.

Here is an extract from the linux kernel `fb` API (`include/uapi/linux/fb.h`):
```
/* Interpretation of offset for color fields: All offsets are from the right,
 * inside a "pixel" value, which is exactly 'bits_per_pixel' wide (means: you
 * can use the offset as right argument to <<). A pixel afterwards is a bit
 * stream and is written to video memory as that unmodified.
 *
 * For pseudocolor: offset and length should be the same for all color
 * components. Offset specifies the position of the least significant bit
 * of the pallette index in a pixel value. Length indicates the number
 * of available palette entries (i.e. # of entries = 1 << length).
 */
struct fb_bitfield {
	__u32 offset;			/* beginning of bitfield	*/
	__u32 length;			/* length of bitfield		*/
	__u32 msb_right;		/* != 0 : Most significant bit is */ 
					/* right */ 
};

struct fb_var_screeninfo {
	__u32 xres;			/* visible resolution		*/
	__u32 yres;
	__u32 xres_virtual;		/* virtual resolution		*/
	__u32 yres_virtual;
	__u32 xoffset;			/* offset from virtual to visible */
	__u32 yoffset;			/* resolution			*/

	__u32 bits_per_pixel;		/* guess what			*/
	__u32 grayscale;		/* 0 = color, 1 = grayscale,	*/
					/* >1 = FOURCC			*/
	struct fb_bitfield red;		/* bitfield in fb mem if true color, */
	struct fb_bitfield green;	/* else only length is significant */
	struct fb_bitfield blue;
	struct fb_bitfield transp;	/* transparency			*/	
	[...]
}
```

I guess my issue is very specific and I managed to get around it by switching channels in `lv_color_t` right before it is copied into the framebuffer device.

But how much work would that represent to get support for color channels into `fbdev`?

Thanks!

kisvegabor, 2020-11-18:
Hi,

I see the issue.
It'd be a little bit difficult to make it as a run time config mainly because of images. Many users convert images to C array (not using PNG or other external files) and the order of bytes is hardcoded into the arrays. 

However, it could be a compile-time config and the converted images could store the RGB and BRG formats as well and the format is selected based on the runtime config. 

Anyway, users noted that LVGL should have support more color formats, even custom ones. So a better "color format handler" should be considered.

kisvegabor, 2020-12-21:
Now we are on v7, and v8 is under development. For v9 I'm planning some graphics related updates. More bpp options will be considered at that time. See the [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md#v9)

closed 2021-01-18 by kisvegabor