lvgl/lv_drivers #151: evdev / libinput: use multiple devices

Johennes, 2021-08-31:
The current evdev and libinput drivers store state in static variables. That means they can only be used with one device at a time but I cannot use, say, two trackpads or a trackpad and a keyboard at the same time (or at least my weak C-fu doesn't let me see how). It would be great if the driver state could instead be stored in an external structure, to be managed by the caller. This way I could connect an arbitrary number of input devices via either of these drivers.

Johennes, 2021-09-03:
Actually we wouldn't even need the state to be managed externally. We could instead store it in the driver's [`user_data`](https://github.com/lvgl/lvgl/blob/master/src/hal/lv_hal_indev.h#L101) field. I made a hacky proof-of-concept [here](https://gitlab.com/cherrypicker/unl0kr/-/blob/master/libinput_multi.c).

kisvegabor, 2021-09-03:
I agree to support various type of indevs (mouse+touch+keypad) but I can't imagine a use case for e.g. 2 keyboard, or 2 mouses. 

If we target only 1 indev per type we can simply hardcode `state_mouse`, `state_touch`, etc.

What do you think?

Johennes, 2021-09-03:
Ah, that's an interesting thought. I suppose that would get me half of the way at least.

There are two cases I can think of where being able to connect multiple devices of the same type might be helpful:

1.A device with built-in keyboard (for instance, a Pinebook) that is also docked to an external USB keyboard
2. Any device docked to an external trackpad and a keyboard with mouse emulation. For example, I have an Ergodox EZ and I occasionally like to control the mouse with it but most of the time I use a trackpad. The keyboard's mouse emulation shows up as a separate device though.

```
/dev/input/event8:      Apple Inc. Magic Trackpad 2
/dev/input/event9:      ZSA Ergodox EZ
/dev/input/event10:     ZSA Ergodox EZ Mouse
/dev/input/event11:     ZSA Ergodox EZ System Control
/dev/input/event12:     ZSA Ergodox EZ Consumer Control
/dev/input/event13:     ZSA Ergodox EZ Keyboard
```

Both of these scenarios are probably not overly common though so if you think it'd add too much complexity to allow for multiple devices of the same type, just permitting one device per type might be enough. On the other hand, the step from that solution to allowing an arbitrary number of devices might also not be too big anymore. :thinking:

kisvegabor, 2021-09-05:
I see, these are really possible scenarios. The problem is that now lvgl's `libinput` driver has an `init` and `read_cb` API. It means the user needs to create the indev driver assign the  `read_cb`. If we need to use the `user_data` to  store e.g. the name of the `dev` in the driver instance we need to pass it to the user too.

It'd be more flexible if an `lv_libput_init()` created as many indev driver's as required. E.g. 2 mouse, 1 keyboard, 1 touchpad.

But such change would be API breaking. Therefore (for now) I vote for supporting one indev per type as it fits into current API schema. 

We have plans to rework the driver and maybe marge into the lvgl repo. We need think about this case when designing a new unified driver API.

Johennes, 2021-09-05:
> If we need to use the `user_data` to store e.g. the name of the `dev` in the driver instance we need to pass it to the user too.

Yeah, that's true. In my current hacky workaround I had to introduce [`void libinput_multi_init_driver(lv_indev_drv_t * indev_drv)`](https://gitlab.com/cherrypicker/unl0kr/-/blob/master/libinput_multi.c#L118) to be called instead of `void libinput_init(void)` for initializing the driver state and [`bool libinput_multi_set_file(lv_indev_drv_t * indev_drv, char* dev_name)`](https://gitlab.com/cherrypicker/unl0kr/-/blob/master/libinput_multi.c#L81) to be called instead of `bool libinput_set_file(char* dev_name)` to store the device file into the driver state.

One thing I wonder is if we could maybe encapsulate all the currently global state into a struct and add a default struct variable that is used whenever the user calls the current `libinput_init` / `libinput_set_file` functions. That way we could keep the current public API that supports only one device at a time but also offer the two additional functions to handle multiple standalone drivers. :thinking: 

Let me maybe play around with this a little and I may send a draft PR to better demonstrate what I mean.

Otherwise, supporting one indev per type would be ok-ish for my use case as well in the short term.

PS: Thanks a ton for considering all my requests! :slightly_smiling_face:

kisvegabor, 2021-09-06:
> PS: Thanks a ton for considering all my requests! 

I also thank you for sending these PRs and issues. This is what moves LVGL forward! :metal: 

As you suggest we can have an other init function that creates all the required indevs. Maybe it's not the cleanest solution  but very useful.

commit 7a0f7887 (2021-09-23): Make it possible to use multiple devices with the libinput and XKB drivers (#165)

This change bundles all of the previously file-global device-specific
state in libinput.c and xkb.c into new driver state structs with a
default value. All of the existing public functions are retained and
operate on the default struct values. To allow the use of multiple
devices, a second set of public functions is introduced that also take
the specific driver state as an argument. It is left up to the caller
where to store the driver state and how to retrieve it when reading
input events.

Closes: #151
closed 2021-09-23 by kisvegabor