lvgl/lv_drivers #153: evdev / libinput: "full" keyboard support

Johennes, 2021-09-01:
When using the `LV_INDEV_TYPE_KEYPAD` type, the evdev driver supports a limited number of [keys](https://github.com/lvgl/lv_drivers/blob/master/indev/evdev.c#L174) (backspace, enter, up right, left, down). Any other keys are ignored. It would be great if the driver would handle input from a regular keyboard so that, for instance, typed characters are forwarded into a textarea.

I'm not sure if there is a better way to achieve this other than adding further cases for the relevant [key codes](https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h#L105) and handling things like SHIFT manually.

Maybe ASCII characters or characters found on common US English keyboard layouts would be enough for a start as that would limit the amount of `switch`ing.

I'm curious if you'd be open to adding this to lv_drivers and would be happy to help implement it.

kisvegabor, 2021-09-02:
For "normal letters" `data->key = in.code;` should work instead of `data->key = 0;` in the `default` branch of the `switch case`.

It really works only for ASCII letters. Isn't there a way to get the "real" letters, e.g. `shift + a = A`?

Johennes, 2021-09-03:
It seems that [libxkbcommon](https://xkbcommon.org/doc/current/modules.html) would be able to do the translation from key codes to key symbols based on the current key map. I'll have to do some more research to understand what would be required to hook into that.

Johennes, 2021-09-03:
Looks like libxkbcommon is working pretty ok for me so far. We'd only have to add a config setting for the key map to use. E.g. this is what I use with my laptop keyboard:

```
struct xkb_rule_names names = {
    .rules = NULL,
    .model = "pc105",
    .layout = "de",
    .variant = "nodeadkeys",
    .options = NULL
};
```

kisvegabor, 2021-09-03:
Good news! :slightly_smiling_face: 

> We'd only have to add a config setting for the key map to use.

Shall we add parts of this config to `lv_drv_conf.h`?

Johennes, 2021-09-03:
Yeah, good idea. :+1:

Let me play with it some more and then I'll try to put it into a pull request.

PR #156 (2021-09-14): Add full keyboard support to libinput/evdev driver
This enhances the existing libinput and evdev driver by adding support for full keyboards. The conversion from Linux scan codes to UTF-8 characters is performed via XKB. To control the behavior a new config setting for opting into or out of using XKB was added. The keymap can either be set via another config option or by calling a function at runtime.

Two small caveats:

1. This only works for a single connected keyboard because static variables are used to store the keyboard state. Supporting multiple devices of the same type is still being  [... cut]
commit 485b150d (2021-09-14): feat(libinput/evdev) add full keyboard support (#156)

This enhances the existing libinput and evdev driver by adding support
for full keyboards. The conversion from Linux scan codes to UTF-8
characters is performed via XKB. To control the behavior a new config
setting for opting into or out of using XKB was added. The keymap can
either be set via another config option or by calling a function at
runtime.

Two small caveats:

(1) This only works for a single connected keyboard because static
variables are used to store the keyboard state. Supporting multiple
devices of the same type is still b [... cut]
closed 2021-09-14 by kisvegabor