lvgl/lv_drivers #155: libinput: automatic device discovery

Johennes, 2021-09-03:
I'm working on an application that is meant to be run on many different types of devices which means I can't hard-code the `/dev/input/event...` paths because I don't know them at build time. It would be great if the libinput driver could provide an interface for automatically detecting touchscreens, pointer devices and keyboards.

I have some very rudimentary code [here](https://gitlab.com/cherrypicker/unl0kr/-/blob/master/libinput_device_discovery.c#L66) that basically just walks the `/dev/input/event...` files, makes a libinput call to determine the device capabilities and then filters the files based on that.

I'd be interested to know if you would consider adding something like this to lv_drivers? I think this could be beneficial to other users as well.

kisvegabor, 2021-09-03:
What if you simply do `#define LIBINPUT_NAME   my_find_input_dev()`? 

Anyway, I believe it'd be useful to add `libinput_find_touch_dev()`, `libinput_find_mose_dev()`, etc  to `libinput.c` so that the user could use them as `LIBINPUT_NAME`. 

Probably we'd need `LIBINPUT_TOUCH_NAME`, `LIBINPUT_MOUSE_NAME`, etc.

What do you think?

Johennes, 2021-09-03:
Yeah, this overlaps a little with #151.

Using it directly in the config file is an interesting idea! The functions I'm using so far return all devices of a certain type. We could maybe wrap them in another set of functions to return only the first  device of a type, that shouldn't be too complicated.

kisvegabor, 2021-09-05:
> Using it directly in the config file is an interesting idea! The functions I'm using so far return all devices of a certain type. We could maybe wrap them in another set of functions to return only the first device of a type, that shouldn't be too complicated.

Based on my [last comment](https://github.com/lvgl/lv_drivers/issues/151#issuecomment-913202619) in #151, I think it'd be the approach now.

PR #157 (2021-09-09): Automatic device discovery via libinput
This adds new convenience functions to the libinput driver for automatically detecting input devices of a certain type (such as keyboards or touchscreens).

My pointer magic is weak so I hope I did everything correctly here. Looking forward to candid feedback. :slightly_smiling_face: 

The logic could also be generalized to work with the evdev driver, too. That would just require replacing the libinput calls with `ioctl`s to the kernel. Since I'm not currently using the evdev driver, I didn't take the time to do that but it should be relatively straightf [... cut]
commit 8860f0dd (2021-09-09): Automatic device discovery via libinput (#157)

This adds new convenience functions to the libinput driver for
automatically detecting input devices of a certain type (such as
keyboards or touchscreens).

Closes: #155
closed 2021-09-09 by kisvegabor