lvgl/lv_drivers #162: sdl.c "SDL_StartTextInput" opens native device keyboard

Slluxx, 2021-09-10:
This function is called inside sdl_init() and just opens the device keyboard when running lvgl, which is not desired for me.

I commented the function and now the app works fine. I am not sure what this function is exactly used for but i'm typing this because it might be looked at. My device is the Nintendo Switch.

embeddedt, 2021-09-11:
We probably need a config option to skip calling this, as it appears that it is supposed to open the system IME. As a result of the reorganization done in https://github.com/lvgl/lv_drivers/commit/2e27975cf5858ddc1ba68d1e46aa869d07df98a4, it is no longer possible to simply skip using `keyboard_init`.

kisvegabor, 2021-09-13:
I thought it's more convenient to initialize keyboard, mouse and mousewheel automatically because if they are present probably the user wants them to use too. 

But we can add `SDL_USE_KEYBOARD/MOUSE/MOUSEWHEEL` [here](https://github.com/lvgl/lv_drivers/blob/master/lv_drv_conf_template.h#L106).  What do you think?

Slluxx, 2021-09-13:
Software/LVGL or Native keyboard does not really matter to me. Sure, native usually is the better choice but if initializing means it just "pops up" when i start my GUI, i rather have it disabled.
Having options to enable/disable those input features seems good.

kisvegabor, 2021-09-17:
I've just realized an issue with the current `SDL_StartTextInput` based implementation. It saves the hit characters into a buffer and generates dummy presses immediatelly followed by releases to pass the characters to lvgl. But hence all the character presses are very short and lvgl can't see long press related events. 

So I suggest adding an `LV_SDL_TEXT_INPUT` to choose between the two modes.

closed 2022-04-27 by stale[bot]