lvgl/lv_drivers #187: Missing Symbols In Generated Static Library File

napperley, 2021-11-01:
After the static library is generated for the LV Drivers library there is an error occurring during link time where the linker reported that the **evdev_init** symbol is missing.

There are some missing symbols in the static library. Here is the output after running `nm liblv_drivers.a`:
```
CMakeCCompilerId.o:
00000000 t $a
00000000 r $d
00000000 d $d
000000e0 t $d
00000008 D info_arch
00000000 D info_compiler
0000000c D info_language_dialect_default
00000004 D info_platform
00000014 R info_version
00000000 T main

GC9A01.o:

ILI9341.o:

R61581.o:

SHARP_MIP.o:

SSD1963.o:

ST7565.o:

UC1610.o:

drm.o:

fbdev.o:
00000000 t $a
00000194 t $a
000001b8 t $a
0000071c t $a
0000077c t $a
00000000 b $d
00000000 r $d
00000160 t $d
000001b4 t $d
00000710 t $d
00000778 t $d
000007b8 t $d
         U close
00000194 T fbdev_exit
000001b8 T fbdev_flush
0000071c T fbdev_get_sizes
00000000 T fbdev_init
0000077c T fbdev_set_offset
000000ec b fbfd
000000e4 b fbp
000000a0 b finfo
         U ioctl
         U lv_disp_flush_ready
         U memcpy
         U memset
         U mmap
         U open
         U perror
000000e8 b screensize
00000000 b vinfo

gtkdrv.o:

AD_touch.o:

FT5406EE8.o:

XPT2046.o:

evdev.o:

libinput.o:

xkb.o:

sdl.o:

sdl_gpu.o:

wayland.o:

win32drv.o:

win_drv.o:
```

Take note of the **o** files that are empty with no symbols (especially **evdev.o** and **libinput.o**). This is very strange to see with the static library file. Some **o** files are included that shouldn’t be present in the static library file (eg **win32drv.o**) when the Linux Frame Buffer backend is used. No **o** files in the static library file should be empty. Why are there missing symbols in the static library file?

kisvegabor, 2021-11-02:
Have you enabled EVDEV in `lv_drv_conf.h`?

napperley, 2021-11-02:
No I haven't. After I enabled **EVDEV** in the header file, rebuilt the LV Drivers library, and reran the linking task for the Kotlin Native program the linking error no longer occurs :). With the static library file I have noticed that there are significantly more symbols included, and the library file is now 2 MB in size instead of 22 KB.

closed 2021-11-02 by napperley

kisvegabor, 2021-11-03:
You can enable/disable only what you really need. The disabled "modules" won't be built.