lvgl/lv_drivers #188: libinput pointer driver doesn't play nicely with screen rotation

Johennes, 2021-11-07:
When processing pointer motion events, the coordinates are [clamped](https://github.com/lvgl/lv_drivers/blob/master/indev/libinput.c#L401) to the display dimensions using `LV_HOR_RES` and `LV_VER_RES`, respectively. When the display is rotated, the latter are adapted according to the angle. The coordinates returned by `libinput_event_pointer_get_dx` and `libinput_event_pointer_get_dy`, however, ignore any rotation. As a result, the clamping doesn't work correctly. If the display is rotated by 90 degrees, for instance, the x coordinate is clamped with `disp->driver->ver_res`.

Using e.g. `lv_disp_get_default()->driver->hor_res` instead of `LV_HOR_RES` fixes the issue for the clamping.

It's possible that I'm not using the display rotation feature correctly so I wanted to open this as an issue first before sending a pull request. :sweat_smile:

kisvegabor, 2021-11-09:
Hi,

The clamping can be safely removed from here as LVGL does it internally. 
https://github.com/lvgl/lvgl/blob/61b0de3bdc46803ae245cf17b453678fd0b30d73/src/core/lv_indev.c#L347-L353

Johennes, 2021-11-09:
Hm, I think the snippet you linked doesn't actually clamp. It just logs a warning when the position is out of the display area but then proceeds with the coordinates regardless, doesn't it?

In my case I'm drawing to a subsection of the display only. When the mouse cursor reaches the boundary of that subsection, I'd like it to stop and not move further. When I remove the clamping in `libinput.c`, the cursor will move beyond the display edge but not draw so in order to get it back into the display I have to move it by an equivalent distance while not being visible.

Maybe the clamping from `libinput.c` could actually move inside of `indev_pointer_proc`? The rotation is already accounted for in there so that problem would be solved. The only issue that I can think of are touchscreen indevs. They don't clamp but [truncate](https://github.com/lvgl/lv_drivers/blob/master/indev/libinput.c#L386) touches that are out of bounds. I think it's not possible to differentiate between a touchscreen pointer indev and a mouse pointer indev in `indev_pointer_proc` though. :thinking:

kisvegabor, 2021-11-16:
> Hm, I think the snippet you linked doesn't actually clamp. It just logs a warning when the position is out of the display area but then proceeds with the coordinates regardless, doesn't it?

Oh, you are right. I was sure that there is clamping in LVGL and on a quick look I thought it's that. But actually, there is no clamping in LVGL. :slightly_smiling_face: 

> I think it's not possible to differentiate between a touchscreen pointer indev and a mouse pointer indev in indev_pointer_proc though.

Yes, they look the same for LVGL.

If the right edge of the "subscreen" is at 480 px but LVGL sees larger coordinates it can only truncate.
```
500->480
1500->480
1400-> 480 but should be 380
```
To manage this in LVGL it should have relative and absolute pointer types. For now, I think it's easier to solve in the driver.

PR #192 (2021-11-22): fix(libinput): read display dimensions from driver
This changes the libinput driver to read unrotated display dimensions directly from the driver. This is necessary because libinput itself doesn't respect any rotation inside of LVGL. As a result, coordinates retrieved from libinput need to be compared to the unrotated dimensions.

Fixes: #188
commit 33983bcb (2021-11-22): fix(libinput) read display dimensions directly from driver (#192)

This changes the libinput driver to read unrotated display dimensions
directly from the driver. This is necessary because libinput itself
doesn't respect any rotation inside of LVGL. As a result, coordinates
retrieved from libinput need to be compared to the unrotated dimensions.

Fixes #188
closed 2021-11-22 by embeddedt