lvgl/lv_drivers #198: DRM:How to enable double buffering

fanday, 2022-02-20:
Hi,
    Currently the DRM display driver works fine,but this driver only use single DRM panle.So, is there any sample code for double DRM panle driver?and in my opinion the double will be more fast and smooth.Thanks.

kisvegabor, 2022-02-20:
The DRM driver was contributed by @superna9999 and @leiflm  so I cc them here. Could take look at this issue?

superna9999, 2022-02-21:
Hi, the DRM driver uses 2 buffers to avoid tearing, it can't be smoother. Maybe tweaking the LVGL config could help.

It's possible to pass the mapped DRM buffers to LVGL and match the buffer pointer and flush the associated BO, but since each frame is redrawn at each vsync, the cost of drawing on uncached video buffers is higher than drawing on cached user buffer then memcpy on the video buffers.

closed 2022-04-27 by stale[bot]