lvgl/lv_drivers #211: sdl_common.c is missing #if USE_SDL || USE_SDL_GPU

johado, 2022-04-26:
When neither USE_SDL or USE_SDL_GPU SDL_INCLUDE_PATH will not be set and you end up with:
workspace/sources/lvgl/lv_drivers/sdl/sdl_common.h:33:10: error: #include expects "FILENAME" or <FILENAME>
   33 | #include SDL_INCLUDE_PATH

johado, 2022-04-26:
The following seems to fix this:
`

diff --git a/sdl/sdl_common.c b/sdl/sdl_common.c
index a636f3c..a26c97c 100644
--- a/sdl/sdl_common.c
+++ b/sdl/sdl_common.c
@@ -4,6 +4,7 @@
 
 #include "sdl_common.h"
 
+#if USE_SDL || USE_SDL_GPU
 /*********************
  *      DEFINES
  *********************/
@@ -262,4 +263,4 @@ uint32_t keycode_to_ctrl_key(SDL_Keycode sdl_key)
     }
 }
 
-
+#endif  /* USE_SDL || USD_SDL_GPU */
diff --git a/sdl/sdl_common.h b/sdl/sdl_common.h
index 7501324..b2832c4 100644
--- a/sdl/sdl_common.h
+++ b/sdl/sdl_common.h
@@ -27,6 +27,8 @@ extern "C" {
 #include "lvgl/lvgl.h"
 #endif
 
+#if USE_SDL || USE_SDL_GPU
+
 #ifndef SDL_INCLUDE_PATH
 #define SDL_INCLUDE_PATH MONITOR_SDL_INCLUDE_PATH
 #endif
@@ -93,6 +95,7 @@ void keyboard_handler(SDL_Event * event);
 /**********************
  *      MACROS
  **********************/
+#endif /* USE_SDL || USE_SDL_GPU */
 
 #ifdef __cplusplus
 } /* extern "C" */
`

kisvegabor, 2022-05-01:
Hi,

Could you send a  pull request?

johado, 2022-05-02:
Created https://github.com/lvgl/lv_drivers/pull/213

closed 2022-05-03 by johado