lvgl/lv_drivers #222: listener function for opcode 2 of xdg_toplevel is NULL

MY201314MY, 2022-06-23:
Afer I build my project, fhe following errors occurred
`listener function for opcode 2 of xdg_toplevel is NULL
Aborted (core dumped)`
My configuration is:
`#if USE_WAYLAND
/* Support for client-side decorations */
#  ifndef LV_WAYLAND_CLIENT_SIDE_DECORATIONS
#    define LV_WAYLAND_CLIENT_SIDE_DECORATIONS 1
#  endif
/* Support for (deprecated) wl-shell protocol */
#  ifndef LV_WAYLAND_WL_SHELL
#    define LV_WAYLAND_WL_SHELL 0
#  endif
/* Support for xdg-shell protocol */
#  ifndef LV_WAYLAND_XDG_SHELL
#    define LV_WAYLAND_XDG_SHELL 1
#    define LV_WAYLAND_TIMER_HANDLER 1
#  endif
#endif`

`#define LV_LOG_LEVEL LV_LOG_LEVEL_WARN`
Did anyone have the same problem as me?

kisvegabor, 2022-06-24:
I cc @simplejack-src who have added some major refactoring and improvements to the Wayland driver.

simplejack-src, 2022-06-24:
I've made a [pull request](#223) for the first one (looks like the drivers missing an event handler for an event that was added to the XDG shell protocol in a later revision). The second one I'm not so sure about, it looks like ```wl_shell``` isn't supported by your compositor (however since our support of that shell is deprecated, I'm not too worried about that one).

MY201314MY, 2022-06-24:
Thank you for your reply.
I did a test with the patch, and it works with the following code:
```
int main() {  
    lv_init();
    lv_wayland_init();

    lv_disp_t * display = lv_wayland_create_window(H_RES, V_RES, "Window Title", NULL);

    lv_example_get_started_1();

    while (1) {
        lv_tick_inc(5);
        lv_task_handler();
        usleep(5*1000);
    }

    return 0;
}
```
But If I add  **_lv_wayland_window_set_fullscreen(display, false);_** after **_lv_wayland_create_window(H_RES, V_RES, "Window Title", NULL);_**, the following errors occured on the window after it runs about 30 seconds later:

> "WIndow Title" is not responding.
> You may choose to wait a short while for it to continue or
> force the application to quit entirely.

simplejack-src, 2022-06-24:
I can't reproduce that (granted my compositor is Wayfire). Is your LVGL configuration correct? i.e. ```LV_WAYLAND_TIMER_HANDLER``` is not defined (since your using the Wayland driver in a polling setup)?