lvgl/lv_drivers #225: LVGL with Wayland on Embedded Linux

symfund, 2022-06-28:
All controls can receive the touch events, but do not redraw the display.

https://github.com/symfund/lvgl-port-ma35d1/blob/master/IMG_2781.PNG

kisvegabor, 2022-06-28:
I moved the issue to the driver's repository. 

@simplejack-src. do you have any ideas?

simplejack-src, 2022-06-28:
Quick look at ```main.c``` in your repository, it seems as though you forgot to setup the ```struct pollfd pfd;``` structure. Adding something like below (before your ```while (1)``` loop) will wait until new incoming data is received on the Wayland file descriptor/pipe:
```
pfd.fd = lv_wayland_get_fd();
pfd.events = POLLIN;
```

In addition if you want to wait and loop again if the Wayland TX pipe gets full, you can add (before the actual ```poll(...)```):
```
if (lv_wayland_window_is_flush_pending(NULL)) {
   pfd.events |= POLLOUT;
} else {
   pfd.events &= (~POLLOUT);
}
```

symfund, 2022-06-28:
Thank  @simplejack-src for your help.
Thank @kisvegabor for your excellent LVGL!
LVGL natively supports Wayland is awesome.

Should we update the README: https://github.com/lvgl/lv_drivers/blob/master/wayland/README.md ?
My main code is borrowed from README.

symfund, 2022-06-29:
I found the problem! UI does not render because the backing buffer is always busy.

symfund, 2022-06-29:
In _lv_wayland_flush(),  It seems no harm to comment out the following lines:
`
#if 0
else if (buffer->busy)
{
   lv_disp_flush_ready(disp_drv);
  return;
}
#endif
`
When touch the calendar's YEAR dropdown, the popup dropdown list appears.

simplejack-src, 2022-06-29:
That check is needed to enforce the Wayland protocol. Effectively, it ensures that the buffer that was "sent" (committed) to the compositor isn't touched by the client until the corresponding release event is received (indicating that the client can reuse that buffer). Since this driver is single-buffered, we need to wait until the compositor releases that buffer before we can begin drawing on it again.

If that section is being hit (should generate an LVGL warning), it would imply that the compositor hasn't released the buffer back to the client by the time it was looking to draw the new frame (LVGL's flush call).

I'll have to think about a proper fix for this. I think the easiest would probably be to go double-buffered, and only commit a single surface/buffer at a time...

simplejack-src, 2022-06-29:
Further thinking on it, a proper fix (which I somewhat lament, as I've been just adding bandage fixes until I can do a full refactor) would be to replace the simple linear buffer allocator with a fixed-size free list and dynamically allocate backing buffers for LVGL to flush into. This keeps LVGL happy (not blocking it's flush routine), would only drop frames on memory starvation, and stays responsive (if flushes occur too often, frame-limiting could also be added).

Unfortunately, as this will be a decently substantial change it'll be a few weeks until I get to it (anyone else wanting to tackle it sooner though is always welcome :)).

closed 2023-07-06 by symfund