lvgl/lv_drivers #235: Wrong DRM_FOURCC for 32 bit?

johado, 2022-08-24:
I'm trying to get an lvgl application running on an embedded linux device that uses the DRM framework and and a MIPI DSI panel using the panel-simple driver.
The current code fails select a plane that is compatible since the closest supported format is XR24 and not AR24, unless i do the following change in drm.c

 #if LV_COLOR_DEPTH == 32
-#define DRM_FOURCC DRM_FORMAT_ARGB8888
+#define DRM_FOURCC DRM_FORMAT_XRGB8888
 #elif LV_COLOR_DEPTH == 16

Is that the right thing to do or should the 32 bit format be configurable?
(On another device the same application would run when configured to use FBDEV or SDL)

kisvegabor, 2022-08-28:
> Is that the right thing to do or should the 32 bit format be configurable?

I'd make `DRM_FORMAT_XRGB8888` the default and add config to use the ARGB version. 

An other option would be to try `DRM_FORMAT_XRGB8888` and if it's  not got for `DRM_FORMAT_XRGB8888` as a fallback.

superna9999, 2022-12-31:
@kisvegabor yes argb should be a fallback of xrgb

closed 2025-05-19 by stale[bot]