lvgl/lv_drivers #241: Difference in SDL & SDL GPU driver with styling an imagebutton

JefkeB, 2022-10-30:
Hi,

I'm evaluating the SDL driver with the lv_port_pc_eclipse simulator.

```
void test()
{
  // setup styles
  lv_style_init(&pageR_Button);
  lv_style_set_img_recolor_opa(&pageR_Button, LV_OPA_COVER);
  lv_style_set_img_recolor(&pageR_Button, lv_palette_main(LV_PALETTE_GREY));

  lv_style_init(&pageR_ButtonPressed);
  lv_style_set_img_recolor(&pageR_ButtonPressed, lv_palette_main(LV_PALETTE_LIME));

  lv_obj_set_flex_flow(lv_scr_act(), LV_FLEX_FLOW_COLUMN);
  lv_obj_set_flex_align(lv_scr_act(), LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);

  // button
  LV_IMG_DECLARE(Home_48);
  lv_obj_t * imgButton = lv_imgbtn_create(lv_scr_act());

  lv_obj_add_flag(imgButton, LV_OBJ_FLAG_CHECKABLE);

  lv_imgbtn_set_src(imgButton, LV_IMGBTN_STATE_RELEASED, &Home_48, NULL, NULL);
  lv_obj_add_style(imgButton, &pageR_Button, LV_IMGBTN_STATE_RELEASED);

  lv_imgbtn_set_src(imgButton, LV_IMGBTN_STATE_CHECKED_RELEASED, &Home_48, NULL, NULL);
  lv_obj_add_style(imgButton, &pageR_ButtonPressed, LV_IMGBTN_STATE_CHECKED_RELEASED);
}
```

When running this in the simulator the working is as expected.
default after starting application, color is grey as expected
[image]

after pressing (checked style ), color is lime as expected
[image]

When the GPU is enabled
```
in lv_conf.h change
#define LV_USE_GPU_SDL 0
to
#define LV_USE_GPU_SDL 1
and
in lv_drv_conf change
# define USE_SDL 1		to 	# define USE_SDL 0
# define USE_SDL_GPU 0 	to 	# define USE_SDL_GPU 1
```

default after starting application Wrong color should be grey
[image]
 

after pressing (checked style )	Wrong color should be lime
[image]
 

Is this a problem in the GPU code or some where else ?
The same problem is visible when running on a raspberry pi (the intended target).

The complete project is at https://github.com/JefkeB/lv_port_pc_eclipse

kisvegabor, 2022-11-01:
cc @mariotaku

JefkeB, 2022-11-17:
Hi,

Been looking in this some more.
When using a white source image it seems to work almost as expected. 
Colors are still a bit off compared to the color without gpu.

As far as I understand it :
The reason must be in the sdl gpu drawing functions.
It seems the problem is in the function static void apply_recolor_opa(SDL_Texture * texture, const lv_draw_img_dsc_t * draw_dsc) in https://github.com/lvgl/lvgl/blob/release/v8.3/src/draw/sdl/lv_draw_sdl_img.c.

The sdl function used is SDL_SetTextureColorMod to do the recolor
Looking @ https://wiki.libsdl.org/SDL_SetTextureColorMod 
call returns 0 so is supported by renderer
The formula 'srcC = srcC * (color / 255)' is used
My image is black, so srcC = 0, the result will always be 0 so no color change ??

Am I correct or is the something else.
Is it possible to use the 'default' lvgl recoloring from https://github.com/lvgl/lvgl/blob/release/v8.3/src/draw/sw/lv_draw_sw_img.c
in combination with the gpu ?

Thanks

JefkeB, 2022-11-17:
Hi,

On PC (linux) and on embedded (Raspberry CM3+)  SDL2-2.0.10

We compiled against SDL version 2.0.10 ...
But we are linking against SDL version 2.0.10

I saw a comment in lv_conf.h

    /*Custom blend mode for mask drawing, disable if you need to link with older SDL2 lib*/
    #define LV_GPU_SDL_CUSTOM_BLEND_MODE (SDL_VERSION_ATLEAST(2, 0, 6))

But ignored it because my version used is newer,
did I do wrong ?

mariotaku, 2022-11-17:
This is strange as it should be enabling by default. Did you try adding breakpoint in blending code?

JefkeB, 2022-11-17:
can you point me to a location in the code to place a breakpoint ?

mariotaku, 2022-11-17:
https://github.com/lvgl/lvgl/blob/6825d4bd1d3d2d14f0f605d0418324ba3df4f9ee/src/draw/sdl/lv_draw_sdl_img.c#L408

Maybe I missed somewhere while drawing image button...

JefkeB, 2022-11-17:
Hi,
nope no stopping in there for imgbutton and button on the screen, not even the function is called.
Must be a problem on my side ?
In the sdl  mode without gpu enabled though all is working as expected.

JefkeB, 2022-11-17:
anything else to enable besides the 2 defines in the lvgl config file ?

JefkeB, 2023-04-20:
For me not stale ...
Nobody can help me ?

Want to have the gpu working cause it gives me much better performance on an embedded rPi system using sdl.
The direct framebuffer shows loads of tearing during changes on the screen

J

mariotaku, 2023-04-20:
@JefkeB Is Home_48 an icon with deep blue color? Due to [SDL_SetTextureColorMod](https://wiki.libsdl.org/SDL_SetTextureColorMod)'s limitation I think it's better to provide a white image.

JefkeB, 2023-04-20:
Hi,

thanks for your reply.
Changed the images from black to white
Now the colors are as expected
Never expected it would be this simple

So solved for me

Thanks again for the help

J

closed 2023-04-20 by JefkeB