lvgl/lv_drivers #255: win32drv has no support for multiple displays

wrgallo, 2022-11-30:
Currently the `win32drv` only support a single display, looking at the source-code there are lots of static variables to control that specific display window.
```C
static HWND g_window_handle = NULL;
static HDC g_buffer_dc_handle = NULL;
static UINT32* g_pixel_buffer = NULL;
static SIZE_T g_pixel_buffer_size = 0;
static lv_disp_t* g_display = NULL;
static bool volatile g_display_refreshing = false;
static bool volatile g_mouse_pressed = false;
static LPARAM volatile g_mouse_value = 0;
static bool volatile g_mousewheel_pressed = false;
static int16_t volatile g_mousewheel_value = 0;
static bool volatile g_keyboard_pressed = false;
static WPARAM volatile g_keyboard_value = 0;
```

Also there are 3 global variables to windows drivers input devices (mouse , keyboard and mousewheel).
```C
EXTERN_C lv_indev_t* lv_win32_pointer_device_object;
EXTERN_C lv_indev_t* lv_win32_keypad_device_object;
EXTERN_C lv_indev_t* lv_win32_encoder_device_object;
```
But, each input driver needs to be assigned to a display which makes those global variable useless for other displays.

I am already working on a pull request to fix this issue.

kisvegabor, 2022-12-01:
cc @MouriNaruto

MouriNaruto, 2022-12-01:
@kisvegabor Got it. I will try to add this feature to win32drv.

Kenji Mouri

MouriNaruto, 2022-12-09:
@wrgallo @kisvegabor 

I have just added the multiple display support and preparing the PR.

[image]

Kenji Mouri

closed 2022-12-15 by kisvegabor