lvgl/lv_drivers #264: LVGL Wayland driver works with Wayland 9.0.0 but not 10.0.2

pclass-sensonix, 2023-01-10:
I'm using the latest version of LVGL from the master branch and the latest Wayland driver from this lv_drivers repository on a small Embedded Linux device. When I run my application again Weston 9.0.0 on my device, it works great - however, when I install Weston 10.0.2, I get a segmentation fault in the Wayland driver at the **lv_obj_set_style_bg_color** line. Chromium works against both Wayland versions on the same hardware. The application is communicating with Wayland because I can see output by exporting the WAYLAND_DEBUG=1 environment variable.

Here is my simple example application:

```
#include <lvgl/lvgl.h>
#include <lv_drivers/wayland/wayland.h>
#include <unistd.h>
#include <pthread.h>
#include <time.h>
#include <sys/time.h>
#include "ui.h"

#define H_RES (800)
#define V_RES (480)

static void * tick_thread(void * data)
{
    (void) data;
    while(true) {
        usleep(5 * 1000);
        lv_tick_inc(5);
    }
}

void hal_init(void)
{

    lv_init();

    lv_wayland_init();

    /* Create a display */
    lv_disp_t * disp = lv_wayland_create_window(H_RES, V_RES, "Window Title", NULL);

    lv_wayland_window_set_fullscreen(disp, true);

}

int main(void)
{
    /*LittlevGL init*/
    lv_init();

    hal_init();

    /*Create a Demo*/
    //ui_init();

    /*Change the active screen's background color*/
    lv_obj_set_style_bg_color(lv_scr_act(), lv_color_hex(0x003a57), LV_PART_MAIN);

    /*Handle LitlevGL tasks (tickless mode)*/
    while(1) {
        lv_timer_handler();
        usleep(5000);
    }

    return 0;
}
```

Here is the output from Weston with WAYLAND_DEBUG=1 exported:

```
[1149019.686] wl_display@1.get_registry(new id wl_registry@2)
[1149020.973]  -> wl_registry@2.global(1, "wl_compositor", 4)
[1149021.402]  -> wl_registry@2.global(2, "wl_subcompositor", 1)
[1149022.652]  -> wl_registry@2.global(3, "wp_viewporter", 1)
[1149023.052]  -> wl_registry@2.global(4, "zxdg_output_manager_v1", 2)
[1149024.408]  -> wl_registry@2.global(5, "wp_presentation", 1)
[1149024.816]  -> wl_registry@2.global(6, "zwp_relative_pointer_manager_v1", 1)
[1149026.034]  -> wl_registry@2.global(7, "zwp_pointer_constraints_v1", 1)
[1149026.452]  -> wl_registry@2.global(8, "zwp_input_timestamps_manager_v1", 1)
[1149026.833]  -> wl_registry@2.global(9, "wl_data_device_manager", 3)
[1149028.123]  -> wl_registry@2.global(10, "wl_shm", 1)
[1149028.530]  -> wl_registry@2.global(11, "weston_touch_calibration", 1)
[1149029.826]  -> wl_registry@2.global(12, "wl_seat", 7)
[1149030.250]  -> wl_registry@2.global(13, "weston_content_protection", 1)
[1149031.518]  -> wl_registry@2.global(14, "wl_output", 3)
[1149031.927]  -> wl_registry@2.global(15, "zwp_input_panel_v1", 1)
[1149032.305]  -> wl_registry@2.global(16, "zwp_input_method_v1", 1)
[1149033.637]  -> wl_registry@2.global(17, "zwp_text_input_manager_v1", 1)
[1149034.039]  -> wl_registry@2.global(18, "xdg_wm_base", 3)
[1149035.300]  -> wl_registry@2.global(19, "weston_desktop_shell", 1)
[1149036.398]  -> wl_registry@2.global(20, "weston_screenshooter", 1)
[1149071.369] wl_registry@2.bind(1, "wl_compositor", 1, new id [unknown]@3)
[1149072.943] wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@4)
[1149074.472] wl_registry@2.bind(10, "wl_shm", 1, new id [unknown]@5)
[1149076.476]  -> wl_shm@5.format(0)
[1149076.714]  -> wl_shm@5.format(1)
[1149077.745]  -> wl_shm@5.format(909199186)
[1149077.972]  -> wl_shm@5.format(875709016)
[1149078.171]  -> wl_shm@5.format(875708993)
[1149078.356]  -> wl_shm@5.format(875714642)
[1149078.541]  -> wl_shm@5.format(875708754)
[1149080.381]  -> wl_shm@5.format(875714626)
[1149080.702]  -> wl_shm@5.format(875708738)
[1149080.902]  -> wl_shm@5.format(808669784)
[1149081.940]  -> wl_shm@5.format(808669761)
[1149082.183]  -> wl_shm@5.format(808665688)
[1149082.379]  -> wl_shm@5.format(808665665)
[1149082.591] wl_shm@5.create_pool(new id wl_shm_pool@6, fd 37, 4096)
[1149084.185] wl_shm_pool@6.resize(8832)
[1149084.567] wl_shm_pool@6.resize(18624)
[1149085.719] wl_registry@2.bind(12, "wl_seat", 1, new id [unknown]@7)
[1149086.404]  -> wl_seat@7.capabilities(6)
[1149086.646] wl_display@1.sync(new id wl_callback@8)
[1149087.813]  -> wl_callback@8.done(1)
[1149088.056]  -> wl_display@1.delete_id(8)
```

And here is the output of my application ( including the segmentation fault ):

```
>> ./splash 
[1149016.050]  -> wl_display@1.get_registry(new id wl_registry@2)
[1149038.173] wl_registry@2.global(1, "wl_compositor", 4)
[1149039.897]  -> wl_registry@2.bind(1, "wl_compositor", 1, new id [unknown]@3)
[1149041.212] wl_registry@2.global(2, "wl_subcompositor", 1)
[1149041.730]  -> wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@4)
[1149043.690] wl_registry@2.global(3, "wp_viewporter", 1)
[1149044.106] wl_registry@2.global(4, "zxdg_output_manager_v1", 2)
[1149044.490] wl_registry@2.global(5, "wp_presentation", 1)
[1149046.146] wl_registry@2.global(6, "zwp_relative_pointer_manager_v1", 1)
[1149048.382] wl_registry@2.global(7, "zwp_pointer_constraints_v1", 1)
[1149048.794] wl_registry@2.global(8, "zwp_input_timestamps_manager_v1", 1)
[1149049.171] wl_registry@2.global(9, "wl_data_device_manager", 3)
[1149050.433] wl_registry@2.global(10, "wl_shm", 1)
[1149050.842]  -> wl_registry@2.bind(10, "wl_shm", 1, new id [unknown]@5)
[1149052.532]  -> wl_shm@5.create_pool(new id wl_shm_pool@6, fd 5, 4096)
[1149055.740]  -> wl_shm_pool@6.resize(8832)
[1149058.772]  -> wl_shm_pool@6.resize(18624)
[1149060.192] wl_registry@2.global(11, "weston_touch_calibration", 1)
[1149061.548] wl_registry@2.global(12, "wl_seat", 7)
[1149061.960]  -> wl_registry@2.bind(12, "wl_seat", 1, new id [unknown]@7)
[1149063.731] wl_registry@2.global(13, "weston_content_protection", 1)
[1149064.167] wl_registry@2.global(14, "wl_output", 3)
[1149065.456] wl_registry@2.global(15, "zwp_input_panel_v1", 1)
[1149065.846] wl_registry@2.global(16, "zwp_input_method_v1", 1)
[1149066.220] wl_registry@2.global(17, "zwp_text_input_manager_v1", 1)
[1149067.445] wl_registry@2.global(18, "xdg_wm_base", 3)
[1149067.831] wl_registry@2.global(19, "weston_desktop_shell", 1)
[1149069.502] wl_registry@2.global(20, "weston_screenshooter", 1)
[1149070.811]  -> wl_display@1.sync(new id wl_callback@8)
[1149088.531] wl_display@1.delete_id(8)
[1149089.677] wl_shm@5.format(0)
[1149089.919] wl_shm@5.format(1)
[1149090.106] wl_shm@5.format(909199186)
[1149090.292] wl_shm@5.format(875709016)
[1149090.469] wl_shm@5.format(875708993)
[1149090.646] wl_shm@5.format(875714642)
[1149091.688] wl_shm@5.format(875708754)
[1149091.904] wl_shm@5.format(875714626)
[1149092.093] wl_shm@5.format(875708738)
[1149092.278] wl_shm@5.format(808669784)
[1149092.459] wl_shm@5.format(808669761)
[1149093.558] wl_shm@5.format(808665688)
[1149093.770] wl_shm@5.format(808665665)
[1149093.956] wl_seat@7.capabilities(6)
[1149094.159]  -> wl_seat@7.get_keyboard(new id wl_keyboard@9)
[1149095.300]  -> wl_seat@7.get_touch(new id wl_touch@10)
[1149095.588] wl_callback@8.done(1)
[1149096.080]  -> wl_compositor@3.create_surface(new id wl_surface@8)
[1149098.150]  -> wl_surface@8.destroy()
Segmentation fault
```

I do not understand what is failing and why it would work with Weston 9.0.0 but not 10.0.2. I'd greatly appreciate any help!

symfund, 2023-01-11:
With Weston 10.0.2, you should use XDG Shell instead of WL SHELL, the WL SHELL is deprecated!

In lv_drv_conf.h,  enable the macro 'LV_WAYLAND_XDG_SHELL'

pclass-sensonix, 2023-01-23:
Thank you for the answer. I do believe that is the problem. I am now trying to enable XDG Shell in Yocto since the last official version of lv_drivers for Bitbake was version 8.3.0. It seems to have trouble compiling with XDG Shell enabled.

pclass-sensonix, 2023-02-27:
Hi @symfund - Have you been able to use the Wayland LVGL drivers with Weston 11.0.1? I was able to build this with Yocto and deploy to my ARM32 board with Weston 11.0.1 running. Chromium is able to render on  using Weston, however the lvl driver does not seem to be able to render onto the display. It does not display any errors and does not set fault, it just does not render. I can see from the Weston side debugging that LVGL is able to communicate with it.

Here is the Wayland debugging I have captured from my LVGL application side ( and the source code is below ):

 ```
>> export WAYLAND_DISPLAY=wayland-1                                                                                                                                       
 >> export WAYLAND_DEBUG=1                                                                                                                                               
 >> export XDG_RUNTIME_DIR=/tmp    
>> splash                                                                                                                                                               
[2832542.019]  -> wl_display@1.get_registry(new id wl_registry@2)
[2832551.152] wl_registry@2.global(1, "wl_compositor", 5)
[2832551.454]  -> wl_registry@2.bind(1, "wl_compositor", 1, new id [unknown]@3)
[2832551.589] wl_registry@2.global(2, "wl_subcompositor", 1)
[2832551.696]  -> wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@4)
[2832551.802] wl_registry@2.global(3, "wp_viewporter", 1)
[2832551.891] wl_registry@2.global(4, "zxdg_output_manager_v1", 2)
[2832551.976] wl_registry@2.global(5, "wp_presentation", 1)
[2832552.139] wl_registry@2.global(6, "wp_single_pixel_buffer_manager_v1", 1)
[2832552.256] wl_registry@2.global(7, "zwp_relative_pointer_manager_v1", 1)
[2832552.347] wl_registry@2.global(8, "zwp_pointer_constraints_v1", 1)
[2832552.431] wl_registry@2.global(9, "zwp_input_timestamps_manager_v1", 1)
[2832552.563] wl_registry@2.global(10, "wl_data_device_manager", 3)
[2832552.665] wl_registry@2.global(11, "wl_shm", 1)
[2832552.761]  -> wl_registry@2.bind(11, "wl_shm", 1, new id [unknown]@5)
[2832553.136]  -> wl_shm@5.create_pool(new id wl_shm_pool@6, fd 5, 4096)
[2832555.011]  -> wl_shm_pool@6.resize(8832)
[2832555.568]  -> wl_shm_pool@6.resize(18624)
[2832555.900] wl_registry@2.global(12, "wl_seat", 7)
[2832556.051]  -> wl_registry@2.bind(12, "wl_seat", 1, new id [unknown]@7)
[2832556.248] wl_registry@2.global(13, "weston_content_protection", 1)
[2832556.348] wl_registry@2.global(14, "wl_output", 3)
[2832556.437] wl_registry@2.global(15, "zwp_input_panel_v1", 1)
[2832556.523] wl_registry@2.global(16, "zwp_input_method_v1", 1)
[2832556.606] wl_registry@2.global(17, "zwp_text_input_manager_v1", 1)
[2832556.688] wl_registry@2.global(18, "xdg_wm_base", 5)
[2832556.781]  -> wl_registry@2.bind(18, "xdg_wm_base", 5, new id [unknown]@8)
[2832556.882] wl_registry@2.global(19, "weston_desktop_shell", 1)
[2832556.968] wl_registry@2.global(20, "weston_screenshooter", 1)
[2832557.068]  -> wl_display@1.sync(new id wl_callback@9)
[2832560.161] wl_display@1.delete_id(9)
[2832560.357] wl_shm@5.format(0)
[2832560.452] wl_shm@5.format(1)
[2832560.534] wl_shm@5.format(909199186)
[2832560.612] wl_shm@5.format(875709016)
[2832560.685] wl_shm@5.format(875708993)
[2832560.759] wl_shm@5.format(875714642)
[2832560.832] wl_shm@5.format(875708754)
[2832560.910] wl_shm@5.format(875714626)
[2832560.985] wl_shm@5.format(875708738)
[2832561.060] wl_shm@5.format(808669784)
[2832561.135] wl_shm@5.format(808669761)
[2832561.209] wl_shm@5.format(808665688)
[2832561.284] wl_shm@5.format(808665665)
[2832561.357] wl_seat@7.capabilities(6)
[2832561.440]  -> wl_seat@7.get_keyboard(new id wl_keyboard@10)
[2832561.541]  -> wl_seat@7.get_touch(new id wl_touch@11)
[2832561.631] wl_callback@9.done(5)
[2832561.960]  -> wl_compositor@3.create_surface(new id wl_surface@9)
[2832562.193]  -> xdg_wm_base@8.get_xdg_surface(new id xdg_surface@12, wl_surface@9)
[2832562.319]  -> xdg_surface@12.get_toplevel(new id xdg_toplevel@13)
[2832562.487]  -> xdg_toplevel@13.set_title("Splash Screen")
[2832562.618]  -> xdg_toplevel@13.set_app_id("Splash Screen")
[2832562.704]  -> wl_surface@9.commit()
[2832562.793]  -> wl_compositor@3.create_surface(new id wl_surface@14)
[2832562.889]  -> wl_compositor@3.create_surface(new id wl_surface@15)
[2832562.991]  -> wl_compositor@3.create_surface(new id wl_surface@16)
[2832563.085]  -> wl_compositor@3.create_surface(new id wl_surface@17)
[2832563.183]  -> wl_compositor@3.create_surface(new id wl_surface@18)
[2832563.278]  -> wl_compositor@3.create_surface(new id wl_surface@19)
[2832563.372]  -> wl_compositor@3.create_surface(new id wl_surface@20)
[... 73 lines cut ...]
int main(void)
{

    lv_init();

    lv_wayland_init();

    /* Create a display */
    lv_disp_t * disp = lv_wayland_create_window(H_RES, V_RES, "Splash Screen", NULL);

    lv_wayland_window_set_fullscreen(disp, false);

    /*Create a Demo*/
    ui_init();

    printf("Done with UI Init\n");

    while(1)
	{
		pause();
	}

    return 0;
}
```

skmodi, 2024-06-27:
I'm facing similar Segmentation fault issue. Do we have any fix for this issue ?