lvgl/lv_drivers #271: drm mode fails to draw on H616 processor

danielwalton, 2023-02-16:
I can't get drm mode to work within lvgl on my SBC (Orange Pi Zero 2).

This example works fine:
[https://github.com/dvdhrm/docs/blob/master/drm-howto/modeset-atomic.c](https://github.com/dvdhrm/docs/blob/master/drm-howto/modeset-atomic.c)

However only fbdev seems to work with lvgl. 

Here is how I'm initializing it. The screen flashes, then I'm back in the console.

`    drm_init();
    lv_disp_draw_buf_init(&disp_buf, buf, buf2, DISP_BUF_SIZE);

    /*Initialize and register a display driver*/
    disp_drv.draw_buf = &disp_buf;
    disp_drv.flush_cb = drm_flush;
    disp_drv.hor_res = 800;
    disp_drv.ver_res = 480;
    disp_drv.full_refresh = true;
    lv_disp_drv_register(&disp_drv);`

kisvegabor, 2023-02-17:
I cc @superna9999 and @leiflm who contributed the DRM driver.
Could you take lake look at this issue?

leiflm, 2023-02-17:
I am sorry. I am currently not using lvgl.

Maybe using [DRM environment variables](https://docs.mesa3d.org/envvars.html) to debug will help you to find precisely what goes wrong.

@danielwalton In case you are addressing the wrong device, you can use the environment variable `DRM_CARD` to specify a different one.

Good luck!

closed 2025-05-19 by stale[bot]