lvgl/lv_drivers #279: Fullscreen window mode in Wayland

symfund, 2023-06-02:
Need a LVGL demo about fullscreen window mode in Wayland.

symfund, 2023-07-06:
[image]
I have improved and corrected the fullscreen support in Wayland environment.

symfund, 2023-07-09:
[image]

kisvegabor, 2023-07-10:
Looks great! 

For LVGL v9 we will host the driver in the main LVGL repo. Would you interested in sending a PR with a Wayland driver there? It can be similar to the [SDL driver](https://github.com/lvgl/lvgl/blob/master/src/dev/sdl/lv_sdl_window.c).

arilink, 2024-05-16:
What platform is this demo running on? I am also trying to run LVGL on arm Linu

hyyoxhk, 2024-07-15:
> What platform is this demo running on? I am also trying to run LVGL on arm Linux

It look like in Weston (A Wayland compositor)