lvgl/lv_drivers #283: how to support(register) multiple input devices?

incity, 2023-06-27:
There is no interface to register devices with  libinput_read_state.(3 parameters)

[image]
[image]
[image]
[image]

kisvegabor, 2023-06-29:
@Johennes could you comment on this?

Johennes, 2023-06-30:
@incity you need to define a custom callback and inside of it call `libinput_read_state` with the state extracted from the driver's `user_data` field. See e.g. https://gitlab.com/cherrypicker/unl0kr/-/blob/master/indev.c#L357.