lvgl/lv_drivers #287: Add lvgl v9 support

nicusorcitu, 2023-08-25:
lv_drivers does not work anylonger with lvgl v9 (due to display rework).
I am trying to build with wayland enabled and got a lot of errors, so I am opening this Issue to track it down.

Is there any roadmap for the fix for lvgl v9 compatibility?

Thanks.
Nicu.

```
In file included from /home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.h:29,
                 from /home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:9:
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lvgl/lvgl.h:126:2: warning: #warning "You are using the development version of LVGL which is not stable at this moment. For production use the release/v8.3 branch. To silence this warning add #define LV_USE_DEV_VERSION to lv_conf.h" [-Wcpp]
  126 | #warning "You are using the development version of LVGL which is not stable at this moment. For production use the release/v8.3 branch. To silence this warning add #define LV_USE_DEV_VERSION to lv_conf.h"
      |  ^~~~~~~
In file included from /home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:9:
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.h:33:2: error: #error "Support for user data is required by wayland driver. Set LV_USE_USER_DATA to 1 in lv_conf.h"
   33 | #error "Support for user data is required by wayland driver. Set LV_USE_USER_DATA to 1 in lv_conf.h"
      |  ^~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:196:5: error: unknown type name ‘lv_disp_drv_t’
  196 |     lv_disp_drv_t lv_disp_drv;
      |     ^~~~~~~~~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:197:5: error: unknown type name ‘lv_disp_draw_buf_t’
  197 |     lv_disp_draw_buf_t lv_disp_draw_buf;
      |     ^~~~~~~~~~~~~~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:200:5: error: unknown type name ‘lv_indev_drv_t’
  200 |     lv_indev_drv_t lv_indev_drv_pointer;
      |     ^~~~~~~~~~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:203:5: error: unknown type name ‘lv_indev_drv_t’
  203 |     lv_indev_drv_t lv_indev_drv_pointeraxis;
      |     ^~~~~~~~~~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:206:5: error: unknown type name ‘lv_indev_drv_t’
  206 |     lv_indev_drv_t lv_indev_drv_touch;
      |     ^~~~~~~~~~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:209:5: error: unknown type name ‘lv_indev_drv_t’
  209 |     lv_indev_drv_t lv_indev_drv_keyboard;
      |     ^~~~~~~~~~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c: In function ‘shm_format’:
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:268:51: warning: unused parameter ‘wl_shm’ [-Wunused-parameter]
  268 | static void shm_format(void *data, struct wl_shm *wl_shm, uint32_t format)
      |                                    ~~~~~~~~~~~~~~~^~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c: In function ‘pointer_handle_leave’:
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:424:65: warning: unused parameter ‘pointer’ [-Wunused-parameter]
  424 | static void pointer_handle_leave(void *data, struct wl_pointer *pointer,
      |                                              ~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:425:43: warning: unused parameter ‘serial’ [-Wunused-parameter]
  425 |                                  uint32_t serial, struct wl_surface *surface)
      |                                  ~~~~~~~~~^~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c: In function ‘pointer_handle_motion’:
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:439:16: warning: unused variable ‘max_y’ [-Wunused-variable]
  439 |     int max_x, max_y;
      |                ^~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:439:9: warning: unused variable ‘max_x’ [-Wunused-variable]
  439 |     int max_x, max_y;
      |         ^~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:435:66: warning: unused parameter ‘pointer’ [-Wunused-parameter]
  435 | static void pointer_handle_motion(void *data, struct wl_pointer *pointer,
      |                                               ~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:436:44: warning: unused parameter ‘time’ [-Wunused-parameter]
  436 |                                   uint32_t time, wl_fixed_t sx, wl_fixed_t sy)
      |                                   ~~~~~~~~~^~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c: In function ‘pointer_handle_button’:
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:466:9: warning: unused variable ‘pos_y’ [-Wunused-variable]
  466 |     int pos_y = app->pointer_obj->input.pointer.y;
      |         ^~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:465:9: warning: unused variable ‘pos_x’ [-Wunused-variable]
  465 |     int pos_x = app->pointer_obj->input.pointer.x;
      |         ^~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:450:66: warning: unused parameter ‘wl_pointer’ [-Wunused-parameter]
  450 | static void pointer_handle_button(void *data, struct wl_pointer *wl_pointer,
      |                                               ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
[... 321 lines cut ...]
 2383 |     window->lv_indev_drv_keyboard.type = LV_INDEV_TYPE_KEYPAD;
      |                                  ^
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:2384:34: error: request for member ‘read_cb’ in something not a structure or union
 2384 |     window->lv_indev_drv_keyboard.read_cb = _lv_wayland_keyboard_read;
      |                                  ^
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:2384:45: error: ‘_lv_wayland_keyboard_read’ undeclared (first use in this function)
 2384 |     window->lv_indev_drv_keyboard.read_cb = _lv_wayland_keyboard_read;
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:2385:34: error: request for member ‘disp’ in something not a structure or union
 2385 |     window->lv_indev_drv_keyboard.disp = window->lv_disp;
      |                                  ^
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:2386:31: warning: assignment to ‘lv_indev_t *’ {aka ‘struct _lv_indev_t *’} from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 2386 |     window->lv_indev_keyboard = lv_indev_drv_register(&window->lv_indev_drv_keyboard);
      |                               ^
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c: In function ‘lv_wayland_close_window’:
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:2401:33: error: dereferencing pointer to incomplete type ‘lv_disp_t’ {aka ‘struct _lv_disp_t’}
 2401 |     struct window *window = disp->driver->user_data;
      |                                 ^~
At top level:
/home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.c:258:21: warning: ‘_atoi’ defined but not used [-Wunused-function]
  258 | static unsigned int _atoi(const char ** str)
      |                     ^~~~~
make: *** [Makefile:43: /home/nxf68195/bitbucket/lvgl_mpu/lvgl_imx_demo/lv_drivers/wayland/wayland.o] Error 1
nxf68195@lw15016 ~/bitbucket/lvgl_mpu/lvgl_imx_demo (master)$
```

kisvegabor, 2023-08-25:
The solution would be to move the Wayland driver to the LVGL repository. 

@WallaceIT @simplejack-src could you help us with that?

The main changes of v9 (so far) are listed here: https://github.com/lvgl/lvgl/issues

nicusorcitu, 2023-09-13:
Hi, Any update with this issue, when it is going to be integrated in v9? Thanks.

kisvegabor, 2023-09-13:
Unfortunately, I don't have no experience with Wayland, so can't really help with that :slightly_frowning_face:

WallaceIT, 2023-09-13:
Hi,

I'm trying to fit some work on this feature on my schedule, but right now I cannot take a commitment.

I took a look at the v9 status and approach to displays and I think the Wayland backend can take a great advantage from them (for example, integrated "dirty rendering" can avoid copies of graphic data). However, such changes also require quite some time of code rework.

nicusorcitu, 2023-09-14:
Is it possible to elaborate a bit more with details what needs to be done?
I am thinking if someone else might find some room, I can bet the info will be much helpful.

WallaceIT, 2023-09-15:
The existing Wayland support shall be adapted to v9, doing the required API conversion and in the mean time aligning to the changes that can lead to performance improvement. As an example: v8 required the Wayland driver to copy the rendered content into its own buffer(s), while with v9 double buffering can be implemented without effort (see explanation of `LV_DISP_RENDER_MODE_DIRECT` [here](https://github.com/lvgl/lvgl/issues/4011)).

While the basics of Wayland are quite simple, resizing the application window is not. Moreover, client-side decorations also add a lot of complexity. I'd say buffer allocation/deallocation is around 90% of the complexity.

Last but not least, buffer allocation logic implemented by @simplejack-src is very flexible but not so straightforward  (at least for me).

kisvegabor, 2023-09-18:
Window resizing is jut a nice to have, absolutely not mandatory. 

I think the most important thing would be have a very basic Wayland driver in v9, which can be improved later.

nicusorcitu, 2024-05-16:
@anagrad

kisvegabor, 2024-05-20:
FYI, @daniel-rossier is working on a Wayland driver of LVGL v9.

nicusorcitu, 2024-05-23:
Hi @kisvegabor @daniel-rossier sorry to ask, do you have any estimation when will be done?

nicusorcitu, 2024-05-23:
Will the driver move from lvgl/lv_drivers/wayland to lvgl/lvgl/src/drivers/wayland?

etag4048, 2024-05-24:
@nicusorcitu Yes I believe it will at some point, but it needs some refactoring.

symfund, 2024-06-17:
Is there any progress weston/wayland and lvgl v9 support?

nicusorcitu, 2024-08-27:
@kisvegabor was wayland refactored and integrated in lvgl v9.2.0?

kisvegabor, 2024-08-29:
Yes, see [here](https://docs.lvgl.io/master/integration/driver/wayland.html)

nicusorcitu, 2024-08-29:
So we can close this issue, as well as the PR https://github.com/lvgl/lv_drivers/pull/313
Correct?

kisvegabor, 2024-08-30:
Yes. 
For future reference:
- The Walan driver is added to LVGL v9: https://github.com/lvgl/lvgl/tree/master/src/drivers/wayland
- `lv_drivers` won't be updated to LVGL v9 as nor the `lvgl` repos has built-in drivers.

closed 2024-08-30 by kisvegabor