lvgl/lv_drivers #288: Add FT81x support to display

ONLYA, 2023-08-31:
The [lvgl_esp32_drivers](https://github.com/lvgl/lvgl_esp32_drivers) repo contains FT81x support in addition to the current display support. 
Would someone fancy migrating the driver to this repo?
Or I could do that if I have time.

ONLYA, 2023-08-31:
I am considering porting the code here. 
FT81x chipset has both display and touch support. Where should I contain the code for it? `display` or `indev`? Or create a new folder like `sdl` named `ft81x`?

kisvegabor, 2023-08-31:
Could you consider rather adding it to the LVGL repository directly? For v9 (under development) we add the driver directly to the LVGL repo. See https://github.com/lvgl/lvgl/tree/master/src/dev

ONLYA, 2023-08-31:
I am using the v8.3 stable release for production so I would port the code to v8.3 first and then migrate to v9.
As I am not quite sure when the release date for v9 will be, I would adhere to v8.3 for now.

kisvegabor, 2023-08-31:
Note that in this repo is not really maintained. Only a few high level drivers are being fixed (SDL, Wayland, etc). 

However if adding the driver here is required step to migrate it to v9 later, I'm ok with that :slightly_smiling_face: