lvgl/lv_drivers #291: endev reverse mapping is broken

elgerg, 2023-09-03:
Hi,

After the introduction of code based on:
https://github.com/lvgl/lv_drivers/pull/251

The reverse functionality based on doing thing like:

```
#  define EVDEV_CALIBRATE         1               /*Scale and offset the touchscreen coordinates by using maximum and minimum values for each axis*/

#  if EVDEV_CALIBRATE
#    define EVDEV_HOR_MIN         2000               /*to invert axis swap EVDEV_XXX_MIN by EVDEV_XXX_MAX*/
#    define EVDEV_HOR_MAX      0               /*"evtest" Linux tool can help to get the correct calibraion values>*/
#    define EVDEV_VER_MIN        2000
#    define EVDEV_VER_MAX      0
#  endif  /*EVDEV_CALIBRATE*/
#endif  /*USE_EVDEV*/
```

Does not work anymore.

I have a screen that I have rotated 180 defgees and when using the current lib with:

```
#if EVDEV_CALIBRATE
    evdev_root_x = LV_CLAMP(EVDEV_HOR_MIN, evdev_root_x, EVDEV_HOR_MAX);
    evdev_root_y = LV_CLAMP(EVDEV_VER_MIN, evdev_root_y, EVDEV_VER_MAX);

    data->point.x = map(evdev_root_x, EVDEV_HOR_MIN, EVDEV_HOR_MAX, 0, drv->disp->driver->hor_res);
    data->point.y = map(evdev_root_y, EVDEV_VER_MIN, EVDEV_VER_MAX, 0, drv->disp->driver->ver_res);
#else
```

Then all touch points seem to register as 0.

Using the following:

```
#if EVDEV_CALIBRATE
    //evdev_root_x = LV_CLAMP(EVDEV_HOR_MIN, evdev_root_x, EVDEV_HOR_MAX);
    //evdev_root_y = LV_CLAMP(EVDEV_VER_MIN, evdev_root_y, EVDEV_VER_MAX);

    data->point.x = map(evdev_root_x, EVDEV_HOR_MIN, EVDEV_HOR_MAX, 0, drv->disp->driver->hor_res);
    data->point.y = map(evdev_root_y, EVDEV_VER_MIN, EVDEV_VER_MAX, 0, drv->disp->driver->ver_res);
#else
```

It all works as expected. 

Related forum post:
https://forum.lvgl.io/t/how-to-invert-x-and-y-axises-not-swap-them/12866/2

Any chance this change can check if a mouse is involves as apposed to a tcouch screen?

Thanks!

kisvegabor, 2023-09-04:
With `lv_map` `LV_CLAMP` is not required indeed. Could you send a Pull request?

elgerg, 2023-09-04:
PR created but it's my first one. Let me know if you need anything correcting.

Thanks