lvgl/lv_drivers #293: Getting mmap fail EINV with RPi4

moroslantia, 2023-09-13:
I try to use RPi4 with DRM dispplay driver

```
  drm_init();

    /*A small buffer for LittlevGL to draw the screen's content*/
    static lv_color_t buf[DISP_BUF_SIZE],buf1[DISP_BUF_SIZE];

    /*Initialize a descriptor for the buffer*/
    static lv_disp_draw_buf_t disp_buf;
    lv_disp_draw_buf_init(&disp_buf, buf, buf1, DISP_BUF_SIZE);

    /*Initialize and register a display driver*/
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);

    disp_drv.draw_buf   = &disp_buf;
    disp_drv.flush_cb   = drm_flush;
    disp_drv.wait_cb    = drm_wait_vsync;
    disp_drv.hor_res    = 800;
    disp_drv.ver_res    = 480;
    lv_disp_drv_register(&disp_drv);
```
When starting my program it fails with 

```
drm: Found plane_id: 86 connector_id: 32 crtc_id: 96
drm: 2560x1440 (550mm X 310mm) pixel format AR24
error: mmap fail (errno 22)
error: DRM buffer allocation failed

```

Any idea what's wrong?

kisvegabor, 2023-09-13:
errno 22 means "invalid argument" in `mmap`. Can you print the arguments passed to `mmap`?

moroslantia, 2023-09-14:
```
	// drm.c 644:649
	/* perform actual memory mapping */
	buf->map = mmap(0, creq.size, PROT_READ | PROT_WRITE, MAP_SHARED, drm_dev.fd, mreq.offset);
	if (buf->map == MAP_FAILED) {
		err("mmap fail (errno=%i), creq.size=%" PRIu64 ", drm_dev.fd=%i, mreq.offset=%" PRIx64, errno, creq.size, drm_dev.fd, mreq.offset);
		return -1;
	}

```

Running the demo gives:
```
drm: Found plane_id: 86 connector_id: 32 crtc_id: 96
drm: 2560x1440 (550mm X 310mm) pixel format AR24
error: mmap fail (errno=22), creq.size=14745600, drm_dev.fd=3, mreq.offset=118a7c000
error: DRM buffer allocation failed
Segmentation fault
```

kisvegabor, 2023-09-14:
These values makes sens to me. I suspect that it's  a Raspberry issue. :slightly_frowning_face: