lvgl/lv_drivers #302: [RK3568 waylay drivce] interface 'xdg_toplevel' has no event 3

eYoung8475, 2023-11-16:
Using the rk3568 development board, driven by wayland （version 1.22.0）, with a prompt of "interface 'xdg_toplevel' has no event 3" and no GUI displayed。[
[image]
](url)

arilink, 2024-05-16:
请问这个有进展吗

arilink, 2024-05-16:
Is there any progress on this issue? I am also trying to run LVGL on rk3568

eYoung8475, 2024-05-17:
No, contacting Rockchip is currently unable to resolve the issue. It may be necessary to replace the DRM or x11 display

kisvegabor, 2024-05-20:
I would be also happy contact Rockchip. Hopefully we could collaborate somehow to make it easier to get started with Rockchip and LVGL. Could you write an introduction email to me and you contact person at Rockchip?

kisvegabor, 2024-05-26:
Thank you! 
We are onboarding a some new chip vendors, but I'll contact them when we will have a little bit free capacity.

eYoung8475, 2024-05-27:
Just tested it, The wayland on rk3568 is now working properly.

eYoung8475, 2024-05-27:
Test on patch 110089d in the master branch。

closed 2024-06-04 by kisvegabor