lvgl/lv_drivers #303: How to implement esp-idf LV_DRV_DISP_SPI_WR_ARRAY.

Cjiio, 2023-11-20:
How to implement the initialization of spi in esp-idf v5.1.2 and the implementation of LV_DRV_DISP_SPI_WR_ARRAY. 
The screen I use is sharp memory lcd 400x240 pixels. I try to integrate it into lvglv8.3.10. 
The results were terrible. Cannot be displayed normally. I tried configuring it, but all I got was gibberish. 
I need help on how to write spi interrupts and device initialization and send spi data under esp-idf.
I've been struggling with this for three weeks and sincerely look forward to your help.

Cjiio, 2023-11-21:
> Hi,
> 
> Exactly which display controller do you use?

sharp memory lcd
ls027b7dh01

kisvegabor, 2023-11-23:
Unfortunately, I have no experience with that controller. I suggest asking about in in our [Forum](https://forum.lvgl.io/).