lvgl/lv_drivers #305: There is no lv_disp_drv_t data type in source code

pouyaimani, 2024-01-15:
I can not find any lv_disp_drv_t data type in source code, however it is used in source code

kisvegabor, 2024-01-15:
I's removed in LVGL v9. Which LVGL version do you use?

pouyaimani, 2024-01-15:
> I's removed in LVGL v9. Which LVGL version do you use?

I’m using master branch.

pouyaimani, 2024-01-15:
> I's removed in LVGL v9. Which LVGL version do you use?

So how should I use lvgl v9 with lv_driver project?

kisvegabor, 2024-01-15:
LVGL v9 has it's own [built in drivers](https://github.com/lvgl/lvgl/tree/master/src/dev) and `lv_drivers` won't be used in the future. 

Which driver would you like to use?

muteryx, 2024-11-27:
> LVGL v9 has it's own [built in drivers](https://github.com/lvgl/lvgl/tree/master/src/dev) and `lv_drivers` won't be used in the future.
> 
> Which driver would you like to use?

I have the same problem。I am using Linux Framebuffer。

kisvegabor, 2024-12-02:
Which LVGL version do you use? See `lvgl.h` or `lv_version.h`