lvgl/lv_drivers #311: LVGL 8.3+wayland can't find wayland shell

arilink, 2024-05-16:
weston version:10.0.0
LVGL version:8.3.0
LVGL driver  8.3.0

I tried to run LVGL on the rk3568 platform. The macro definition enables wayland and XDG shell, but the segment err is prompted when lvgl demo is run, so I further debug the problem that lvgl cannot find wayland shell. But my qt program can work normally, what could be the problem?

[image]

arilink, 2024-05-16:
[image]

arilink, 2024-05-16:
After searching the problem step by step, I found that weston10 did not have wl_shell interface, instead of xdg_shell, so I opened the macro definition of XDG_SHELL, searched the header file and lib again, and compiled and ran it. At this time, the following error occurred, I do not know how to solve it

arilink, 2024-05-16:
[image]

kisvegabor, 2024-05-20:
@daniel-rossier it might be useful for the LVGL v9 Wayland driver.

arilink, 2024-05-21:
So the lvgl v9 doesn't have this problem?

etag4048, 2024-05-21:
@arilink Did you update the ``lv_drv_conf.h`` and set ``LV_WAYLAND_XDG_SHELL`` to 1 ? Support for ``wl-shell`` has been deprecated for 3 years, it has been dropped in Weston 10.x

arilink, 2024-05-23:
Yes, I have set LV_WAYLAND_XDG_SHELL to 1 but it still won't display properly,

etag4048, 2024-05-24:
@arilink Looking at the screenshots more carefully, there is something abnormal in the wayland debug logs. It seems like ``create_window`` is called two times since ``xdg_toplevel@12.set_title`` is present two times, are you calling ``lv_wayland_create_window`` multiple times ? 

At it's core this problem is caused by a race condition that is apparent on  some weston configurations. The buffer is attached to the surface before weston sends the configure event. Upon the configure event ``xdg_surface_handle_configure`` is called where the ``surface_configured`` flag is set. Instead, in your case the ``surface_configured`` is set in ``create_graphic_obj`` and when  LVGL calls the flush cb ``_lv_wayland_flush``  on line 1992 the buffer gets attached too early.. 

Try commenting out the line 1411 ``obj->surface_configured = true;``
Add ``wl_display_dispatch(application.display);`` in ``create_window`` right after ``wl_surface_commit(window->body->surface);`` on line 1880. 

Also check why  ``xdg_toplevel@12.set_title`` is present two times.

Please avoid screenshots, you can capture the weston logs like so ``./your_app 2>/tmp/wayland_debug`` and then paste the logs it makes life easy :)

arilink, 2024-05-24:
@etag4048  I only called lv_wayland_create_window once in main.c file. You mean that surface was not fully configured when flush cb was used, so this problem occurred, right? I uploaded wayland.c, could you please help me adjust the code, because the code line number you described seems to be inconsistent with mine
[wayland.zip](https://github.com/lvgl/lv_drivers/files/15431558/wayland.zip)

etag4048, 2024-05-24:
@arilink Please apply the patch https://github.com/lvgl/lv_drivers/commit/c8d483f308abaf5d6796e1fbe7605af610a31709 this issue has been fixed in commit ``c8d483f`` by previous contributors. Another solution could be to update to ``release/v8.3`` the above answer is applicable on ``8cdabe8`` you  have to checkout the correct branch to be on version 8.3

arilink, 2024-05-24:
OK. I will try release/v8.3

arilink, 2024-05-27:
@etag4048   Still have err

root@ATK-DLRK356X:/opt/lvgl# ./lvgl-wayland2
[2572087.020]  -> wl_display@1.get_registry(new id wl_registry@2)
[2572087.187] wl_display@1.get_registry(new id wl_registry@2)
[2572087.561]  -> wl_registry@2.global(1, "wl_compositor", 4)
[2572088.017]  -> wl_registry@2.global(2, "wl_subcompositor", 1)
[2572088.511]  -> wl_registry@2.global(3, "wp_viewporter", 1)
[2572088.984]  -> wl_registry@2.global(4, "zxdg_output_manager_v1", 2)
[2572089.525]  -> wl_registry@2.global(5, "wp_presentation", 1)
[2572090.008]  -> wl_registry@2.global(6, "zwp_relative_pointer_manager_v1", 1)
[2572090.615]  -> wl_registry@2.global(7, "zwp_pointer_constraints_v1", 1)
[2572091.189]  -> wl_registry@2.global(8, "zwp_input_timestamps_manager_v1", 1)
[2572091.801]  -> wl_registry@2.global(9, "wl_data_device_manager", 3)
[2572092.340]  -> wl_registry@2.global(10, "wl_shm", 1)
[2572092.765]  -> wl_registry@2.global(11, "mali_buffer_sharing", 4)
[2572093.294]  -> wl_registry@2.global(12, "wl_seat", 7)
[2572093.730]  -> wl_registry@2.global(13, "zwp_linux_dmabuf_v1", 3)
[2572094.247]  -> wl_registry@2.global(14, "weston_direct_display_v1", 1)
[2572094.821]  -> wl_registry@2.global(15, "wl_output", 3)
[2572095.260]  -> wl_registry@2.global(16, "wl_output", 3)
[2572095.707]  -> wl_registry@2.global(17, "zwp_input_panel_v1", 1)
[2572096.233]  -> wl_registry@2.global(18, "zwp_input_method_v1", 1)
[2572096.750]  -> wl_registry@2.global(19, "zwp_text_input_manager_v1", 1)
[2572097.324]  -> wl_registry@2.global(20, "xdg_wm_base", 3)
[2572097.793]  -> wl_registry@2.global(21, "weston_desktop_shell", 1)
[2572098.325]  -> wl_registry@2.global(22, "weston_screenshooter", 1)
[2572098.982] wl_registry@2.global(1, "wl_compositor", 4)
[2572099.321]  -> wl_registry@2.bind(1, "wl_compositor", 1, new id [unknown]@3)
[2572099.907] wl_registry@2.global(2, "wl_subcompositor", 1)
[2572100.372]  -> wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@4)
[2572101.001] wl_registry@2.global(3, "wp_viewporter", 1)
[2572101.441] wl_registry@2.global(4, "zxdg_output_manager_v1", 2)
[2572101.956] wl_registry@2.global(5, "wp_presentation", 1)
[2572102.414] wl_registry@2.global(6, "zwp_relative_pointer_manager_v1", 1)
[2572102.983] wl_registry@2.global(7, "zwp_pointer_constraints_v1", 1)
[2572103.528] wl_registry@2.global(8, "zwp_input_timestamps_manager_v1", 1)
[2572104.109] wl_registry@2.global(9, "wl_data_device_manager", 3)
[2572104.608] wl_registry@2.global(10, "wl_shm", 1)
[2572105.008]  -> wl_registry@2.bind(10, "wl_shm", 1, new id [unknown]@5)
[2572105.683]  -> wl_shm@5.create_pool(new id wl_shm_pool@6, fd 5, 4096)
[2572106.353]  -> wl_shm_pool@6.resize(8832)
[2572106.520]  -> wl_shm_pool@6.resize(18624)
[2572106.848] wl_registry@2.global(11, "mali_buffer_sharing", 4)
[2572107.327] wl_registry@2.global(12, "wl_seat", 7)
[2572107.727]  -> wl_registry@2.bind(12, "wl_seat", 1, new id [unknown]@7)
[2572108.300] wl_registry@2.global(13, "zwp_linux_dmabuf_v1", 3)
[2572108.791] wl_registry@2.global(14, "weston_direct_display_v1", 1)
[2572109.325] wl_registry@2.global(15, "wl_output", 3)
[2572109.745] wl_registry@2.global(16, "wl_output", 3)
[2572110.166] wl_registry@2.global(17, "zwp_input_panel_v1", 1)
[2572110.654] wl_registry@2.global(18, "zwp_input_method_v1", 1)
[2572111.142] wl_registry@2.global(19, "zwp_text_input_manager_v1", 1)
[2572111.690] wl_registry@2.global(20, "xdg_wm_base", 3)
[2572112.125]  -> wl_registry@2.bind(20, "xdg_wm_base", 3, new id [unknown]@8)
[2572112.725] wl_registry@2.global(21, "weston_desktop_shell", 1)
[2572113.225] wl_registry@2.global(22, "weston_screenshooter", 1)
[2572113.737]  -> wl_display@1.sync(new id wl_callback@9)
[2572114.279] wl_registry@2.bind(1, "wl_compositor", 1, new id [unknown]@3)
[2572114.756] wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@4)
[2572115.347] wl_registry@2.bind(10, "wl_shm", 1, new id [unknown]@5)
[2572115.882]  -> wl_shm@5.format(0)
[2572116.162]  -> wl_shm@5.format(1)
[2572116.450]  -> wl_shm@5.format(909199186)
[2572116.793]  -> wl_shm@5.format(842093913)
[2572117.135]  -> wl_shm@5.format(842094158)
[2572117.478]  -> wl_shm@5.format(909203022)
[2572117.831]  -> wl_shm@5.format(1448695129)
[2572118.176]  -> wl_shm@5.format(1448434008)
[2572118.531]  -> wl_shm@5.format(808665665)
[... 27 lines cut ...]
[2572129.005]  -> xdg_surface@12.get_toplevel(new id xdg_toplevel@13)
[2572129.536]  -> xdg_toplevel@13.set_title("Window Title")
[2572129.993]  -> xdg_toplevel@13.set_app_id("Window Title")
[2572130.504]  -> wl_shm@5.create_pool(new id wl_shm_pool@14, fd 6, 1536000)
[2572131.040]  -> wl_shm_pool@14.create_buffer(new id wl_buffer@15, 0, 800, 480, 3200, 0)
[2572152.284]  -> wl_surface@9.damage(0, 0, 800, 480)
[2572152.392]  -> wl_surface@9.attach(wl_buffer@15, 0, 0)
[2572152.693]  -> wl_surface@9.commit()
[2572153.205] wl_seat@7.get_keyboard(new id wl_keyboard@10)
[2572153.678]  -> wl_keyboard@10.keymap(1, fd 57, 48967)
[2572153.922] wl_seat@7.get_touch(new id wl_touch@11)
[2572154.312] wl_compositor@3.create_surface(new id wl_surface@9)
[2572154.850] xdg_wm_base@8.get_xdg_surface(new id xdg_surface@12, wl_surface@9)
[2572155.437] xdg_surface@12.get_toplevel(new id xdg_toplevel@13)
[2572155.928] xdg_toplevel@13.set_title("Window Title")
[2572156.357] xdg_toplevel@13.set_app_id("Window Title")
[2572156.794] wl_shm@5.create_pool(new id wl_shm_pool@14, fd 55, 1536000)
[2572157.382] wl_shm_pool@14.create_buffer(new id wl_buffer@15, 0, 800, 480, 3200, 0)
[2572158.014] wl_surface@9.damage(0, 0, 800, 480)
[2572158.396] wl_surface@9.attach(wl_buffer@15, 0, 0)
[2572158.802] wl_surface@9.commit()
[2572159.173]  -> wl_display@1.error(xdg_surface@12, 3, "xdg_surface has never been configured")
[08:18:15.109] libwayland: error in client communication (pid 5980)
[2572163.329] wl_display@1.error(xdg_surface@12, 3, "xdg_surface has never been configured")
xdg_surface@12: error 3: xdg_surface has never been configured

arilink, 2024-05-28:
@etag4048    
After replacing the Driver of release/v8.3, lvgl can display normally, but there will be other problems, as follows
1.RK3568 hardware supports multi-screen display, so does weston. I plan to use HDMI as the main display, run Qt app, and DSI as the secondary display to display lvgl app
2. I start the Qt app first, and once I run lvgl->demo qt app the app screen disappears
3. In the/etc/XDG/weston/weston. Ini. I configured two outputs and corresponding app ids
Where could this be a problem?

arilink, 2024-05-28:
lvgl app start log is blow:
root@ATK-DLRK356X:~# /opt/lvgl/lvgl-wayland2
[ 854435.514]  -> wl_display@1.get_registry(new id wl_registry@2)
[ 854435.711] wl_display@1.get_registry(new id wl_registry@2)
[ 854435.773]  -> wl_registry@2.global(1, "wl_compositor", 4)
[ 854435.824]  -> wl_registry@2.global(2, "wl_subcompositor", 1)
[ 854435.848]  -> wl_registry@2.global(3, "wp_viewporter", 1)
[ 854435.870]  -> wl_registry@2.global(4, "zxdg_output_manager_v1", 2)
[ 854435.894]  -> wl_registry@2.global(5, "wp_presentation", 1)
[ 854435.917]  -> wl_registry@2.global(6, "zwp_relative_pointer_manager_v1", 1)
[ 854435.940]  -> wl_registry@2.global(7, "zwp_pointer_constraints_v1", 1)
[ 854435.962]  -> wl_registry@2.global(8, "zwp_input_timestamps_manager_v1", 1)
[ 854435.987]  -> wl_registry@2.global(9, "wl_data_device_manager", 3)
[ 854436.009]  -> wl_registry@2.global(10, "wl_shm", 1)
[ 854436.034]  -> wl_registry@2.global(11, "mali_buffer_sharing", 4)
[ 854436.058]  -> wl_registry@2.global(12, "wl_seat", 7)
[ 854436.081]  -> wl_registry@2.global(13, "zwp_linux_dmabuf_v1", 3)
[ 854436.103]  -> wl_registry@2.global(14, "weston_direct_display_v1", 1)
[ 854436.125]  -> wl_registry@2.global(15, "wl_output", 3)
[ 854436.148]  -> wl_registry@2.global(16, "wl_output", 3)
[ 854436.172]  -> wl_registry@2.global(17, "xdg_wm_base", 3)
[ 854436.195]  -> wl_registry@2.global(18, "weston_screenshooter", 1)
[ 854436.342] wl_registry@2.global(1, "wl_compositor", 4)
[ 854436.403]  -> wl_registry@2.bind(1, "wl_compositor", 1, new id [unknown]@3)
[ 854436.447] wl_registry@2.global(2, "wl_subcompositor", 1)
[ 854436.511]  -> wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@4)
[ 854436.548] wl_registry@2.global(3, "wp_viewporter", 1)
[ 854436.604] wl_registry@2.global(4, "zxdg_output_manager_v1", 2)
[ 854436.647] wl_registry@2.global(5, "wp_presentation", 1)
[ 854436.699] wl_registry@2.global(6, "zwp_relative_pointer_manager_v1", 1)
[ 854436.741] wl_registry@2.global(7, "zwp_pointer_constraints_v1", 1)
[ 854436.771] wl_registry@2.global(8, "zwp_input_timestamps_manager_v1", 1)
[ 854436.799] wl_registry@2.global(9, "wl_data_device_manager", 3)
[ 854436.856] wl_registry@2.global(10, "wl_shm", 1)
[ 854436.919]  -> wl_registry@2.bind(10, "wl_shm", 1, new id [unknown]@5)
[ 854437.105]  -> wl_shm@5.create_pool(new id wl_shm_pool@6, fd 5, 4096)
[ 854437.881]  -> wl_shm_pool@6.resize(8832)
[ 854438.011]  -> wl_shm_pool@6.resize(18624)
[ 854438.077] wl_registry@2.global(11, "mali_buffer_sharing", 4)
[ 854438.139] wl_registry@2.global(12, "wl_seat", 7)
[ 854438.188]  -> wl_registry@2.bind(12, "wl_seat", 1, new id [unknown]@7)
[ 854438.263] wl_registry@2.global(13, "zwp_linux_dmabuf_v1", 3)
[ 854438.317] wl_registry@2.global(14, "weston_direct_display_v1", 1)
[ 854438.361] wl_registry@2.global(15, "wl_output", 3)
[ 854438.412] wl_registry@2.global(16, "wl_output", 3)
[ 854438.461] wl_registry@2.global(17, "xdg_wm_base", 3)
[ 854438.501]  -> wl_registry@2.bind(17, "xdg_wm_base", 3, new id [unknown]@8)
[ 854438.575] wl_registry@2.global(18, "weston_screenshooter", 1)
[ 854438.620]  -> wl_display@1.sync(new id wl_callback@9)
[ 854438.757] wl_registry@2.bind(1, "wl_compositor", 1, new id [unknown]@3)
[ 854438.888] wl_registry@2.bind(2, "wl_subcompositor", 1, new id [unknown]@4)
[ 854438.955] wl_registry@2.bind(10, "wl_shm", 1, new id [unknown]@5)
[ 854438.997]  -> wl_shm@5.format(0)
[ 854439.026]  -> wl_shm@5.format(1)
[ 854439.048]  -> wl_shm@5.format(909199186)
[ 854439.084]  -> wl_shm@5.format(842093913)
[ 854439.105]  -> wl_shm@5.format(842094158)
[ 854439.118]  -> wl_shm@5.format(909203022)
[ 854439.130]  -> wl_shm@5.format(1448695129)
[ 854439.159]  -> wl_shm@5.format(1448434008)
[ 854439.194]  -> wl_shm@5.format(808665665)
[ 854439.221]  -> wl_shm@5.format(808665688)
[ 854439.244] wl_shm@5.create_pool(new id wl_shm_pool@6, fd 48, 4096)
[ 854439.373] wl_shm_pool@6.resize(8832)
[ 854439.436] wl_shm_pool@6.resize(18624)
[ 854439.483] wl_registry@2.bind(12, "wl_seat", 1, new id [unknown]@7)
[ 854439.573]  -> wl_seat@7.capabilities(6)
[ 854439.608] wl_registry@2.bind(17, "xdg_wm_base", 3, new id [unknown]@8)
[ 854439.700] wl_display@1.sync(new id wl_callback@9)
[ 854439.725]  -> wl_callback@9.done(4)
[... 101 lines cut ...]
[ 854570.332]  -> wl_surface@9.attach(wl_buffer@16, 0, 0)
[ 854570.414]  -> wl_surface@9.commit()
[ 854570.565] wl_surface@9.damage(572, 185, 110, 46)
[ 854570.666] wl_surface@9.attach(wl_buffer@16, 0, 0)
[ 854570.718] wl_surface@9.commit()
[ 854572.269]  -> wl_buffer@16.release()
[ 854581.056] wl_buffer@16.release()
[ 854591.880]  -> wl_surface@9.damage(572, 185, 110, 46)
[ 854591.984]  -> wl_surface@9.attach(wl_buffer@16, 0, 0)
[ 854592.079]  -> wl_surface@9.commit()
[ 854592.232] wl_surface@9.damage(572, 185, 110, 46)
[ 854592.379] wl_surface@9.attach(wl_buffer@16, 0, 0)
[ 854592.439] wl_surface@9.commit()
[ 854594.091]  -> wl_buffer@16.release()
[ 854602.975] wl_buffer@16.release()
[ 903844.951] wl_surface@21.frame(new id wl_callback@30)
[ 903854.355] wl_display@1.sync(new id wl_callback@31)
[ 903854.540]  -> wl_callback@31.done(9)
[ 903854.729]  -> wl_display@1.delete_id(31)
[ 903854.841] wl_surface@21.attach(wl_buffer@27, 0, 0)
[ 903855.158] wl_surface@21.damage(0, 0, 2147483647, 2147483647)
[ 903855.737] wl_surface@21.commit()
[ 903856.043]  -> wl_buffer@29.release()
[ 903862.831]  -> wl_callback@30.done(604276)
[ 903862.941]  -> wl_display@1.delete_id(30)

arilink, 2024-05-29:
I think this problem is about window geometry

etag4048, 2024-05-29:
@arilink Yes it is indeed, this is a different problem that is also apparent when running older versions of the wayland driver on weston.. The flush occurs before the buffers are resized. 

It looks like weston runs the application in fullscreen right at the beginning. Upon the configure event the compositor communicates the size it wants the client app to have:

`[ 854478.215] [xdg_toplevel@13.configure] xdg_toplevel@13.configure(720, 1280, array[4])`

In this case 720*1280

As a workaround, try passing these values to  the ``lv_wayland_create_window`` function in your main.c file

This issue is fixed in the opened PR. 

BR

arilink, 2024-05-31:
@etag4048  
I found a way to do it, which is as follows:
1.lvgl seems to prefer to display on the main display
2. I designated DSI as the primary monitor and HDMI as the secondary monitor
3. Run Qt program on the secondary monitor Qt can be specified by set_geometry on HDMI display
4. Run lvgl's demo on the main display
5. It seems to be working well and there is no tear. The photos are as follows:
[image]

In theory, lvgl should call xdg_surface_set_window_geometry to specify the display location, but when I try, I'm told that xdg_surface must have a role. I think this is related to the way I call, could you please help confirm the correct calling method, the specific log is as follows:
[13:45:23.809] libwayland: error in client communication (pid 3277)
xdg_surface@12: error 1: xdg_surface must have a role

etag4048, 2024-06-01:
@arilink 
A xdg surface can have a top-level role or popup role.
Before sending any request related to a surface the role needs to be set, the ``get_xdg_surface`` request assigns the top-level role to a  ``wl_surface``, So you need to 'paste' the function after ``get_xdg_surface`` in ``create_window``.

However, sending a ``xdg_surface_set_window_geometry`` request to weston, won't solve this issue. The book on wayland explains the purpose of this request better than the official spec https://wayland-book.com/xdg-shell-basics/xdg-surface.html

' ...is used primarily for applications using client-side decorations, to distinguish the parts of their surface which are considered a part of the window, and the parts which are not. Most commonly, this is used to exclude client-side drop-shadows rendered behind the window from being considered a part of it.'

There are no decorations and no windowing in the above screenshot, both apps are running in full screen.

The proper solution to this is to resize the underlying SHM buffers before the first flush occurs.

BR

arilink, 2024-06-03:
@etag4048 
As you said, set_window_geometry doesn't work. How do I resize the shm?