lvgl/lv_drivers #314: in windows，cmake is error

xinsuinizhuan, 2024-05-30:
the lv_drv_conf.h need ../../lv_drv_conf.h， but only have the ../lv_drv_conf.h
to install is error, to loop create he path inlcude/install not stop

kisvegabor, 2024-06-10:
How to reproduce this error?

xinsuinizhuan, 2024-06-21:
> How to reproduce this error?

in windwos use the cmake make the cmakelist, then compile by vs2019

kisvegabor, 2024-06-21:
I'm a Linux guy, cannot comment on vs2019 :slightly_frowning_face:

xinsuinizhuan, 2024-06-25:
in linux,run make install, could generate the all install file?

kisvegabor, 2024-06-25:
Ah, just realized that it's about `lv_driver`s. It seems you shall to create an `lv_drv_conf.h` file too where you enable the drievrs you need.

Note that this repo will be archived soon.

xinsuinizhuan, 2024-07-24:
> I'm a Linux guy, cannot comment on vs2019 🙁
i I compile lv_driver separately
in linux is same，i make it, then make install, loop create /build/install, not stop:

[100%] Built target lvgl_examples
firefly@firefly:~/lvgl/lvgl-release-v8.3/build$ make install
[ 44%] Built target lvgl
[ 59%] Built target lvgl_demos
[100%] Built target lvgl_examples
Install the project...
-- Install configuration: ""
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/stm32_dma2d
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/lv_draw_label.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/lv_img_decoder.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/lv_draw_triangle.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/lv_draw_arc.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/vglite
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/vglite/lv_draw_vglite_blend.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/vglite/lv_draw_vglite.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/vglite/lv_vglite_buf.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/vglite/lv_vglite_utils.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/vglite/lv_draw_vglite_line.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/vglite/lv_draw_vglite_rect.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/vglite/lv_draw_vglite_arc.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/pxp
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/pxp/lv_gpu_nxp_pxp_osa.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/pxp/lv_gpu_nxp_pxp.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/pxp/lv_draw_pxp.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/nxp/pxp/lv_draw_pxp_blend.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/lv_img_cache.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/lv_img_buf.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/swm341_dma2d
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/swm341_dma2d/lv_gpu_swm341_dma2d.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/lv_draw_mask.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/lv_draw_layer.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/renesas
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/renesas/lv_gpu_d2_ra6m3.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sw
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sw/lv_draw_sw_blend.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sw/lv_draw_sw_gradient.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sw/lv_draw_sw.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sw/lv_draw_sw_dither.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl_texture_cache.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl_stack_blur.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl_utils.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl_priv.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl_composite.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl_rect.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl_img.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl_layer.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/draw/sdl/lv_draw_sdl_mask.h
[... cut ...]
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/extra/others/gridnav
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/extra/others/gridnav/lv_gridnav.h
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/extra/widgets
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/extra/widgets/chart
-- Installing: /home/firefly/lvgl/lvgl-release-v8.3/build/install/include/lvgl//src/extra/widgets/chart/lv_chart.h

kisvegabor, 2024-08-03:
It seems it was working. If so, please close this issue.