lvgl/lv_drivers #316: only have the lv_win32_init interface, how to de-init?

xinsuinizhuan, 2024-06-21:
in windows:
i init,then uinit，the re-init, false:

bool single_display_mode_initialization()
{
	HMODULE curmodel = GetModuleHandleW(NULL);
	if (!lv_win32_init(
		curmodel,
		SW_SHOW,
		480,
		320,
		NULL))
	{
		return false;
	}
	lv_win32_add_all_input_devices_to_group(NULL);
	return true;
}

init:
	lv_init();
	if (!single_display_mode_initialization())
	{
		return false;
	}
unit:
lv_deinit();

i init the window, but i want destroy it, then re-init the windows,but lv_win32_init return false! how to uint?

xinsuinizhuan, 2024-07-08:
how about this issue?

kisvegabor, 2024-07-10:
cc @MouriNaruto 

Note that this repository will be archived in the feature. Instead, I recommend using LVGL v9 where we have the same driver [built into LVGL](https://github.com/lvgl/lvgl/tree/master/src/drivers/windows).

xinsuinizhuan, 2024-07-18:
I want upgrade my project from v8 to v9, but so many functions dismiss，could not inherit。I hope could repair this issue, 3q

xinsuinizhuan, 2024-07-29:
could repaire this issue?

kisvegabor, 2024-08-07:
I don't understand the issue 🙁 Could you explain it in a different way?

xinsuinizhuan, 2024-08-12:
I use lv_win32_init to init a window,and pop it, but now i need close it, then repop, how to unit the window? like create and destroy, i lv_win32_init (create), how to lv_win32_uinit(destroy)?

kisvegabor, 2024-08-13:
If there is no `lv_win32_uinit` function now, probably you need to add it manually. :slightly_frowning_face: