lvgl/lv_drivers #317: RK3568 LVGL8.3.10 wayland display data abnormality

eYoung8475, 2024-10-09:
RK3568 weston version: 12.0.1
LVGL: 8.3.10
wayland: 7482c7c1ed33c8b6944241aed9f38208a880a507

I created a tabview, one page is [商品分类], and the other page is [快速查找].
When I click the page [商品分类] product button, the product will be added to the shopping cart on the right. 
If I quickly click, it is easy for some content from the [商品分类] page to overlap with the [快速查找] page.

Previously, I was on RK1808, LVGL8.3.10, weston 3.0， I did not find this issue when using Wayland driver without smm.

[image]
[image]
[image]

https://github.com/user-attachments/assets/ed5659b4-2a70-4c91-9774-0b9df60f89f6

kisvegabor, 2024-10-14:
I suggest printing the x and y coordinates in the touch's `read_cb` to better understand what's happening.

eYoung8475, 2024-10-15:
The following are the touch coordinates printed during the process from normal display to abnormal display. The screen resolution is 1366 * 768, and there seems to be no issue with abnormal touch points.
I tried to replace Wayland with release/v8.3 and repeated the test again, and it showed no problem. Is it possible that the data was dirty due to smm.

[image]

touch: x: 742, y: 201
touch: x: 741, y: 204
touch: x: 742, y: 201
touch: x: 708, y: 272
touch: x: 708, y: 272
  10:57:47 交易1.000个的<20020:多味瓜子>
touch: x: 736, y: 190
touch: x: 736, y: 190
touch: x: 736, y: 190
  10:57:47 交易1.000个的<20020:多味瓜子>
touch: x: 687, y: 256
touch: x: 687, y: 256
  10:57:47 交易1.000个的<20020:多味瓜子>
touch: x: 754, y: 205
touch: x: 753, y: 205
touch: x: 752, y: 207
touch: x: 748, y: 210
touch: x: 743, y: 215
touch: x: 736, y: 222
touch: x: 728, y: 230
touch: x: 720, y: 238
touch: x: 720, y: 238
  10:57:47 交易1.000个的<20020:多味瓜子>
touch: x: 746, y: 213
touch: x: 746, y: 212
touch: x: 745, y: 212
touch: x: 743, y: 214
touch: x: 740, y: 217
touch: x: 735, y: 221
touch: x: 729, y: 226
touch: x: 729, y: 226
  10:57:47 交易1.000个的<20020:多味瓜子>
touch: x: 742, y: 199
touch: x: 742, y: 199
  10:57:47 交易1.000个的<20020:多味瓜子>
touch: x: 689, y: 266
touch: x: 689, y: 266
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 740, y: 207
touch: x: 740, y: 207
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 692, y: 275
touch: x: 692, y: 275
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 730, y: 187
touch: x: 730, y: 187
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 687, y: 263
touch: x: 687, y: 263
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 736, y: 216
touch: x: 735, y: 217
touch: x: 734, y: 218
touch: x: 730, y: 221
touch: x: 726, y: 224
touch: x: 720, y: 230
touch: x: 713, y: 236
touch: x: 706, y: 243
touch: x: 699, y: 249
touch: x: 699, y: 249
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 734, y: 228
touch: x: 733, y: 228
touch: x: 732, y: 228
touch: x: 732, y: 228
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 686, y: 277
touch: x: 686, y: 277
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 717, y: 210
touch: x: 717, y: 209
touch: x: 717, y: 209
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 680, y: 260
touch: x: 680, y: 260
  10:57:48 交易1.000个的<20020:多味瓜子>
touch: x: 726, y: 189
touch: x: 726, y: 189
  10:57:49 交易1.000个的<20020:多味瓜子>
touch: x: 690, y: 270
touch: x: 690, y: 270
  10:57:49 交易1.000个的<20020:多味瓜子>
touch: x: 745, y: 199
touch: x: 745, y: 199
  10:57:49 交易1.000个的<20020:多味瓜子>
touch: x: 705, y: 268
touch: x: 705, y: 268
  10:57:49 交易1.000个的<20020:多味瓜子>
touch: x: 753, y: 204
touch: x: 751, y: 205
touch: x: 748, y: 209
touch: x: 742, y: 214
touch: x: 734, y: 221
touch: x: 726, y: 230
touch: x: 717, y: 238
touch: x: 708, y: 247
touch: x: 700, y: 255
touch: x: 694, y: 262
touch: x: 694, y: 262
  10:57:49 交易1.000个的<20020:多味瓜子>
touch: x: 720, y: 213
touch: x: 719, y: 213
touch: x: 717, y: 215
touch: x: 714, y: 218
touch: x: 710, y: 222
touch: x: 704, y: 226
touch: x: 699, y: 231
touch: x: 694, y: 236
touch: x: 689, y: 241
touch: x: 686, y: 245
touch: x: 686, y: 245
  10:57:49 交易1.000个的<20020:多味瓜子>
touch: x: 732, y: 197
touch: x: 731, y: 198
touch: x: 729, y: 200
touch: x: 729, y: 200
  10:57:49 交易1.000个的<20020:多味瓜子>
touch: x: 680, y: 260
touch: x: 680, y: 260
  10:57:49 交易1.000个的<20020:多味瓜子>
touch: x: 740, y: 200
touch: x: 739, y: 201
touch: x: 737, y: 203
touch: x: 734, y: 207
touch: x: 730, y: 213
touch: x: 725, y: 221
touch: x: 720, y: 229
touch: x: 714, y: 237
touch: x: 709, y: 244
touch: x: 709, y: 244
  10:57:50 交易1.000个的<20020:多味瓜子>
touch: x: 721, y: 205
touch: x: 721, y: 205
  10:57:50 交易1.000个的<20020:多味瓜子>
touch: x: 692, y: 273
touch: x: 693, y: 273
touch: x: 694, y: 273
touch: x: 694, y: 274
touch: x: 694, y: 274
  10:57:50 交易1.000个的<20020:多味瓜子>
touch: x: 741, y: 209
touch: x: 741, y: 209
  10:57:50 交易1.000个的<20020:多味瓜子>
touch: x: 714, y: 290
touch: x: 714, y: 290
touch: x: 757, y: 226
touch: x: 757, y: 227
touch: x: 756, y: 229
touch: x: 753, y: 232
touch: x: 750, y: 238
touch: x: 745, y: 246
touch: x: 740, y: 256
touch: x: 735, y: 265
touch: x: 730, y: 274
touch: x: 726, y: 282
touch: x: 726, y: 282
  10:57:50 交易1.000个的<20020:多味瓜子>
touch: x: 742, y: 220
touch: x: 741, y: 220
touch: x: 740, y: 221
touch: x: 740, y: 221
  10:57:50 交易1.000个的<20020:多味瓜子>
touch: x: 692, y: 288
touch: x: 692, y: 289
touch: x: 692, y: 290
touch: x: 691, y: 292
touch: x: 691, y: 292
  10:57:50 交易1.000个的<20020:多味瓜子>
touch: x: 710, y: 228
touch: x: 710, y: 228
  10:57:50 交易1.000个的<20020:多味瓜子>
touch: x: 679, y: 286
touch: x: 679, y: 286
  10:57:51 交易1.000个的<20020:多味瓜子>
touch: x: 710, y: 190
touch: x: 710, y: 190
  10:57:51 交易1.000个的<20020:多味瓜子>
touch: x: 686, y: 268
touch: x: 686, y: 268
  10:57:51 交易1.000个的<20020:多味瓜子>
touch: x: 710, y: 176
touch: x: 710, y: 176
  10:57:51 交易1.000个的<20020:多味瓜子>
touch: x: 686, y: 261
touch: x: 686, y: 261
  10:57:51 交易1.000个的<20020:多味瓜子>
touch: x: 725, y: 189
touch: x: 725, y: 189
  10:57:51 交易1.000个的<20020:多味瓜子>
touch: x: 690, y: 271
touch: x: 690, y: 271
  10:57:51 交易1.000个的<20020:多味瓜子>
touch: x: 751, y: 192
touch: x: 751, y: 192
  10:57:51 交易1.000个的<20020:多味瓜子>
touch: x: 720, y: 262
touch: x: 720, y: 262
  10:57:51 交易1.000个的<20020:多味瓜子>
touch: x: 758, y: 200
touch: x: 758, y: 200
  10:57:51 交易1.000个的<20020:多味瓜子>

kisvegabor, 2024-10-16:
The data rally look good. I cc @etag4048 who implemented our new Wayland driver.